zephyrproject-rtos/zephyr #57682: LE Audio: TMAP BT Tester application

Thalley, 2023-05-09:
**Is your feature request related to a problem? Please describe.**
We should have a TMAP BT Tester implementation. The implementation should support the BTP similar to the other profile implementation for the BT tester, and thus be usuable for both PTS testing, as well as other automatic testing that supports the BTP. 

**Describe the solution you'd like**
Implement support for TMAP in the bttester (`tests/bluetooth/tester`) which support the BTP commands such that the Zephyr application can perform specific operations. 

**Describe alternatives you've considered**
See the other profile implementations for inspiration or contact @sjanc for more information. 

**Additional context**
For TMAP we probably want to support multiple connections, which may not have been done before with BTP, so some new things may need to be designed.

Thalley, 2023-05-09:
@sp3030 will take this, but I cannot assign them to it.

sp3030, 2024-02-12:
Covered by PR https://github.com/zephyrproject-rtos/zephyr/pull/68675

closed 2024-03-01 by aescolar