zephyrproject-rtos/zephyr #57690: twister: Allow specifying location of source files

nordicjm, 2023-05-09:
**Is your enhancement proposal related to a problem? Please describe.**
At present, a twister .yaml file needs to be present in the same folder as the zephyr project files, it would be useful if this could e expanded so that the .yaml could (optionally) specify a different folder where the files are located.

**Describe the solution you'd like**
An optional field in the .yaml file which specifies where the project files are located, this could be a relative or absolute path, it would be beneficial if it also supported using zephyr module path variables.

**Describe alternatives you've considered**
Leaving it as it is

nordicjm, 2023-05-16:
Fixed by #57806

PerMac, 2023-05-18:
I don't think #57806 is solving this request (even though @nordicjm you're the one creating it;)) The functionality introduced in #57086 is very specific one. It allows to shadow the yaml file from zephyr tree with one from another repo, if the repos' structures are matched. I believe there is still a need to be able to have an in tree folder with yaml, and in that yaml it is explicitly told where sources are located. A use case for that are e.g. pytest tests for already existing samples/apps

nashif, 2023-05-18:
> A use case for that are e.g. pytest tests for already existing samples/apps

this leaves me owndering why would you have this out of the tree, pytest is the test itself which means if you are going to change that, it will be a different test and having this split across trees does not make sense to me at all and will cause conflicts and confusion, i.e. test fails out of tree and bug is reported in zephyr or vice versa. At this point, if you have custom tests you want to maintain downstream, I would maintain them in their entirty there.

nordicjm, 2023-05-22:
The original feature that I was asking for has been implemented

nashif, 2023-08-17:
looking at this again, I believe it is implemented already.

closed 2023-08-17 by nashif