zephyrproject-rtos/zephyr #57696: ESP32-C3 not displaying "hello world!"

Ananyaaynana, 2023-05-09:
I am using the zephyr sample application from github to understand the zephyr RTOS. I modified the main.c to "Hello world!" on ESP32-C3.
`/*
 * Copyright (c) 2012-2014 Wind River Systems, Inc.
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr/kernel.h>

void main(void)
{
	printk("Hello World! %s\n", CONFIG_BOARD);
}`

To add the ESP32-C3 files such as hal_espressif and stm32 to my sample-application I modified the west.yml in the my-workspace2/example-application directory:

`# Copyright (c) 2021 Nordic Semiconductor ASA

manifest:
  self:
    west-commands: scripts/west-commands.yml

  remotes:
    - name: zephyrproject-rtos
      url-base: https://github.com/zephyrproject-rtos

  projects:
    - name: zephyr
      remote: zephyrproject-rtos
      revision: main
      import:
        # By using name-allowlist we can clone only the modules that are
        # strictly needed by the application.
        name-whitelist:
          - cmsis      # required by the ARM port
          - edtt
          - hal_espressif
          - hal_stm32
          - tinycrypt
          - zscilib` 
   
The prj.conf file is as follows:
`
CONFIG_ZSL=y
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y

CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y

CONFIG_THREAD_NAME=y

CONFIG_REBOOT=y

CONFIG_EVENTS=y
CONFIG_HWINFO=y

CONFIG_LOG=y
CONFIG_LOG_MODE_MINIMAL=y

CONFIG_SOC_ESP32C3=y
CONFIG_CONSOLE=y
CONFIG_SERIAL=y
CONFIG_UART_CONSOLE=y
CONFIG_UART_INTERRUPT_DRIVEN=y

CONFIG_I2C=y

CONFIG_CAN=y
CONFIG_ISOTP=y
[... 333 lines cut ...]
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
I (43) boot: ESP-IDF 0b5aa325a 2nd stage bootloader
I (43) boot: compile time 16:46:18
I (43) boot: chip revision: 3
I (45) boot.esp32c3: SPI Speed      : 40MHz
I (49) boot.esp32c3: SPI Mode       : DIO
I (54) boot.esp32c3: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00100000
I (98) boot: End of partition table
I (102) esp_image: segment 0: paddr=00010020 vaddr=00000020 size=0001ch (    28) 
I (110) esp_image: segment 1: paddr=00010044 vaddr=3fc89b00 size=009a8h (  2472) load
I (119) esp_image: segment 2: paddr=000109f4 vaddr=3fc8a4a8 size=0030ch (   780) load
I (127) esp_image: segment 3: paddr=00010d08 vaddr=40380000 size=07df8h ( 32248) load
I (143) esp_image: segment 4: paddr=00018b08 vaddr=00000000 size=07530h ( 30000) 
I (150) esp_image: segment 5: paddr=00020040 vaddr=3c000040 size=016e0h (  5856) map
I (153) esp_image: segment 6: paddr=00021728 vaddr=00000000 size=0e8f0h ( 59632) 
I (173) esp_image: segment 7: paddr=00030020 vaddr=42010020 size=04178h ( 16760) map
I (180) boot: Loaded app from partition at offset 0x10000

sylvioalves, 2023-05-09:
@Ananyaaynana I have just tried `samples/hello_world` embeddding your configuration and it worked. What is the revision you've been working on? Have you run `west update` as well?

Ananyaaynana, 2023-05-12:
I am using the v3.2-branch revision of zephyr. Yes I have run west update.

closed 2023-05-22 by Ananyaaynana

MrSuicideParrot, 2023-07-01:
@Ananyaaynana could you share how did you solve this issue?

Ananyaaynana, 2023-07-06:
@MrSuicideParrot I ran west update again.

alkurop, 2024-06-28:
Hello. I also stumbled upon this issue of serial monitor.
After some investigation, it appears that the `uart` port on Esp32c3 is desabled by default configuration.

To make my life easier, I used `nrfConnect` plugin for the VSCode. In there I can reconfigure tasks to build project and debug.
```
{
	"version": "2.0.0",
	"tasks": [
		{
			"label": "west build esp32c3",
			"type": "shell",
			"command": "west build -p always -b esp32c3_devkitm  ."
		},
		{
			"label": "west flash esp32c3",
			"type": "shell",
			"command": "west flash --esp-baud-rate 115200 --esp-device /dev/cu.usbmodem1114201 ;  west espressif monitor --p /dev/cu.usbmodem1114201 "
		},
	]
}
```
And then click on the Flash and Build button and reconfigure to use tasks

This nrf plugin has a very nice feature `ViewDeviceTree`. It is a gui to configure the `esp32_....overlay`. I played a little bit with the configuration, and it turned out that to enable the monitor output, there is a line of code to add to the overlay.
```
&uart0 {
	zephyr,deferred-init;
};
```
 Moreover, I'm now following nrf's tutorials - they are pretty useful in my case for a board Esp23C3, because nrf totally uses Zephyr. https://academy.nordicsemi.com/courses/nrf-connect-sdk-fundamentals/lessons/lesson-1-nrf-connect-sdk-introduction/topic/exercise-2-1/
 The only difference is I have to tweak the overlay file for the Esp32C3 board.

sylvioalves, 2024-06-28:
@alkurop which Zephyr revision (and SDK)?

alkurop, 2024-06-28:
`zephyr-sdk-0.16.1`. I probably need to update to 16.8
I'm on master branch for the `zephyrproject`

sylvioalves, 2024-06-28:
Can you paste the whole output and branch revision? I can successfully run that here:
```c
[esp32c3] [INF] IROM segment: paddr=00010000h, vaddr=42000000h, size=03092h ( 12434) map
*** Booting Zephyr OS build v3.7.0-rc1-542-g1159c2addaac ***
Hello World! esp32c3_devkitm/esp32c3
```

alkurop, 2024-06-28:
I'm on this commit. 
```
commit 895f246227e2ce78fcea54d29dcfd730633ad9be (HEAD -> main)
Author: Glenn Andrews <glenn.andrews.42@gmail.com>
Date:   Sat Jun 15 08:32:37 2024 -0700
```
Do you think it's better to check out a release branch?

My log without the modification of the overlay looks like this
```
Wrote 147456 bytes at 0x00000000 in 1.0 seconds (1198.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
--- idf_monitor on /dev/cu.usbmodem1114201 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038168a
0x4038168a: arch_cpu_idle at /Users/alex/zephyrproject/zephyr/arch/riscv/core/cpu_idle.c:13

SPIWP:0xee
mode:DIO, clock div:2
load:0x3fc89820,len:0x1a6c
load:0x40380000,len:0x9810
0x40380000: iram_vma at ??:?

SHA-256 comparison failed:
Calculated: 25b01b3a6ef98d3662b1b3da27872db316ccb0107dcdbc649e9b3eb49dbbf428
Expected: 00000000404d0000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x403815c2
0x403815c2: __start at /Users/alex/zephyrproject/zephyr/soc/espressif/common/loader.c:227

I (60) boot: ESP Simple boot
I (60) boot: compile time Jun 28 2024 21:51:16
I (60) boot: Multicore bootloader
I (61) spi_flash: detected chip: generic
I (62) spi_flash: flash io: dio
W (65) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (77) boot: chip revision: v0.4
I (80) boot.esp32c3: SPI Speed      : 40MHz
I (84) boot.esp32c3: SPI Mode       : SLOW READ
I (88) boot.esp32c3: SPI Flash Size : 4MB
I (92) boot: Enabling RNG early entropy source...
[esp32c3] [INF] DRAM: lma 0x00000020 vma 0x3fc89820 len 0x1a6c   (6764)
[esp32c3] [INF] IRAM: lma 0x00001a94 vma 0x40380000 len 0x9810   (38928)
0x40380000: iram_vma at ??:?

[esp32c3] [INF] padd: lma 0x0000b2b8 vma 0x00000000 len 0x4d40   (19776)
[esp32c3] [INF] IMAP: lma 0x00010000 vma 0x42000000 len 0x32f0   (13040)
0x42000000: _OffsetAbsSyms at /Users/alex/zephyrproject/zephyr/arch/riscv/core/offsets/offsets.c:140

[esp32c3] [INF] padd: lma 0x000132f8 vma 0x00000000 len 0xcd00   (52480)
[esp32c3] [INF] DMAP: lma 0x00020000 vma 0x3c010000 len 0xff0    (4080)
[esp32c3] [INF] Image with 6 segments
[esp32c3] [INF] DROM segment: paddr=00020000h, vaddr=3c010000h, size=00FF0h (  4080) map
[esp32c3] [INF] IROM segment: paddr=00010000h, vaddr=42000000h, size=032F0h ( 13040) map
```

alkurop, 2024-06-28:
With overlay as described, it now looks like this
```
Wrote 147456 bytes at 0x00000000 in 1.0 seconds (1205.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
--- idf_monitor on /dev/cu.usbmodem1114201 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038168a
0x4038168a: arch_cpu_idle at /Users/alex/zephyrproject/zephyr/arch/riscv/core/cpu_idle.c:13

SPIWP:0xee
mode:DIO, clock div:2
load:0x3fc89820,len:0x1a6c
load:0x40380000,len:0x9810
0x40380000: iram_vma at ??:?

SHA-256 comparison failed:
Calculated: 09ccfa51459f0b85fa20c39d4065c9d917a10f7d3e536253558340a78768d789
Expected: 00000000404d0000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x403815c2
0x403815c2: __start at /Users/alex/zephyrproject/zephyr/soc/espressif/common/loader.c:227

I (60) boot: ESP Simple boot
I (60) boot: compile time Jun 28 2024 21:53:50
I (60) boot: Multicore bootloader
I (61) spi_flash: detected chip: generic
I (62) spi_flash: flash io: dio
W (65) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (77) boot: chip revision: v0.4
I (80) boot.esp32c3: SPI Speed      : 40MHz
I (84) boot.esp32c3: SPI Mode       : SLOW READ
I (88) boot.esp32c3: SPI Flash Size : 4MB
I (92) boot: Enabling RNG early entropy source...
[esp32c3] [INF] DRAM: lma 0x00000020 vma 0x3fc89820 len 0x1a6c   (6764)
[esp32c3] [INF] IRAM: lma 0x00001a94 vma 0x40380000 len 0x9810   (38928)
0x40380000: iram_vma at ??:?

[esp32c3] [INF] padd: lma 0x0000b2b8 vma 0x00000000 len 0x4d40   (19776)
[esp32c3] [INF] IMAP: lma 0x00010000 vma 0x42000000 len 0x32f0   (13040)
0x42000000: _OffsetAbsSyms at /Users/alex/zephyrproject/zephyr/arch/riscv/core/offsets/offsets.c:140

[esp32c3] [INF] padd: lma 0x000132f8 vma 0x00000000 len 0xcd00   (52480)
[esp32c3] [INF] DMAP: lma 0x00020000 vma 0x3c010000 len 0xff0    (4080)
[esp32c3] [INF] Image with 6 segments
[esp32c3] [INF] DROM segment: paddr=00020000h, vaddr=3c010000h, size=00FF0h (  4080) map
[esp32c3] [INF] IROM segment: paddr=00010000h, vaddr=42000000h, size=032F0h ( 13040) map

*** Booting Zephyr OS build v3.7.0-rc1-313-g895f246227e2 ***
Set up button at gpio@60004000 pin 1
Set up LED at gpio@60004000 pin 0
Press the button
```

sylvioalves, 2024-06-28:
I think I missed something in there. Last post shows it worked. What is the issue then?

alkurop, 2024-06-28:
It only worked after I modified, the overlay, but not out of the box. Basically, I added my comment for anyone else with the same problem, to know how to modify the overlay to get the monitor working

sylvioalves, 2024-06-28:
You mean you only got it working after deferring uart0 init by the below?
```c
&uart0 {
	zephyr,deferred-init;
};
```

alkurop, 2024-06-28:
Yes this. And also I have a lot of warnings. 
`0x40380000: iram_vma at ??:?`

`Calculated: 9b9ca907870209d00e0377014be2a2168815abc1a093d70dc8da2a708ed6a6f8
Expected: 0000000040620000000000000000000000000000000000000000000000000000
Attempting to boot anyway...`

And also rtc timer address warning in the DeviceTreeOverlay preview.

A bit of an off top, but maybe you know, what esle to modify in the overlay to fix them?

heepdog, 2024-07-12:
I had the same issue with a fresh install, and flashing to an esp32c3 supermicro.  
after adding `zephyr,deferred-init;` to the `&uart0` section of "esp32c3_devkitm.dts"