zephyrproject-rtos/zephyr #57703: Re-Evaluate the requirments tool

romkell, 2023-05-09:


gregshue, 2023-05-09:
Here's a start on user needs for an RM solution.

User Need:
As a Chip Vendor, 
I need to capture and trace requirements and design decisions to Zephyr from my commercially licensed HAL in a non-public repository, 
so that I can generate additional revenue through licensed value-added functionality.

User Need:
As a downstream user with access to relevant Zephyr release certification artifacts,
I need to capture and trace requirements in a Zephyr release to my bootloader requirements without modifying the Zephyr or MCUboot repositories,
so that I can re-submit a subset of the certification artifacts to reduce the time and cost of certifying the image on my device.

User Need:
As a downstream user following the T2 workspace topology with my manifest-module's build interface files located in the repository's `zephyr/` subdirectory,
I need the Zephyr-associated requirements to also be beneath the repository's `zephyr/` subdirectory,
so that I can avoid conflicts with pre-existing non-zephyr content located at the repository top level.

User Need:
As a downstream user following the T3 topology with my module assigned a fixed name in `zephyr/module.yml`,
I need the requirement UIDs to be independent of the pathname of my module repository,
so that I can continue to have the file organization freedom the Zephyr build system currently supports.

User Need:
As a downstream user of both Zephyr and a Zephyr fork (e.g., NCS),
I need an individual requirement UID to have consistent meaning across all forks,
so that I can avoid hard-to-detect errors in requirements tracing.

User Need:
As a manufacturer,
I want a requirements management solution from Zephyr Project that can scale up to cover my entire system (including cloud services) and the multiple levels of regulations my product must meet,
so that I do not have to split my requirements and verification of them across multiple tools.

User Need:
As a manufacturer,
I need to filter the requirements and links based on my product feature set (product configuration),
so that I can evaluate and track only those requirements that apply to my product variant.

User Need:
As a downstream module provider,
I need to capture my module requirements in a form and structure that aligns with the upstream modules and integrates with end user content consistent with upstream modules,
so that fewer barriers exist to others using my module.

tobiaskaestner, 2023-05-10:
The focus on downstream user needs resonates very much with me. Zephyr has no application on its own and will always be treated as some component in whatever safety-critical device you may envision. Treating your product development as a downstream activity also makes a ton of sense, in fact the entire tooling of zephyr is pushing you into this direction. And I mean this in the most positive sense possible. Zephyr is more than a RTOS kernel under the kernel folder. It also provides many integration points (for silicon vendors, chip manufacturers, middleware providers etc.) It is opionated to the greatest extent possible (again in the most positive sense) and if it starts to capture requirements or (much more interesting for me) design specifications it should be done in a way that downstream projects could build upon the tooling by using extension points than offered as we can extend CMake/KConfig/DeviceTree/West modules today. 
Nordic provides a nice example with how they build the docs for their connectSDK  on how to even extend things not yet meant to be extended.

gregshue, 2023-05-10:
>  if it starts to capture ... design specifications

When, not if. :-) As downstream users integrate multiple subsystems into a device they must not violate the design assumptions of any subsystem. The cleanest way to communicate this to product integrators and subsystem maintainers is through design specifications.

gregshue, 2023-05-11:
User Need:
As a secure IoT device manufacturer,
I need to clearly identify which requirements apply to each of my immutable images,
so that I can identify security risks that may require device HW replacement to mitigate and follow the appropriate SDLC.

User Need:
As a secure IoT device manufacturer,
I need to clearly identify which requirements apply to each of my mutable images,
so I can identify security risks that may estimate the risk exposure on units that are not allowed to be automatically updated.

nashif, 2023-05-11:
> Here's a start on user needs for an RM solution.
>
> User Need: As a Chip Vendor, I need to capture and trace requirements and design decisions to Zephyr from my commercially licensed HAL in a [...]

I am lost here and not sure why any of this applies to selecting a requirement management tool for Zephyr. As a chip vendor, manufacturer, downstream user you can use whatever tool you like and everyone will have different requirements for tooling. None of that is relevant to what we are trying to do here. We are looking for a tool to manage our own requirements for the zephyr project, full stop. Nothing more. This is not about providing tools for downstream users to manage their own requirements. Maybe the only relevant item we need to make sure a tool provides, hence a requirement on the tool, is to be able to export zephyr requirements  in some standard format that can be imported into other tools, thats it. Everything else should be about how such tool integrates with zephyr workflows and how Zephyr's own requirements are managed and maintained.

gregshue, 2023-05-11:
>  how Zephyr's own requirements are managed and maintained.

Zephyr's own requirements will include constraints on workflows, SDLC, and the standards it has committed to meeting (e.g., IEC 62443). For Zephyr to be desirable, many of the requirements on Zephyr should be based on realistic user needs and usage models. (See @tobiaskaestner comment above.) Zephyr is not very useful in isolation. 

> As a chip vendor, manufacturer, downstream user you can use whatever tool you like

Agreed. Many will be very interested in using Zephyr's OSS solution for managing their own requirements.  Ignoring that now is inconsiderate of Zephyr's target userbase and contrary to the ZP vision statement.

> Maybe the only relevant item we need to make sure a tool provides, hence a requirement on the tool, is to be able to export zephyr requirements in some standard format that can be imported into other tools,

If that were the only requirement on Zephyr's tool then it wouldn't be useful for producing safety- and security- certification inputs. 

If you prefer, we can use `example-application or mcuboot module maintainer` for downstream end user, and `hal_<whomever> maintainer` for chip vendor (since they may distribute HAL fixes before they get merged into zephyrproject-rtos mainlines). The needs are still the same.

nashif, 2023-05-11:
> Zephyr's own requirements will include constraints on workflows, SDLC, and the standards it has committed to meeting (e.g., IEC 62443). For Zephyr to be desirable, many of the requirements on Zephyr [...]

This is not about doing things isolation. You are mixing requirements with a requirement management tool. this issue is about selecting a tool, not about setting requirements. Requirements can be managed in different ways, a text file, a csv file, a database, GH issues, Jira, some commercial tool, etc etc and any of those is acceptable for whatever standard you target. 

Right now, we are trying to identify a tool that is to be used to manage Zephyr's own requirements. This is not about finding a tool that fits needs the rest of the world. This is what the issue about, nothing more. If you want to contribute to this, start looking at possible options and how they fit the needs of the project, not your needs as a downstream user.

> > As a chip vendor, manufacturer, downstream user you can use whatever tool you like
> 
> Agreed. Many will be very interested in using Zephyr's OSS solution for managing their own requirements. Ignoring that now is inconsiderate of Zephyr's target userbase and contrary to the ZP vision statement.

There is nothing here that is contrary to any vision. We are looking for a tool, we have options, we will pick one, just like we did in the past, as a project. Thats why we are having this discussion, some think whatever was selected is not good enough, we are reconsidering that. This is very considerate of the userbase.

> If that were the only requirement on Zephyr's tool then it wouldn't be useful for producing safety- and security- certification inputs.

Fully disagree. Standards do not have any requirements on the tools. That requirement of being able to import/export is to satisify  interchangability of information in a standard format. We do not need more than that. 
And, I noted this being the only relevant item in the context of reuse and end-users. I did not say this is the only requirements on a tool.

> 
> If you prefer, we can use `example-application or mcuboot module maintainer` for downstream end user, and `hal_<whomever> maintainer` for chip vendor (since they may distribute HAL fixes before they get merged into zephyrproject-rtos mainlines). The needs are still the same.

No, I do not prefer that. I do not think any of that has any relevance here.

romkell, 2023-05-11:
@gregshue What Zephyr OS safety concerns I look at it as if I buy an safe RTOS, e.g. Wittensteins SafeRTOS or MS Azure RTOS ThreadX, ask for / buy from my favorite MCU vendor a safety certified HAL and start building a Safety application. If I need a bootloader, the I will have to develop that similar to my application either I use the same certified RTOS or I do that bare-metal with the certified HAL. And I will have to do the same puzzle with the requirements management.
What you are asking for or would like to build is, as I understand it, a requirements management framework around or to the existing ecosystem, that allows requirements management and traceability to the full extent, from HAL (module), to RTOS (zephyr repo), middleware (other modules) to user application(s).
It would be cool to get that - one day.
At the moment we are struggling to get the core tasks done and I therefore feel we need to focus on the basics. Get the Zephyr OS safety scope described with requirements and achieve traceability there. If you need traceability to those requirements from "outside" (I would not even know If you get the full requirements set from Wittenstein for SafeRTOS, and if so, the as reqif or pdf), you will need to link from your tool and your higher level requirements to those of Zephyr OS or, as Anas mentioned, you need requirements export/import facilities and formats to transfer at least the layer of your "greybox" you want to link to.
For the moment, I feel, you are asking for too much. I doubt that we can manage that right now.

gregshue, 2023-05-11:
So I will need some guidance on what kinds of things are in/out of the safety scope:

The boundaries may need to contain code inside/outside trusted execution environments, which may be one of the decomposition layers.  Does it need to create an independent executable with its own certification criteria?

Could any module content be pulled into the safety-critical code (including any headers with macros or inline implementations)? Carefully consider HALs, tfm, picolibc, crypto content.  Any of these will bring up the question of cross-repo linking. Is this currently an issue?  If not and we choose to ignore it, how will we prevent it from becoming an issue?

Requirements for secure code will need to use the same tool and probably exist in the same dataset. How does the safety-critical code boundary align with the boundary needed for secure code? Do we need to identify a requirement as being relevant for safety-critical? secure? secure-trusted-environment? secure-immutable?

Are we looking for a tool that is ready to handle the needs as the safety- and security- scopes grow over time?

nashif, 2023-05-12:
> So I will need some guidance on what kinds of things are in/out of the safety scope:
>
> The boundaries may need to contain code inside/outside trusted execution environments, which may be one of the [...]

once again, I am not sure why all of this is relevant to selecting a requirement management tool.  Would be progress if we could see list of available tools, how they differ, what they offer, maybe in form of a table so we can make a decision and STOP referring to modules and HALS and security.

gregshue, 2023-05-12:
What safety claims can you make that don't depend on a secured implementation?

romkell, 2023-07-05:
## Sphinx-needs specification sandbox for Zepyhr

https://github.com/romkell/zephyr/blob/spec_with_sphinx_needs

https://github.com/romkell/zephyr/blob/spec_with_sphinx_needs/specs

https://github.com/romkell/zephyr/blob/spec_with_sphinx_needs/specs/README.md

I couldn't get the vscode plugin to work. 

@danwos would you mind having a look at the sandbox and at the [Requirement Tool requirements](https://docs.google.com/spreadsheets/d/12DBzOCXW3vUJc1cfnA5AIK_JIdCKqUcIkFuO3jeFIqM/edit?pli=1#gid=0) me not stating anything wrong concerning sphinx-needs? I would also appreciate some support on the vscode plugin. Does not show anything in the window so far.

danwos, 2023-07-05:
@romkell: I wrote some comments in the Sphinx-Needs column.
I hope this helps a little bit and I wrote 1-2 comments about in what kind of projects Sphinx-Needs is already used.
Currently, I have really zero time to take a deep look into the sandbox (projects needs to be finished, vacation time starts, ..).  Sorry, will try to come back to it in 3-4 weeks.

From the requirement management perspective, the written requirements are quite good.
Here are some additional points (but I may have missed them):

* Analysis: Show filtered results in tables, flows, piecharts and Co.
* Automation: Calculate values. For instance, the status of a sys-req is based on the status of linked comp-reqs.
* Templating: Add req-specific analysis under each req. E.g. show the dependency tree
* Layout: Customize what data to show in the HTML representation
* Styles: Set colors based on status or other data

You may also think about how separated the requirement definition shall be from the other documentation.
Does it make sense to have normal rst/md-text between reqs, to maybe explain something which shall not be part of an official req?
Or can reqs be part of a SW-Architecture page?
Can they be linked/combined with other Documentation features, like Test-Results or imported Data from systems like JIRA, github Issues, ...
So the core of this question is: Are there benefits for you that Sphinx-Needs extends Sphinx and feels therefore like just writing Sphinx docs, no extra files or format?

romkell, 2023-07-10:
@stanislaw Since I directed Daniel Woste, the main developer of sphinx-needs to the Projects [Requirement Tool requirements](https://docs.google.com/spreadsheets/d/12DBzOCXW3vUJc1cfnA5AIK_JIdCKqUcIkFuO3jeFIqM/edit?pli=1#gid=0) I would like to give you the same opportunity to comment on the requirement fulfillment of StrictDoc (entries done by Roberto Bagnara).
I feel it would be helpful if the main developer can comment on it.

romkell, 2023-08-15:
- Decision taken in Safety Committee in favor of StrictDoc (see [Requirement Tool requirements](https://docs.google.com/spreadsheets/d/12DBzOCXW3vUJc1cfnA5AIK_JIdCKqUcIkFuO3jeFIqM/edit?pli=1#gid=0)
- Decision communicated to Safety WG, no objections received form WG
- Safety Committee decision to be taken to TSC for approval

closed 2023-08-15 by romkell