zephyrproject-rtos/zephyr #57715: With Add pytest plugin #57117 and Add renode-test integration #56770 we have the start of a common API

DRuffer, 2023-05-09:
**Both of these PRs describe similar solution to a common problem. Can we call different Twister runners?**
[Add pytest plugin #57117](https://github.com/zephyrproject-rtos/zephyr/pull/57117)
[Add renode-test integration #56770](https://github.com/zephyrproject-rtos/zephyr/pull/56770)
It is my understanding, they they both use pyTest, in different manors, and would benefit from a common front end that already chooses tests plans, and a common back end that using report formats that can be scrapped to be integrated into regression analysis report. I suspect that those parts might remain the same and the runner in the middle can be extended in multiple ways 

**I'd like us to start looking at the common parts and lets those be tools so the end parts look the same**
They can be interegratre as needed in the new systems. Initially, there may be little overlap, but the integrators and consider the possiblities for a later integration stage.

**Alternatives are already in play in both solutions**
The goal is NOT to slow down merging those solutions in ASAP, but the establish another phase, where this integration can become universal.

hakehuang, 2023-05-10:
for top-down view, do we have a use case diagram on what we expect the control flow should be? As now we have two approaches, and the acrually all act as an twister plugin way, and use twister as a shell and takeover the control by calling its internal calls.  and to unify the apis, we need:
1. do we need unify the API calls?
2.  does each plugin in need shall the same data such as the functionality can easily shared/overried by other?
3. do we expect the same reporting result of each plugin? I think we do.

PerMac, 2023-05-10:
Just a small clarification. Both solutions have python on it's base. However one is based on robot framework and the other on pytest.

1.) I don't think the exact same api is possible since different frameworks have their own interfaces. However, having a unified integration of multiple framework with twister can be beneficial. Already, both PRs are based on `harness` functionality in twister. 
2.) Indeed, both PRs show how to use these frameworks for shell testing but they use different approaches and test's source code. I don't think one test code can be reused by another framework (but the app under test can).
3.) Those are different frameworks and one test cannot be rerun with other framework, hence having "the same" result might not be desired. Although, the reporting should be unified in a way that in the end all results are seen in the final twister report and maintain common format.

I will be showing some simplified flow charts on how the integration with twister is done during our plugin presentation. Maybe it can serve as a starting point for what you're asking Dennis.

DRuffer, 2023-05-11:
> I second what @PerMac <https://github.com/PerMac> says - these are
> conceptually similar, but different solutions addressing different needs
> and based on different technical details.
>
> Regarding result reporting, it's difficult to generate identical results,
> because we test different things.
> What we can do is to produce test results in the xUnit format.
>
> Of course Twister also gathers these results as usual.

I was not suggesting "identical results", but rather a common
reporting framework. From what I've seen, it's hard for even the current
Twister to get "identical" results on some intermittent tests, but both
results are reported in a common format/API. That is the desire, common
reportability tools, not repeatability, at least not yet. ;)

nashif, 2024-02-08:
common results reported in json format by twister is exactly what we have now.

closed 2024-02-08 by nashif