zephyrproject-rtos/zephyr #57720: Dissaccociate struct device from devicetree nodes for application layer

bjarki-trackunit, 2023-05-09:
## Introduction

Using sections in elf files, it is possible to find all struct device defined using DEVICE_DEFINE, and what API type the device implements (its class). From this, we can create macros to find and iterate through struct devices by class agnostic from the devicetree.

We will be able to do this from the application

```
const struct device *rtc = DEVICE_CLASS_GET(rtc, 0);`
const struct device *counter0 DEVICE_CLASS_GET(counter, 0);`
const struct device *counter1 = DEVICE_CLASS_GET(counter, 1);`
```

To get pointers to struct devices, without caring where the device came from. This dissociates struct devices from the devicetree nodes, allowing us to create multiple struct devices for each devicetree node, of the same or different class.

Its essentially Linux SYSFS but static and using macros :)

### Problem description

We are currently limited by having struct devices associated with a devicetree node, which results in issues when no driver exists to create a struct device for said devicetree node, and has no guarantee of which API the potential struct device implements, only a vague naming contract like counter0, counter1 etc, which can lead to undefined behavior, and makes it impossible to create multiple struct devices for each devicetree node, without having to iterate through devices by name, which is slow and only possible at runtime.

### Proposed change

Place struct device APIs like rtc_driver_api in sections by type, to be recognized by an elf parser. By parsing the elf file, we can determine which API each struct device points to, and from the section name, we can determine the class of the device.

We then split up the build into a kernel build, which builds zephyr only against an empty main.c file, use the elf parser to identify all struct devices, generate macros and externs for them, then build the application, which can now include the header with the new macros as shown in the introduction.

### Dependencies

Applications will now be encouraged to use the new class based macros to identify and use devices, instead of devicetree nodes. Devicetree nodes will still work as before, but may become less useful as nodes become dissassociated from the struct devices generated from them by the drivers.

### Concerns and Unresolved Questions

* Is it even possible? 
* How much does it break? 
* Is a device class based approach (like Linux sysfs) something we want on an embedded platform given its abstraction from the devicetree?

## Alternatives

Keep the strict 1-1 binding between devicetree nodes and struct devices.

bjarki-trackunit, 2023-05-09:
I have done a bit of testing on this draft PR https://github.com/zephyrproject-rtos/zephyr/pull/57622

bjarki-trackunit, 2023-05-11:
@gmarull What do you think of this idea?

carlescufi, 2023-05-16:
Architecture WG:

- @henrikbrixandersen states that this has been brought up before. If the only purpose is to be able to error out if the struct device that is passed around does not conform to a specific API, then this is overkill. @bjarki-trackunit answers that this is not the only purpose. Having sections allows you to list instances of a specific device class. An elf parser suddenly can figure out what class of device a particular `struct device` is. This is similar to `sysfs` in Linux.

bjarki-trackunit, 2023-05-22:
Here is a proper attempt at adding the device enumeration feature https://github.com/zephyrproject-rtos/zephyr/pull/58112