zephyrproject-rtos/zephyr #57722: Possible bug with samples/net/sockets/echo_server overlay-tunnel.conf

glarsennordic, 2023-05-10:
**Describe the bug**
The echo_server does not get past boot when `overlay-tunnel.conf` is enabled. Since there are no instructions on how to test with the tunnel, it isn't 100% clear whether this is due to misuse or a bug in the overlay.

This is using the qemu_x86 platform on Ubuntu Linux.

**To Reproduce**
I derived the following test procedure by combining instructions from the following two pages:
- https://docs.zephyrproject.org/3.2.0/samples/net/capture/README.html (Contains a very similar tunneling feature that I was told [here](https://github.com/zephyrproject-rtos/zephyr/pull/56549#issuecomment-1524882810) should be roughly equivalent to the `overlay-tunnel.conf` overlay.)
- https://docs.zephyrproject.org/2.7.0/guides/networking/qemu_setup.html (standard instructions for the echo_server sample without overlays on qemu_x86)

Terminal 1, in net-tools: `sudo ./net-setup.sh -c zeth-tunnel.conf`
Terminal 2, in net-tools: `./loop-socat.sh`
Terminal 3, in net-tools: `sudo ./loop-slip-tap.sh`
Terminal 4, in echo-server sample: `west build -p always -b qemu_x86 samples/net/sockets/echo_server -- -DCONF_FILE=overlay-tunnel.conf` followed by `west build -t run`

**Expected behavior**
Sample should make it past boot and I should be presented with a shell terminal.

**Observed behavior**
Stalls on boot:
```
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..
*** Booting Zephyr OS build zephyr-v3.2.0-7691-ga01d40f6049a ***
```

rlubos, 2023-05-25:
@glarsennordic I've tried to reproduce the issue, and failed. I've tried `qemu_x86` and `native_posix`, same Zephyr revision as in the log you provide, and observe no issues with boot at all.

The virtual interface seems to be working correctly. The interface starts in admin DOWN state, but after bringing it up, it seems to be listed correctly:
```
uart:~$ net iface up 1
Interface 1 is up
uart:~$ net iface 

Interface 0x149ea4 (Virtual) [1]
==================================
Name      : IPv4 tunnel
Attached  : 2 (Ethernet / 0x149f70)
Link addr : 91:8E:EF:EC:F2:BF
MTU       : 1480
Flags     : POINTOPOINT,NO_AUTO_START,IPv4,IPv6
IPv6 unicast addresses (max 3):
	2001:db8:200::1 manual preferred infinite
IPv6 multicast addresses (max 4):
	<none>
IPv6 prefixes (max 2):
	<none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 44408
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 1):
	<none>
IPv4 multicast addresses (max 1):
	<none>
```
And once the interface is up, I can ping the tunnel interface from my Linux host (after running `sudo ./net-setup.sh -c zeth-tunnel.conf` - with `native_posix` this works out of the box, `qemu_x86` requires slightly different configuration, but also works). 

I do agree though, the documentation on the tunneling topic could be better though.

Anyways, since the original concern was about not breaking the tunneling with the new `conn_mgr` extension, and it doesn't seems to be the case (`main` works fine for me as well), maybe we should consider closing the issue? I don't see anything to fix here.

glarsennordic, 2023-05-25:
Strange -- I'd be curious what your exact build command was, but I agree, this is fine to close since the primary concern was making sure tunneling isn't broken after conn_mgr. I'm not sure why it fails to get past boot on my system

closed 2023-05-25 by glarsennordic