zephyrproject-rtos/zephyr #57735: USB mass storage data corruption when using flashdisk and jedec, spi-nor flash device with 4096 sector size

justinleavesley, 2023-05-10:
I am using CONFIG_SPI_NOR_SFDP_RUNTIME=y to configure the spi nor flash device which has a sector size of 4096.
I am using this In conjunction with flash disk and usb mass storage to enable support for windows usb mass storage device.

The flash disk  ("zephyr,flash-disk") configuration supports sector-size property which if set to some value other than 512 causes zephyr\subsys\usb\device\class\msc.c to throw a runtime error "Block Size reported by the storage side is different from Mass Storage Class page Buffer - Aborting". This is due to msc.c testing the returned sector size from the flash disk with its hardcoded value of 512.

If using a spi nor flash device with a sector size of 4096 then leaving the sector size as the default 512 allows the system to run but causes data corruption when used with windows 11 to format or read/write files to it (not tested on anything else). 

Changing **#define BLOCK_SIZE	512** to **#define BLOCK_SIZE	4096** in the msc.c file
 and adding the property **sector-size = < 4096 >;** to my overlay for the flash disk fixed the problem.
 
 Obviously having to edit the Zephyr code is not an ideal situation. Maybe msc.c should accept the value it gets from the flash disk device for sector size and not impose its hardcoded value.

OS: windows
Toolchain: NRF SDK 2.3.0

tmon-nordic, 2023-05-11:
> Obviously having to edit the Zephyr code is not an ideal situation. Maybe msc.c should accept the value it gets from the flash disk device for sector size and not impose its hardcoded value.

This is how the new MSC class implementation in next USB device stack works. All you have to do is just to make sure `CONFIG_USBD_MSC_SCSI_BUFFER_SIZE` is at least as big as the sector-size (default is 512, but can be set to any value).

Note that while new MSC class implementation has no restriction on the sector-size, major OSes seem to support only the most common values (512 and 4096 are supported on pretty much every OS).

justinleavesley, 2023-05-11:
If I understand you correctly, then the next nrf connect sdk release should no longer have this issue as it should contain this new MSC implementation.

That would be great!

tmon-nordic, 2023-05-11:
> If I understand you correctly, then the next nrf connect sdk release should no longer have this issue as it should contain this new MSC implementation.

Not really. It only covers the workaround you use with 4096, i.e. the workaround will work with the next USB device stack without having to modify Zephyr code (i.e. can be done solely in application).

However, I can reliably reproduce Windows 11 "Windows was unable to complete the format." error with 8 MiB flashdisk using 512 emulated sectors on 4 KiB SPI NOR sectors on both the old and next device stack. The error happens for every "Allocation unit size" except for 4096.

The problem is that data written does not match data read back, which confirms your data corruption observation.

tmon-nordic, 2023-05-11:
I can confirm the issue on nRF52840 DK if I power the board solely from nRF USB. If I connect both nRF USB and JLinkOB USB, then everything works as expected, which suggests the issue might not be in the code itself.

Is your device bus-powered or self-powered?

justinleavesley, 2023-05-11:
I see. At least I will be able to make it work without zephyr code changes. That is the most important to me.

To confirm, I see the problem on the nRF5340DK without or without powering via bus or self powered. And the problem vanishes if 4096 is used without changes power approach.

tmon-nordic, 2023-05-11:
I cannot reproduce this on nRF5340DK, neither with old nor next device stack. On nRF5340DK everything works as expected if I have only nRF USB connected.

How do you determine that there is a problem? Can you share your application and/or do you see the issue with any sample?

justinleavesley, 2023-05-11:
I noted your issue with different power scenarios and I just changed the nrf5340dk to run in high power mode (3.3v).

Now it is working with either nRF USB alone or with jlink power. Previously it was running at 3v (still well within my spi-nor voltage range).

Both default 512 sector size or my hardcoded 4096 it is now working.

Also worth noting that I had to give a larger stack size to the msc.c thread with CONFIG_MASS_STORAGE_STACK_SIZE otherwise I was getting stack overflow. (That might be because I am running a slow SPI bus to the nor flash).

Since this is working now for the default sector size I guess there isn't a bug. Although I am glad that will be able to set the sector size from the application in the next SDK release.

tmon-nordic, 2023-05-12:
I measured MX25R6435F Vcc voltage at SB16 on nRF52840DK (PCA10056) during the failing format operation (powered only via nRF USB). The voltage repeatedly does drop below the minimum 1.65V for approximately 100us. There is also some ringing visible. This pretty much convinces me that the issue is not in the software.

[image]

jfischer-no, 2023-05-12:
>Since this is working now for the default sector size I guess there isn't a bug.

Let's turn this into a discussion. Perhaps some clarification is needed in the board documentation.