zephyrproject-rtos/zephyr #57746: Changing UART1 baud rate on runtime for ESP32C3 does not work

abdulayubzoi, 2023-05-10:
**Describe the bug**
I'm working on changing UART1 baud rate on runtime, on ESP32C3. I have tried using `uart_configure` and `uart_line_ctrl_set` but both of them don't seem to work, i.e. they execute without actually changing the baud rate and don't return a negative error code. The communication simply stops. Moreover, the `uart_config_get` reports the same baud rate before and after changing it.

**To Reproduce**
Take any sample project with UART1 and build it for ESP32C3. Use the relevant `prj.conf`, `.dts` and code snippet in the last section **Additional context** to be able to reproduce the issue.

**Expected behavior**
Changing baud rate works fine, or at least reports an error.

**Impact**
Being unable to change the baud rate dynamically is a showstopper for my use-case.

**Environment (please complete the following information):**
- Devcontainer based on image: zephyrprojectrtos/zephyr-build:v0.25.0
- Toolchain Zephyr SDK 0.16.0
- Zephyr branch: v3.3.0
- Target: esp32c3 (DevkitM-1)

**Additional context**
Use the following (relevant) `prj.conf`:
```
CONFIG_SERIAL=y
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_USE_RUNTIME_CONFIGURE=y
CONFIG_UART_LINE_CTRL=y
CONFIG_UART_DRV_CMD=y
```

Use the following (relevant) `.dts` section:
```
&uart1 {
	status = "okay";
	current-speed = <115200>;
	pinctrl-0 = <&uart1_tx_gpio19 &uart1_rx_gpio18 &uart1_cts_gpio2 &uart1_rts_gpio3>;
	pinctrl-names = "default";
};
```

Use the following code to attempt to change the baud rate on runtime:
```
int configure_device_uart_baud_rate(uint32_t device_uart_baud_rate)
{
	int error = 0;

	const struct device *uart_device = DEVICE_DT_GET(DT_ALIAS(uart1));
	if (uart_device == NULL) {
		LOG_ERR("UART Device not found");
		return -ENODEV;
	}

	error = device_is_ready(uart_device);
	if (error != 1) {
		LOG_ERR("UART Device not ready (error %d)", error);
		return error;
	}

	struct uart_config uart_config;
	error = uart_config_get(uart_device, &uart_config);
	if (error) {
		LOG_ERR("uart_config_get failed (error %d)", error);
		return error;
	}
	LOG_DBG("UART current baud rate = %lu", (unsigned long)uart_config.baudrate);

	uart_config.baudrate = device_uart_baud_rate;
	LOG_DBG("UART setting baud rate = %lu", (unsigned long)uart_config.baudrate);

	error = uart_configure(uart_device, &uart_config);
	if (error) {
		LOG_ERR("uart_line_ctrl_set failed (error %d)", error);
		return error;
	}

	/* This doesn't work either */
	error = uart_line_ctrl_set(uart_device, UART_LINE_CTRL_BAUD_RATE, device_uart_baud_rate);
	if (error) {
	 	LOG_ERR("uart_line_ctrl_set failed (error %d)", error);
	 	return error;
	}

	error = uart_config_get(uart_device, &uart_config);
	if (error) {
		LOG_ERR("uart_config_get failed (error %d)", error);
		return error;
	}
	LOG_DBG("UART updated baud rate = %lu", (unsigned long)uart_config.baudrate);

	return error;
}
```

**Note:**
A discussion for the same issue has already been opened before, for reference: https://github.com/zephyrproject-rtos/zephyr/discussions/57029

sylvioalves, 2023-05-11:
Hi, indeed there was an error on the return value only.  
The code below was updating the baudrate properly:  
```c
	error = uart_configure(uart_device, &uart_config);
	if (error) {
		LOG_ERR("uart_line_ctrl_set failed (error %d)", error);
		return error;
	}
```
However, the return value wasn't being update, causing error at:  
```c
	error = uart_config_get(uart_device, &uart_config);
	if (error) {
		LOG_ERR("uart_config_get failed (error %d)", error);
		return error;
	}
```
ESP32 LINE_CTRL is not supported, so the below will fail with -88 I think:  
```c
	/* This doesn't work either */
	error = uart_line_ctrl_set(uart_device, UART_LINE_CTRL_BAUD_RATE, device_uart_baud_rate);
	if (error) {
	 	LOG_ERR("uart_line_ctrl_set failed (error %d)", error);
	 	return error;
	}
```

abdulayubzoi, 2023-05-11:
Thank you for the fast response! I had to switch back to v3.1.0 but will test again soon with v3.3.0 and will let you know.

PR #57761 (2023-05-12): drivers: uart: esp32: fix baudrate return value
Baudrate value was not updated properly when requested.

Fixes #57746
commit 01da4df1 (2023-05-12): drivers: uart: esp32: fix baudrate return value

Baudrate value was not updated properly when requested.

Fixes #57746
closed 2023-05-11 by abdulayubzoi