zephyrproject-rtos/zephyr #57763: Cannot enable GPROF with C++

yperess, 2023-05-11:
**Describe the bug**
(Possibly due to `main` renaming) it's currently not possible to use `CONFIG_GPROF` with C++. I've tried a few things to understand how the renaming works so I may resolve the issue, but with little to no success.

Please also mention any information which could help others to understand
the problem you're facing:
 - Platform: `native_posix`
 - I've tried building `samples/philosophers` with various additional configs
 - This is not a regression, just something that was never tested

**To Reproduce**
Steps to reproduce the behavior:
1. Add the following line to `samples/philosophers/CMakeLists.txt`:
```
set_source_files_properties(src/main.c PROPERTIES LANGUAGE CXX)
```
2. `$ west -v build -p -b native_posix zephyr/samples/philosophers/ -- -DCONFIG_GPROF=y -DCONFIG_CPP=y -DCONFIG_STD_CPP17=y -DCONFIG_CPP_MAIN=y`
3. `$ ./build/zephyr/zephyr.elf` let it run for a while then ctrl+c
4. `$ find . -name "gmon.out"`

**Expected behavior**
The run should generate a gmon.out like it does when the program is compiled as a C program.

**Impact**
This prevents us from profiling several key components for sensors common logic which we plan to use in a CI which will verify against performance regressions.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: HEAD

yperess, 2023-05-11:
@stephanosio I hope you don't mind, I assigned this to you as you're the most knowledgeable person I know in this area. Please feel free to re-assign as needed. I'm at a loss of what to do with this.

stephanosio, 2023-05-11:
> @stephanosio I hope you don't mind, I assigned this to you as you're the most knowledgeable person I know in this area. Please feel free to re-assign as needed. I'm at a loss of what to do with this.

That is fine. I will try to see if I can figure out what is going on when time permits (probably not going to be anytime soon though).

If anyone else wants to look into this, please feel free to assign themselves.

stephanosio, 2023-05-11:
cc @nashif

yperess, 2023-05-11:
A quick update, I started looking at workarounds instead of proper solutions and I came across one. If my main function includes the following, everything works:

```c
int main() {
  // My application code here...

  extern void _mcleanup(void);
  _mcleanup();
  return 0;
}
```

I then moved the logic into `native_posix/main.c::posix_exit()` in the form of:
```c
#ifdef CONFIG_GPROF
	extern void _mcleanup(void);
	_mcleanup();
#endif
```
What are your thoughts about this being the fix?

PR #57841 (2023-05-12): lib: cpp: Do not define __cxa_atexit() for the posix arch
For the POSIX architecture we do not define an empty `__cxa_atexit()` as it otherwise would
override its host libC counterpart. And this would both disable the atexit() hooks, and prevent possible test code global destructors from being registered.

---------
Fixes #57763
Supersedes #57801 
See https://github.com/zephyrproject-rtos/zephyr/pull/57801#issuecomment-1545444148 for more background.
closed 2023-05-12 by nashif