zephyrproject-rtos/zephyr #57778: usb_dc_mcux: Failed to allocate memory

jeppenodgaard, 2023-05-11:
**Describe the bug**
Calling usb_enable(NULL) which is done by multiple samples can fail because usb_dc_mcux.c:289 fails to allocate memory.
This has been seen on an mimxrt1050_evk.

**To Reproduce**
Steps to reproduce the behavior:
1. 
```
west build -b mimxrt1050_evk -d build_bl bootloader/mcuboot/boot/zephyr -- -DCONFIG_IMG_ERASE_PROGRESSIVELY=y && west build -b mimxrt1050_evk -d build zephyr/samples/subsys/usb/dfu -- -DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE=\"bootloader/mcuboot/root-rsa-2048.pem\" -DCONFIG_IMG_ERASE_PROGRESSIVELY=y && west flash -d build_bl && west flash
```
2. Reset the device until it fails

Step 2 can be automated so it reboots fast and without using the board's switch.
Create a test.sh script:
```bash
#!/bin/bash

LOG_FILE=/tmp/tio.log

JLINK_SCRIPT=$(mktemp)
cat > $JLINK_SCRIPT <<- EOL
        reset
        exit
EOL

reset()
{
        JLinkExe -nogui 1 -if swd -speed auto -device MCIMXRT1052 \
        -CommanderScript $JLINK_SCRIPT > /dev/null
}

await_boot()
{
        while ! grep "USB DFU class." $LOG_FILE; do
                sleep .1
        done
        echo > $LOG_FILE
}

(
        RUNS=1000
        for i in $(seq $RUNS); do
                echo "Run $i of $RUNS"
                echo > $LOG_FILE
                reset
                await_boot
        done
)
```

In one terminal: `tio -m INLCRNL /dev/ttyACM0 --log --log-file /tmp/tio.log`
In another terminal: `bash test.sh`

It might take more than 100 reset before the error occurs.

**Expected behavior**
usb_enable() does not fail.

**Impact**
Sample sometimes does not work and applications that uses usb in general might fail.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.3.0-3745-ga638223df37b ***
*** Booting Zephyr OS build zephyr-v3.3.0-3745-ga638223df37b ***
[00:00:00.000,000] <inf> main: This device supports USB DFU class.

*** Booting Zephyr OS build zephyr-v3.3.0-3745-ga638223df37b ***
*** Booting Zephyr OS build zephyr-v3.3.0-3745-ga638223df37b ***
[00:00:00.000,000] <inf> main: This device supports USB DFU class.

*** Booting Zephyr OS build zephyr-v3.3.0-3745-ga638223df37b ***
*** Booting Zephyr OS build zephyr-v3.3.0-3745-ga638223df37b ***
[00:00:00.000,000] <err> usb_dc_mcux: Failed to allocate memory
[00:00:00.000,000] <err> main: Failed to enable USB
[00:00:00.527,000] <err> usb_dc_mcux: No cb pointer for endpoint 0x00
[00:00:05.743,000] <err> usb_dc_mcux: No cb pointer for endpoint 0x00
[00:00:10.911,000] <err> usb_dc_mcux: No cb pointer for endpoint 0x00
[00:00:16.031,000] <err> usb_dc_mcux: No cb pointer for endpoint 0x00
[00:00:21.471,000] <err> usb_dc_mcux: No cb pointer for endpoint 0x00
[00:00:26.735,000] <err> usb_dc_mcux: No cb pointer for endpoint 0x00
[00:00:31.899,000] <err> usb_dc_mcux: No cb pointer for endpoint 0x00
[00:00:37.019,000] <err> usb_dc_mcux: No cb pointer for endpoint 0x00
[00:00:43.011,000] <err> usb_dc_mcux: No cb pointer for endpoint 0x00
[00:00:48.235,000] <err> usb_dc_mcux: No cb pointer for endpoint 0x00
[00:00:53.719,000] <err> usb_dc_mcux: No cb pointer for endpoint 0x00

```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain  Zephyr SDK 0.16.0
 - Commit SHA: a638223df37b4e2e0789dee0b2600a0ee0a90a37

jeppenodgaard, 2023-05-11:
I think the issue was introduced in 90570c6b9dcffb3f82f0e23e45624a5d2c6e11d2

mmahadevan108, 2023-05-25:
@jeppenodgaard I am unable to reproduce the failure. I am using a MIMXRT1050 EVKB board. Below is the log after 1000 RUNS

```
*** Booting Zephyr OS build zephyr-v3.3.0-4350-g8959b838bf15 ***
*** Booting Zephyr OS build zephyr-v3.3.0-4350-g8959b838bf15 ***
[00:00:00.000,000] <inf> main: This device supports USB DFU class.

*** Booting Zephyr OS build zephyr-v3.3.0-4350-g8959b838bf15 ***
*** Booting Zephyr OS build zephyr-v3.3.0-4350-g8959b838bf15 ***
[00:00:00.000,000] <inf> main: This device supports USB DFU class.
```

Log on the second terminal:
```
Run 996 of 1000
Binary file /tmp/tio.log matches
Run 997 of 1000
Binary file /tmp/tio.log matches
Run 998 of 1000
Binary file /tmp/tio.log matches
Run 999 of 1000
Binary file /tmp/tio.log matches
Run 1000 of 1000
Binary file /tmp/tio.log matches
```

jeppenodgaard, 2023-05-25:
Thanks for testing this so thoroughly @mmahadevan108.

We are using different commits. The hal_nxp has been updated since I tested. `drivers/usb/device/usb_dc_mcux.c` seems to be identical.

I'll close this since it does not seem to be reproducible.

closed 2023-05-25 by jeppenodgaard