zephyrproject-rtos/zephyr #57817: Littlefs sample failing on mimxrt1050_evk

bdunlay, 2023-05-11:
**Describe the bug**
Littlefs sample does not work on mimxrt1050_evk. 

 - What target platform are you using? mimxrt1050_evk
 - What have you tried to diagnose or workaround this issue? I haven't attempted to debug the issue.
 - Is this a regression? Not that I am aware of.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -b mimxrt1050_evk -d nxp-fs-sample samples/subsys/fs/littlefs/`
2. `west flash -d nxp-fs-sample`
3. Observe serial output

```
*** Booting Zephyr OS build zephyr-v3.3.0-2305-gdf7871c41230 ***
Sample program to r/w files on littlefs
Area 4 at 0x700000 on s26ks512s@0 for 59768832 bytes
E: Erasing flash area ... 0
I: LittleFS version 2.5, disk version 2.0
I: FS at s26ks512s@0:0x700000 is 228 0x40000-byte blocks with 512 cycle
I: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
E: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1234: Corrupted dir pair at {0x0, 0x1}
W: can't mount (LFS -84); formatting
W: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1897: Superblock 0x0 has become unwritable
E: format failed (LFS -28)
E: fs mount error (-28)
FAIL: mount id 4 at /lfs: -28
```

**Expected behavior**
I expect the device to erase, format, and successfully mount the storage partition.

**Impact**
Flash storage unusable with littlefs (hyperflash, I believe).

**Logs and console output**
See above.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) OS X
 - Toolchain (e.g Zephyr SDK, ...) Zephyr SDK 0.15.4
 - Commit SHA or Version used Zephyr 3.3.0

bdunlay, 2023-05-11:
I attempted to build the sample with a similar Kconfig overlay to the 1060 board (https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/fs/littlefs/boards/mimxrt1060_evk.conf) but this did not seem to make any difference.

danieldegrasse, 2023-05-16:
@bdunlay Could you try with #53857? There are some known issues with the hyperflash driver's write command on the RT1050, and this PR should fix them

bdunlay, 2023-05-16:
With the changes you referenced applied to v3.3.0, it appears to work:

```
*** Booting Zephyr OS build zephyr-v3.3.0-2307-g9fb867cfdb0f ***
Sample program to r/w files on littlefs
Area 4 at 0x700000 on s26ks512s@0 for 59768832 bytes
I: LittleFS version 2.5, disk version 2.0
I: FS at s26ks512s@0:0x700000 is 228 0x40000-byte blocks with 512 cycle
I: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
E: WEST_TOPDIR/modules/fs/littlefs/lfs.c:1234: Corrupted dir pair at {0x0, 0x1}
W: can't mount (LFS -84); formatting
I: /lfs mounted
/lfs mount: 0
/lfs: bsize = 16 ; frsize = 262144 ; blocks = 228 ; bfree = 226

Listing dir /lfs ...
/lfs/boot_count read count:0 (bytes: 0)
/lfs/boot_count write new boot count 1: [wr:1]
I: Test file: /lfs/pattern.bin not found, create one!
------ FILE: /lfs/pattern.bin ------
01 55 55 55 55 55 55 55 02 55 55 55 55 55 55 55
03 55 55 55 55 55 55 55 04 55 55 55 55 55 55 55
05 55 55 55 55 55 55 55 06 55 55 55 55 55 55 55
07 55 55 55 55 55 55 55 08 55 55 55 55 55 55 55
09 55 55 55 55 55 55 55 0a 55 55 55 55 55 55 55
0b 55 55 55 55 55 55 55 0c 55 55 55 55 55 55 55
0d 55 55 55 55 55 55 55 0e 55 55 55 55 55 55 55
0f 55 55 55 55 55 55 55 10 55 55 55 55 55 55 55
11 55 55 55 55 55 55 55 12 55 55 55 55 55 55 55
13 55 55 55 55 55 55 55 14 55 55 55 55 55 55 55
15 55 55 55 55 55 55 55 16 55 55 55 55 55 55 55
17 55 55 55 55 55 55 55 18 55 55 55 55 55 55 55
19 55 55 55 55 55 55 55 1a 55 55 55 55 55 55 55
1b 55 55 55 55 55 55 55 1c 55 55 55 55 55 55 55
1d 55 55 55 55 55 55 55 1e 55 55 55 55 55 55 55
1f 55 55 55 55 55 55 55 20 55 55 55 55 55 55 55
21 55 55 55 55 55 55 55 22 55 55 55 55 55 55 55
23 55 55 55 55 55 55 55 24 55 55 55 55 55 55 55
25 55 55 55 55 55 55 55 26 55 55 55 55 55 55 55
27 55 55 55 55 55 55 55 28 55 55 55 55 55 55 55
29 55 55 55 55 55 55 55 2a 55 55 55 55 55 55 55
2b 55 55 55 55 55 55 55 2c 55 55 55 55 55 55 55
2d 55 55 55 55 55 55 55 2e 55 55 55 55 55 55 55
2f 55 55 55 55 55 55 55 30 55 55 55 55 55 55 55
31 55 55 55 55 55 55 55 32 55 55 55 55 55 55 55
33 55 55 55 55 55 55 55 34 55 55 55 55 55 55 55
35 55 55 55 55 55 55 55 36 55 55 55 55 55 55 55
37 55 55 55 55 55 55 55 38 55 55 55 55 55 55 55
39 55 55 55 55 55 55 55 3a 55 55 55 55 55 55 55
3b 55 55 55 55 55 55 55 3c 55 55 55 55 55 55 55
3d 55 55 55 55 55 55 55 3e 55 55 55 55 55 55 55
3f 55 55 55 55 55 55 55 40 55 55 55 55 55 55 55

41 55 55 55 55 55 55 55 42 55 55 55 55 55 55 55
43 55 55 55 55 55 55 55 44 55 55 55 55 55 55 55
45 55 aa
I: /lfs unmounted
/lfs unmount: 0
```

danieldegrasse, 2023-05-22:
@bdunlay I am going to close this issue, as #53857 has merged. Please feel free to comment if you'd like me to reopen it.