zephyrproject-rtos/zephyr #57844: Using mold for faster linking

arbrauns, 2023-05-12:
**Is your enhancement proposal related to a problem? Please describe.**
A significant part of the total build time for a zephyr project can be taken up by the linking of up to three ELF binaries, especially for incremental builds. The [mold linker](https://github.com/rui314/mold) is specifically designed as a significantly faster, modern alternative to existing linkers.

Unfortunately, it is currently targeted mostly at (linux) OS packaging rather than embedded use. As such, there are at least a few outstanding issues that make it unsuitable for the time being, among them:

- No support for linker file [`SECTIONS`](https://github.com/rui314/mold/issues/243)
- No support for `--sort-common=descending`

**Describe the solution you'd like**
The build system automatically uses mold if available, for faster builds without user action.

**Describe alternatives you've considered**
The user can opt-in to using mold using a CMake directive.

nashif, 2023-05-12:
@tejlmand  @stephanosio  FYI

stephanosio, 2023-05-12:
I cannot possibly think of a way we can use this linker when it does not have a proper support for the linker scripts; moreover, its author does not seem to be very keen on supporting one either.

I would suggest closing this for now and re-opening it after `mold` becomes a proper linker that supports linker scripts.

arbrauns, 2023-05-12:
Sure, I wasn't suggesting supporting it in its current state, just wanted to post a "tracking issue" to be referred to in the future.

nordicjm, 2023-05-15:
> A significant part of the total build time for a zephyr project can be taken up by the linking of up to three ELF binaries, especially for incremental builds.

I mean, really? Are you running on windows? And how slow is slow? Because I've certainly not noticed linking to be slow on ESXi VMs, desktops or laptops on linux.

arbrauns, 2023-05-15:
> I mean, really? Are you running on windows? And how slow is slow? Because I've certainly not noticed linking to be slow on ESXi VMs, desktops or laptops on linux.

In one of my current projects, touching a single source file causes a subsequent `west build` to take 28 seconds. Actually recompiling the C file only takes a fraction of a second, of course, the rest is linking `zephyr_pre0.elf`, `zephyr_pre1.elf` and `zephyr.elf`.

Linux, i7-11700, project directory on an NVMe SSD.

arbrauns, 2024-02-09:
I don't think there's anything actionable here yet, this is waiting on improvements in upstream mold.