zephyrproject-rtos/zephyr #57851: adc: ads1115: the internal voltage reference is wrong

efra-mx-aqua, 2023-05-12:
I am developing an application using the ADC ADS1115, but the conversion from adc-raw to mV fails when using the internal reference. I have notice that the upstream driver has  a flaw: the the voltage reference is fixed, but the device has a dynamic voltage reference:

> 9.3.4 Voltage Reference
The ADS111x have an integrated voltage reference. An external reference cannot be used with these devices.
source: https://www.ti.com/lit/ds/symlink/ads1115.pdf

Currently is is:
ref_internal = 2048

| Gain | FSR | LSB size |
| --------|--------|------------- |
|2/3 | ±6.144 V | 187.5 μV
|1 | ±4.096 V | 125 μV
| 2* | ±2.048 V | 62.5 μV
| 4| ±1.024 V | 31.25 μV
| 8| ±0.512 V | 15.625 μV
| 16| ±0.256 V | 7.8125 μV

\* Default

This is the configuration would not work,
```
     gain:            2.0
     reference:       Internal
     resolution:      16
```
V = 1 V

```
value = 1 / 62.5e-6 = 16000
adc_raw_to_millivolts(ref_mv=2048, gain=2, resolution=16, val=16000)
= (Gain * FSR * value) / Gain / 2^resolution
= (2048 * 16000) / 2 / 2^16 = 250 mV  
```
 **INCORECT**

This issue has been present since the first version of the driver.
The commit 9b7d71e0446ce8aa70301ae9f5433c406e2707e6 fixes the driver for ADS1113. It introduces a regression when using a ADS1115 and with defaults (Gain=2), but it would work fine when using a Gain=8

**workaround**

If we define `vref` to neutralize the LSB factor:
```
vref = 1 / LSB = Gain * FSR
```

We can resolve:
```
adc_raw_to_millivolts(ref_mv=2 * 4096, gain=2, resolution=16, val=16000)
= (2 * 4096 * 16000) / 2 / 2^16 = 1000 mV
```
**CORRECT**

This means that we have to change the voltage reference every time we change the gain

**To Reproduce**
Use any board with ADC ADS1115, using different gains. Run sample similar to:
 boards/nrf/battery

**Expected behavior**
I expect to be able to convert ADC raw value to millivolts using the internal reference
adc_raw_to_millivolts(adc_ref_internal(dev),...)

**Impact**
Severe when using a ADS115 and ADS1114

**Environment:**
 - OS: Linux
 - Toolchain: gnuarmemb
 - Zephyr 2.7.4

**Additional context**
I am using a similar driver, for zephyr 2.7.2, but the issue seems to be present in version zephyr 3.3. I only have access to our custom boards.

I don't see any logic in version 3.3 to address this issue, so in invite others to test it

efra-mx-aqua, 2023-05-12:
I already have a fix, i will submit it next week

XenuIsWatching, 2023-05-13:
I am on vacation this week, so I’ll take a deeper look when I get back, but my initial thoughts are below…

I don't not see any issue with the driver, but only with how you are calling the api for converting to millivolts. You are calling adc_raw_to_millivolts(ref_mv=2048, gain=2, resolution=16, val=16000) with a gain of 2 and that table you gave for gain is incorrect with the new commit. Everything is now scaled by 1/2 and the default is now gain of 1 (not 2) which means 2.048V corresponds to gain of 1. I did take a look at the source code again and I noticed that the code comment was forgotten to be updated for the enum. I’ll try to get an PR fix for that comment when I get back from vacation. You also should be passing the updated gain value to adc_raw_to_millivolts every time you change the gain value…. not a new reference voltage.

efra-mx-aqua, 2023-05-15:
@XenuIsWatching  Could you point out where it is scaled by half?

The table I gave is from the data sheet.

The internal reference is is dynamic for ads1115 and ads1114. It changes when the gain changes

The current Api is expecting a constant value.

XenuIsWatching, 2023-05-15:
> @XenuIsWatching  Could you point out where it is scaled by half?
> 
> 
> 
> The table I gave is from the data sheet.
> 
> 
> 
> The internal reference is is dynamic for ads1115 and ads1114. It changes when the gain changes
> 
> 
> 
> The current Api is expecting a constant value. 
> 
> 

There is no gain number in the data sheet, you pulled that gain 2/3,1,2,4 etc from the code. There is only FSR voltage. The internal ref voltage never changes... only the gain on that changes which multiples the internal reference voltage.

XenuIsWatching, 2023-05-15:
Also, I just noticed your math is wrong on 2 issues. If you're single-ended, then the resolution is 15b, not 16b. Also, for 1V at 2.048V and gain of 2, that would give you a FSV of 1.048V and but that somewhere around 32000 counts.

efra-mx-aqua, 2023-05-15:
> There is no gain number in the data sheet, you pulled that gain 2/3,1,2,4 etc from the old code

Mmmm, i afraid that is correct. 

> If you're single-ended, then the resolution is 15b
Where is that coming from? 

I will update the issue considering the gain changes, I did not consider that. I am working in running  sample with the latest driver, so I will update the results later

cfriedt, 2023-05-15:
@XenuIsWatching @efra-mx-aqua - if there is a bugfix to backport to v2.7 please add the `backport 2.7-branch` label

efra-mx-aqua, 2023-05-15:
@XenuIsWatching the ASD115 is 16-bit

From the datasheet

> Single-ended signal measurements, where VAINN = 0 V and VAINP = 0 V to +FS, only use
the positive code range from 0000h to 7FFFh. However, because of device offset, the
ADS111x can still output negative codes in case VAINP is close to 0 V

Therefore even though the result would be 15-bit, it is not correct to consider a 16-bit resolution invalid. The best is to get rid of the sign or clear the upper bits with a mask.

or Am I missing something?

closed 2023-05-16 by XenuIsWatching

XenuIsWatching, 2023-05-16:
Comment fixing the code comment
https://github.com/zephyrproject-rtos/zephyr/pull/57919

sslupsky, 2024-03-14:
> It's 16b because it's a differential ADC, I know it's in the fine print but read the api doc carefully. You would send it 15b in differential mode and single-ended mode

@XenuIsWatching There is some confusion about the hardware configuration of this part.  The ADS1115 inputs can be configured as either four singled ended inputs or two differential input.  If the inputs are configured as singled ended inputs, the result is 16 bits.