zephyrproject-rtos/zephyr #57866: Need Vibration support for PineTime DevKit0 in dts

ljunquera, 2023-05-13:
I don't see any vibration support in the PineTime DevKit
boards/arm/pinetime_devkit0/pinetime_devkit0.dts

**Describe why you are asking for this support?**
I don't really know how to add it. My understanding of the devicetree is limited.

**Additional context**
I have seen other projects with an overlay to support it:
https://github.com/ck-telecom/pinetime/blob/develop/app/boards/pinetime_devkit0.overlay

We know what the pins are from here, can we add them all into the board:
https://github.com/joaquimorg/MY-Time/blob/master/src/pinetime_board.h

This is a zephyr example with the vibration:
https://github.com/ck-telecom/pinetime/blob/develop/app/src/service/vibrator.c

nordicjm, 2023-05-15:
This would need a driver, not a very complex driver but probably one that turns a GPIO on and off, or just to work similarly to https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/basic/custom_dts_binding. The one linked in your post is an abuse of the GPIO key driver, it is **not** a button and should not pretend to be one.

ljunquera, 2023-05-15:
I'm no expert on any of this. Does that mean something should be added to the pinetime_devkit0 dts file? Can somebody add it?

ljunquera, 2023-05-15:
Having me add it might be a little dangerous. I would be glad to do it, but think I need to figure out what I need to add. You've given me some good direction. I think I need to sort through it.

nordicjm, 2023-05-15:
Trying is a great way to learn, people will review it and leave comments if they think things can be improved

closed 2023-07-31 by github-actions[bot]