zephyrproject-rtos/zephyr #57867: cbprintf package doesn't handle non-NUL terminated strings with precision correctly

trentp-igor, 2023-05-13:
**Describe the bug**
It's possible to use a string with printf-style formats that is not NUL terminated by specifying a field precision. E.g.:
```c
printf("%.10s", str);  // Print up to the first 10 characters of str.  str does not need to NUL terminated.
```

cbprintf, when used with a string format specifier with a precision, does not work unless the string is also NUL terminated.  If the string does have a terminating NUL, then the string will be correctly truncated to the precision specified (possibly less than the total length of string).  But a string without a terminating NUL will cause a crash or a failure to create a log message, depending on the context and memory contents.

The fundamental error is that that the code in cbprintf finds the length of strings to copy into the package with strlen(), regardless of the precision value.  If a precision is present, then it should use strnlen().

**To Reproduce**
Set these options:
```
CBPRINTF_COMPLETE=y
LOG_PRINTK=n
LOG_MODE_DEFERRED=y
LOG_ALWAYS_RUNTIME=n or y
```

Test program:
```c
#include <zephyr/kernel.h>
#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(main, LOG_LEVEL_INF);
char buffer[2048]; // Must be large
void main(void)
{
        memset(buffer, 'a', sizeof(buffer));  // No NULs in buffer!
        while(1) {
                printk("%.1s\n", buffer);
                LOG_INF("%.1s", buffer);
                k_sleep(K_SECONDS(1));
        }
}
```

The printk(), when log_printk is off, will correctly print 'a' every second.  But the `LOF_INF()` version will not appear.  It is dropped as the package size will include the entire buffer and possibly more and this is too large to log.

If `LOG_MODE_IMMEDIATE` is used, then the `LOG_INF` call causes an exception.

**Expected behavior**
The cbprintf() system is C99 compliant, except as noted in the documentation.

**Impact**
Log messages are silently dropped or random crashes, depending on log mode selected and how long past the end of a string the cbprintf code searches before it finds a NUL byte.

**Environment (please complete the following information):**
zephyr-v3.3.0-3578-g1f43e126a8
Tested on ESP32, should be present on all architectures.

nordic-krch, 2023-05-15:
Unfortunately, this mode is not handled correctly. I will see what can be done but still there will be modes of logging operation where it will be recommended to not use this because there are mode where format string is not available (dictionary mode) or inspecting format string is not feasible (frontend mode which targets highest speed).

Even when format string is present, it might be tricky as so far processing relied on arguments types only.

trentp-igor, 2023-05-17:
t looked to me like this would be difficult to fix with the design of cbprintf, due to package creation only using argument types and not a parsed format string.  It is like the problem that arises with `("%p", (char *)string)`

But in that case, using `%p` with an argument that might be interpreted as a string, there is a warning.  It would be nice to get a warning for using field precisions with strings too.

The current situation is really quite bad.  Usually there will be a zero byte near the string to be printed, just by chance, and so it appears to work perfectly.  But sometimes there won't be, in which case the log message is silently dropped with no error or warning or log dropped message.  It's the worst possible outcome.  Crashing would be better.

So the logging system appears trustworthy, but it is not, and randomly loses the occasional message.  One can spend quite a bit of time trying to figure out how their code is doing what the logs say it is doing before coming to conclusion the logs themselves that are in error.

closed 2023-08-01 by github-actions[bot]

joel-jetcharge, 2023-12-01:
I spent a few days tracking down weird behaviour in my application and have traced it back to this. I agree with [trentp-igor](https://github.com/trentp-igor) that this should produce a warning or at least include documentation under https://docs.zephyrproject.org/latest/services/formatted_output.html#limitations-and-recommendations.

trentp-igor, 2023-12-01:
I'm very disappointed in Nordic for this.  I go to the trouble to track this down and document exactly what is wrong, and they just ignore it. They know the issue is there and what a problem it will create for those who hit it, and they won't even document it?!  It is understood the cbprintf system has limitations for the sake of efficiency, there's no shame in admitting to them.

Having a limitations section and not listed this is just evil.  If there was no documentation, then one would know that they didn't know what was supported and what wasn't.  But to have a list and intentionally omit something, then the user will trust the documentation that is lying to them.

nordic-krch, 2023-12-01:
I'm sorry to hear that. It's not that we try to hide anything. It's just that I'm currently assigned to other tasks and have no time to look into that. Opened PR with documentation: #66013.