zephyrproject-rtos/zephyr #57868: [esp32] The Wifi code runs on zephyr 3.2 can't run on v3.3

ashegheparyshun, 2023-05-14:
**Describe the bug**
I have a project which includes WiFi on esp32 which works fine on zephyr v3.2. Now I'm trying to port the zephyr version to 3.3. which break the WiFi. After some googling I find this code [https://github.com/craigpeacock/Zephyr_WiFi](url) which also works on v3.2 (its wifi status not working but it doesn't care for now) but fail to work on v3.3.
Although in the log it says the esp32 is connected in both v3.2 and v3.3 versions but when I see the connected devices on my router, when esp32 runs v3.2 really it connects to the router while when esp32 runs v3.3 not.

**To Reproduce**
Follow the instruction on [https://github.com/craigpeacock/Zephyr_WiFi](url)

**Logs and console output**
The output for Zephyr v3.3
�*** Booting Zephyr OS build zephyr-v3.3.0-1125-gb9dabf04f0e0 ***
WiFi Example
Board: esp32
Connecting to SSID: Naser
Connected
WiFi Status Request Failed
Disconnected
The output for Zephyr v3.2
�** Booting Zephyr OS build zephyr-v3.2.0-707-g9f652ea04a6d  **
WiFi Example
Board: esp32
Connecting to SSID: Naser
Connected
WiFi Status Request Failed

IPv4 address: 192.168.251.137
Subnet: 255.255.255.0
Router: 192.168.251.66
Ready...

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Toolchain: zephyr-sdk-0.16.0
 - Hardware: esp32 wrover

sylvioalves, 2023-05-18:
@ashegheparyshun, can you check the instructions link you shared?

sylvioalves, 2023-05-18:
@ashegheparyshun, I tested the Wi-Fi on v3.2, v3.3 and main. All replies the below when Wi-Fi connected:  
```c
IPv4 address: 192.168.xxx
Subnet: 255.255.255.0
Router: 192.168.xxx
```
However, getaddr_info() fails with err 12, meaning there is not enough space. What is the issue in here? Wi-Fi IP address or something else?

ashegheparyshun, 2023-05-19:
@sylvioalves Yes, the problem is with the Wi-Fi IP address which can’t be obtained. In fact, I’m testing the code with two different laptops. On both laptops, Ubuntu 22.04 is installed. The first one uses Zephyr v3.3 and zephyr-sdk-0.16.0. The code generated with this laptop fails to connect to the network. The second laptop uses Zephyr v3.2 and zephyr-sdk-0.15.1 version. The code generated using this laptop works as expected. As you tested the code on v3.2 and v3.3, I updated the zephyr v3.2 on the second laptop. The code generated by the second laptop is working as expected. I’m trying to deliver the project as soon as possible and quit the job, so I’m following the rule: if the code works, don’t touch it :). So, I don’t want to upgrade the zephyr-sdk on the second laptop to check whether it works or not. But I’m curious to know which zephyr sdk you are compiling the project with?
In conclusion:
Ubuntu 22.04 + Zephyr v3.3 + zephyr-sdk-0.15.1 works as it is expected.
Ubuntu 22.04 + Zephyr v3.3 + zephyr-sdk-0.16.0 doesn't work for me.

closed 2023-08-07 by github-actions[bot]