zephyrproject-rtos/zephyr #57870: http_client_req: abort reception of data in the response callback

joelguittet, 2023-05-14:
**Is your feature request related to a problem? Please describe.**
I would like to have a way to abort the reception of data in the `response` callback function of `http_client_req` of "subsys/net/lib/http/http_client.c`  (in case the application don't want to continue because of it's own decision).

**Describe the solution you'd like**
I would like the `response` function of `http_client_req` be able to close the connection to stop reading data.

**Describe alternatives you've considered**
Closing the socket in the `response` function, but this hangs and the `http_client_req` function do not returns.
I would like to suggest the `response` function of `http_client_req` should not be void. It should returns something (enum ? integer ?) to indicate if the callback has properly executed, or if an error occurred and the connection should be aborted.

**Additional context**
No additional information.

rlubos, 2025-05-21:
Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/90287 - the response callback function can now return an error to abort the HTTP exchange.

joelguittet, 2025-05-21:
Wonderful 👍

PR #90287 (2025-05-23): net: http: client: Allow to abort download from response callback 
Modify `http_response_cb_t` signature to allow to return an error from the callback to abort the HTTP exchange.
Add test suite for HTTP client, as we now have HTTP sever we can utilise it to verify HTTP client functionality.

Resolves #57870
closed 2025-05-23 by kartben