zephyrproject-rtos/zephyr #57887: Wi-Fi example fails if Ethernet is present in device tree

ShahinHaque97-Corintech, 2023-05-15:
Using the Wi-Fi example on NPX RT1040 through UART to an ESP32-C3 running ESP-AT, This example works fine, provided that the device tree does not contain Ethernet (&enet). 

Wifi Scan Fails. 
Wifi Status results in: 
`uart:~$ wifi status
[00:00:12.234,000] <err> os: ***** USAGE FAULT *****
[00:00:12.234,000] <err> os:   Illegal use of the EPSR
[00:00:12.234,000] <err> os: r0/a1:  0x6001c8a0  r1/a2:  0x80003b50  r2/a3:  0x80003b50
[00:00:12.234,000] <err> os: r3/a4:  0x60022334 r12/ip:  0x600212d3 r14/lr:  0x6000b9cf
[00:00:12.234,000] <err> os:  xpsr:  0x60000000
[00:00:12.234,000] <err> os: Faulting instruction address (r15/pc): 0x60022334
[00:00:12.234,000] <err> os: >>> ZEPHYR FATAL ERROR 35: Unknown error on CPU 0
[00:00:12.234,000] <err> os: Current thread: 0x800012c0 (shell_uart)
[00:00:12.289,000] <err> os: Halting system
 ` 
I believe the issue could be that it is trying to use wifi commands with an ethernet iface.

ShahinHaque97-Corintech, 2023-05-15:
Solution, If both are present on the board, change ethernet priority to be after Wi-Fi.  If net default is set to ethernet, change it to the Wi-Fi module as follows
`net_if_set_default(net_if_lookup_by_dev(DEVICE_DT_GET(DT_NODELABEL(esp32c3))));`

rlubos, 2023-05-17:
This is a known limitation of the Wi-Fi shell, which uses `net_if_get_default()` to obtain interface to work with (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/l2/wifi/wifi_shell.c#L1018). The proposed workaround is what we also use for now.

closed 2023-05-17 by carlescufi

ShahinHaque97-Corintech, 2023-05-17:
Update, 

"wifi scan" works fine, however "wifi connect ssid password" causes the device to restart.

I put a switch on the power to the ethernet so that I am able to kill power to it and "wifi connect" works with ethernet power killed during that command

My question is, even though  net_if_get_default is set to the ESP32C3, Ethernet functions must still be being used in the Wifi functions, I am just unsure where.

Can this thread be re-opened to discuss this or shall I make a new one?

rlubos, 2023-05-18:
@krish2718 Did you experience any problems with `wifi connect` when running in multi-interface environment?

@ShahinHaque97-Corintech It's really hard to find the answer out of thin air, especially that you run a custom hardware configuration. Would be good if you could debug the issue a bit and provide more specific information.

I'll convert the issue into a discussion for now.