zephyrproject-rtos/zephyr #57890: Make Sparkfun Thing Plus 9160 build with secure boot

ssharks, 2023-05-15:
When building the B0 image as part of the secure boot, the CONFIG_GPIO=y is dropped by the build process. This causes a compile error in board.c.

**Describe the solution you'd like**
Option 1:
The setting of the power pin in the function board_sparkfun_thing_plus_nrf9160_init is disabled when CONFIG_GPIO is not defined.

Option 2:
A solution to make sure that the CONFIG_GPIO=y is set during the build processes of the secure boot.

**Additional context**
The error shows during linking, where it has an undefined reference to __device_dts_ord_XXX.

joerchan, 2023-05-15:
Please report on devzone: https://devzone.nordicsemi.com/. B0 is not part of zephyr project.

ssharks, 2023-05-15:
Thanks created a support question: https://devzone.nordicsemi.com/f/nordic-q-a/99767/make-sparkfun-thing-plus-9160-build-with-secure-boot

PR #57893 (2023-06-17): boards: sparkfun_thing_plus_nrf9160: Always enable GPIO
The GPIO driver is required by this board's initialization code (board.c), so it is forced to be enabled always for the board Kconfig, not only enabled by default (in defconfig).

Fixes: #57890
commit 7c1c619c (2023-06-17): boards: sparkfun_thing_plus_nrf9160: Always enable GPIO

The GPIO driver is required by this board's initialization code
(board.c), so it is forced to be enabled always for the board Kconfig,
not only enabled by default (in defconfig).

Fixes: #57890
closed 2023-06-17 by nashif