zephyrproject-rtos/zephyr #57898: Improving the shared memory mechanism for zbus channel message

lgl88911, 2023-05-15:
The following content is mentioned in https://docs.zephyrproject.org/latest/services/zbus/index.html#delivery-guarantees
> Zbus always delivers the messages to the listeners. However, there are no message delivery guarantees for subscribers because zbus only sends the notification, but the message reading depends on the subscriber’s implementation. This is because channels have a mutex protected singleton objects for which message transfer is used. In other words, it can be seen as a single size queue where publishers always overwrite if queue is full.

But I believe that the following issues may also arise from it:
* Messages from the publisher may get lost.
* Subscribers may receive the same message multiple times.
* Subscribers and Listener may observe inconsistent messages.
* Publishers are not aware when their messages are overwritten.

This is because after `zbus_chan_pub` is called to publish a message, the lock on the channel is released. At this point, the subscriber's thread may not be scheduled immediately. If a higher priority thread publishes a message to the channel at this time, the original message in the channel will be overwritten, **causing the old message to be lost and the new message to be received twice**.

Requiring the consumer's priority to be higher than the producer's is one way to avoid this problem. But,requiring the user of zbus to consider the limitations of single message shared memory and carefully design the process or priority from the beginning can be challenging and may cause difficulties for zbus users. It is also not conducive to the modularity and decoupling of the zbus module, as it adds additional considerations in the decoupling process between modules.

The following is a possible solution. If everyone agrees that it is feasible, I can provide a patch：
Replce single message shared memory to FIFO, allowing the channel to retain multiple messages and allowing publishers to wait for a timeout when the FIFO is full, it would be possible for the publisher to know if the message publication fails due to the timeout.

rodrigopex, 2023-05-15:
Hi, @lgl88911. Thanks for your contribution. I will be glad to discuss that with you.

### Context

Since the beginning, I have been following the performant and low-memory path with zbus. The focus always was on constrained devices because if I could improve the communication in constrained devices, the powerful ones would "fly" with that. 

I have studied several approaches to implement that, but shared memory seemed the best choice. It is fast, reliable, requires less memory (compared with message-passing), and enables messages to be persistent. Persistent communication means the message is available (for reading) after the communication; it is not transient. 

For constrained devices, a persistent bus is a place to keep systems states and events. Zbus is excellent for that. I see zbus as a mid-level communication mechanism. Zbus looks like a TCP Socket; you can develop the HTTP over it.

### Commenting on the issues you listed

> * Messages from the publisher may get lost.

That is true. Losses will happen even in FIFO scenarios if the developer mismeasures time. To be sure you will not lose messages with zbus, use listeners for that. They implement a flavor of message-passing, and they will receive the message for each publication. 

> * Subscribers may receive the same message multiple times.

That is also true; you can avoid that by choosing the threads' priorities wisely.

> * Subscribers and Listener may observe inconsistent messages.

What do you mean by inconsistent? They always will access the latest message on the channel. I need help to see when it is consistent. Please describe a scenario. 

> * Publishers are not aware when their messages are overwritten.

Is it essential? Please describe a scenario where it is necessary.

### Talking about the solution

A FIFO is suitable for data items of arbitrary size. In zbus, the message is constant per channel. So, when I started developing zbus, a message queue per channel was one option. With that, some concerns arise: 

1. Memory; the amount of memory would increase a lot since all channels would have their queue;
1. Configuration complexity; the user should choose the size of the queue for each channel, which seems to be an overwhelming requirement;
1. Message persistency; using a message queue makes it hard to keep track of the items in the queue. So persistent communication would be a challenge;
1. Even with a message queue per channel, messages would be lost if the producer performs quicker than the consumers. Maintaining the current behavior;
1. The mutex priority inheritance would be lost as well since zbus uses mutexes to lock channels. It enables the VDED to finish the publication quicker than without it.

lgl88911, 2023-05-16:
@rodrigopex Thank you for the detailed explanation, which enabled me to understand the original intention behind the design of zbus.

**Firstly, let's discuss the issue raised earlier：**
  > That is true. Losses will happen even in FIFO scenarios if the developer mismeasures time. To be sure you will not lose messages with zbus, use listeners for that. They implement a flavor of message-passing, and they will receive the message for each publication.  

Yes, for consumers who don't want to lose messages, they can use listeners. After this point is explained in the documentation, zbus users can easily ensure this.

>That is also true; you can avoid that by choosing the threads' priorities wisely.  

I think this may pose some difficulty for zbus users. In practice, even if all consumers are given a higher priority than the producer, there may still be situations where a consumer cannot timely retrieve data from the channel due to waiting for other resources. This could also result in the message being overwritten.

>What do you mean by inconsistent? They always will access the latest message on the channel. I need help to see when it is consistent. Please describe a scenario.  

For example, when there are two messages A and B, and  occurs, the listener may see both A and B, but the subscriber may see B and B, which could cause some issues in practical use.

> Is it essential? Please describe a scenario where it is necessary.  

For example, when both the temperature sensor and the  send  to a channel at the same time, one of the alarms may be lost due to overwriting, causing necessary processing steps to be missed. Of course, this can be addressed by using separate channels for each sensor.

**Talking about the solution**
> Memory; the amount of memory would increase a lot since all channels would have their queue;  

Indeed, compared to FIFO consume much more memory and can also have efficiency issues. But , nowadays most MCU have sufficient RAM resources, so enabling FIFO is no longer a burden for them.

> Configuration complexity; the user should choose the size of the queue for each channel, which seems to be an overwhelming requirement;  

If using FIFO, this seems to be unavoidable.

> Message persistency; using a message queue makes it hard to keep track of the items in the queue. So persistent communication would be a challenge;  

I don't quite understand what "persistent" refers to. Does it mean that using a  is more advantageous for tracking messages? If not, could you provide more detailed explanations? 

> Even with a message queue per channel, messages would be lost if the producer performs quicker than the consumers. Maintaining the current behavior;  

I am more concerned about , as generating too many messages may lead to unexpected results.

> The mutex priority inheritance would be lost as well since zbus uses mutexes to lock channels. It enables the VDED to finish the publication quicker than without it.  

Writing a message to FIFO occurs after the lock mutex , and it simply replaces the memcpy operation on the shared memory. This should not affect priority inheritance.

**Summarizing my perspective:**
- Losing messages may be tolerable, but it is better to have a mechanism that allows the publisher to know about it.
- **Duplicated messages can have a significant impact**, and under the shared memory mechanism, resetting the msgq may be a way to avoid it.
- Perhaps we can incorporate both the FIFO and shared memory , with shared memory being the default option. Users can choose which mechanism to use through configuration, provided that memory resources are not a concern.

rodrigopex, 2023-05-16:
@lgl88911 thanks again for your comments. 

> Yes, for consumers who don't want to lose messages, they can use listeners. After this point is explained in the documentation, zbus users can easily ensure this.

Great!

> I think this may pose some difficulty for zbus users. In practice, even if all consumers are given a higher priority than the producer, there may still be situations where a consumer cannot timely [...]

For critical scenarios, developers must use listeners. Someone should use subscribers only for parts of the code where message duplication and losses do not affect the application's behavior.

> For example, when there are two messages A and B, and  occurs, the listener may see both A and B, but the subscriber may see B and B, which could cause some issues in practical use.

If B and B are an issue, consider using a listener. If you need to store the message, add that to a queue just for the critical channel, not all of them. 

> For example, when both the temperature sensor and the  send  to a channel at the same time, one of the alarms may be lost due to overwriting, causing necessary processing steps to be missed. Of course, this can be addressed by using separate channels for each sensor.

That is indeed a case of separating channels. 

> Indeed, compared to FIFO consume much more memory and can also have efficiency issues. But , nowadays most MCU have sufficient RAM resources, so enabling FIFO is no longer a burden for them.

As an embedded systems engineer, I always try to use the cheapest MCU possible to solve a problem. So memory is always a problem, IMHO. For sure, several developers share your vision. As I have said, if I solve some problem with the tiny, the giant will solve that with no difficulty. 

> I don't quite understand what "persistent" refers to. Does it mean that using a  is more advantageous for tracking messages? If not, could you provide more detailed explanations?

In this case, the persistent means the message is still on the channel after the communication. On the other hand, in transient communication, no entity can retrieve the information, but who sent and received that on the communication act after the message exchange. In the zbus case, after publishing a message, the message will still be there on the channel to be retrieved by any code portion. It enables state storing. 

> Writing a message to FIFO occurs after the lock mutex , and it simply replaces the memcpy operation on the shared memory. This should not affect priority inheritance.

That would make the communication slower than it is. To write in the message queue, you would lock that as well. So you would lock twice to read or publish a channel. If using the message queue, I would consider removing the mutex. 

> Summarizing my perspective:
> * Losing messages may be tolerable, but it is better to have a mechanism that allows the publisher to know about it.

It would be good, but we must consider how "expensive" it is.

> * Duplicated messages can have a significant impact, and under the shared memory mechanism, resetting the msgq may be a way to avoid it.

I suggest using a listener + dedicated msqg in that case. The msgq for all channels will be "too much" for the majority of the "producer->subscriber" communication. 

> *Perhaps we can incorporate both the FIFO and shared memory , with shared memory being the default option. Users can choose which mechanism to use through configuration, provided that memory resources are not a concern.

I can't see that as a progress on the zbus implementation. Zbus can already solve the scenario you pointed out by using listeners in conjunction with FIFO or message queue. It would increase the communication latency and memory usage with zbus.

lgl88911, 2023-05-17:
@rodrigopex - Thank you for patiently explaining. I think I have a rough understanding of your design intention and the way you handle the existing defects of zbus:
Users should be aware that the messages obtained by subscribers from zbus may be lost or duplicated, and handle them in the following manner:
- For scenarios that do not allow duplication or loss, zbus users should assign them as listeners and use other asynchronous message processing mechanisms in listener.
- For scenarios that can tolerate lost or duplicated messages, subscribers should be used.

I believe that the above processing methods should be documented in zbus's documentation and highlighted to users.
Compared with replacing with FIFO, handing the task of handling lost messages over to users undoubtedly improves the flexibility of zbus and greatly reduces its memory usage.

rodrigopex, 2023-05-20:
@lgl88911 PR submitted: #58101. Thank you for your contribution.

lgl88911, 2023-05-23:
@rodrigopex Thank you for your update. This way, users will have a clearer understanding of how to use zbus in different scenarios.