zephyrproject-rtos/zephyr #57920: i2c_ll_stm32_v2.c - I2c driver not supporting I2C bus recovery.

ashutoshpandey-eaton, 2023-05-16:
**i2c_ll_stm32_v2.c I2c driver not supporting I2C bus recovery** [zephyr/drivers/i2c/i2c_ll_stm32_v2.c]

Description: When I2c lock-up occurs, there is no support for bus recovery.
NOTE: We are using this driver for the stm32-h743 and stm32-u575 controllers.

I found an issue open against bus recovery since 2020 https://github.com/zephyrproject-rtos/zephyr/issues/27715.
Still waiting for the bus recovery feature in the i2c_ll_stm32_v2.c driver.

ashutoshpandey-eaton, 2023-06-05:
We are waiting for it; can we know an approximation of when it will be taken care of?

erwango, 2023-06-05:
> We are waiting for it; can we know an approximation of when it will be taken care of?

Sorry, but there is no plan yet on this specific feature. If you have any kind of urgent need, please your official ST support.