zephyrproject-rtos/zephyr #57940: net: lwm2m: lwm2m thread blocks while holding rd clients mutex lock

m-wenzkowski, 2023-05-16:
**Describe the bug**

For resolving the hostname via DNS, the lwm2m library uses the blocking function [zsock_getaddrinf](https://github.com/zephyrproject-rtos/zephyr/blob/164ad1065f6a6d39e9a31abb7387ffa2c4cb3f7e/subsys/net/lib/lwm2m/lwm2m_message_handling.c#LL3144C17-L3144C17).

This lead to the following problem when the RD client does a regstration:
The registration is done in the lwm2m thread in the function [lwm2m_rd_client_service](https://github.com/zephyrproject-rtos/zephyr/blob/164ad1065f6a6d39e9a31abb7387ffa2c4cb3f7e/subsys/net/lib/lwm2m/lwm2m_rd_client.c#L1247) which holds the rd clients mutex lock for its entire execution time. From there we get the following call chain:

- https://github.com/zephyrproject-rtos/zephyr/blob/164ad1065f6a6d39e9a31abb7387ffa2c4cb3f7e/subsys/net/lib/lwm2m/lwm2m_rd_client.c#L1286
- https://github.com/zephyrproject-rtos/zephyr/blob/164ad1065f6a6d39e9a31abb7387ffa2c4cb3f7e/subsys/net/lib/lwm2m/lwm2m_rd_client.c#L1068
- https://github.com/zephyrproject-rtos/zephyr/blob/164ad1065f6a6d39e9a31abb7387ffa2c4cb3f7e/subsys/net/lib/lwm2m/lwm2m_engine.c#L943
- https://github.com/zephyrproject-rtos/zephyr/blob/164ad1065f6a6d39e9a31abb7387ffa2c4cb3f7e/subsys/net/lib/lwm2m/lwm2m_message_handling.c#L3144

At this point I coud observe that `zsock_getaddrinfo()` call was blocking the lwm2m thread (while still holding the rd client lock).

In the meantime other code of our firmware was calling ` lwm2m_rd_client_is_registred` from the system work queue. `lwm2m_rd_client_is_registred` also needs to hold the RD client lock, hence our system work queue was also blocked, which prevented other time critical tasks from running on the work queue.

After about 35 seconds  the lwm2m library logged `<err> net_lwm2m_message_handling: Unable to resolve address` and the lwm2m thread followd by the system work queue thread were unblocked.

**To Reproduce**
I wasn't able to reproduce it.

**Expected behavior**
The RD clients mutex lock isn't held for longer periods of time.

**Logs and console output**
I added additional log statements to trace locking behaviour of rd client mutex.

Sending fails, rd client schedules re-connect and register:
```
[00:03:09.030,242] <wrn> net_lwm2m_message_handling: Send Timeout
[00:03:09.030,242] <err> net_lwm2m_rd_client: lock rd_client mutex (sm_is_registered)
[00:03:09.030,242] <err> net_lwm2m_rd_client: unlock rd_client mutex (sm_is_registered)
[00:03:09.030,273] <err> net_lwm2m_rd_client: lock rd_client mutex (lwm2m_rd_client_timeout)
[00:03:09.030,273] <wrn> net_lwm2m_rd_client: Confirmable Timeout -> Re-connect and register
[00:03:09.030,273] <err> net_lwm2m_rd_client: unlock rd_client mutex (lwm2m_rd_client_timeout)
[00:03:09.030,303] <err> net_lwm2m_rd_client: lock rd_client mutex (sm_is_registered)
[00:03:09.030,303] <err> net_lwm2m_rd_client: unlock rd_client mutex (sm_is_registered)
```
`lwm2m_rd_client_service()` is executed in the lwm2m thread and takes the lock:
```
[00:03:09.092,437] <err> net_lwm2m_rd_client: lock rd_client mutex (lwm2m_rd_client_service)
[00:03:09.092,468] <err> net_lwm2m_rd_client: lock rd_client mutex (get_sm_state)
[00:03:09.092,468] <err> net_lwm2m_rd_client: unlock rd_client mutex (get_sm_state)
[00:03:09.092,559] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x200144c0 for path 3/0/0
[00:03:09.092,773] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x200144f0 for path 3/0/1
[00:03:09.092,987] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x20014520 for path 3/0/2
[00:03:09.093,170] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x20014550 for path 3/0/3
[00:03:09.093,353] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x20014580 for path 3/0/7
[00:03:09.093,566] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x200145b0 for path 3/0/18
[00:03:09.093,750] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x200145e0 for path 3/0/19
[00:03:09.093,933] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x20014610 for path 4/0/0
[00:03:09.094,146] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x20014640 for path 4/0/2
[00:03:09.094,329] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x20014670 for path 5/0/3
[00:03:09.094,512] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x200146a0 for path 5/0/5
[00:03:09.094,726] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x200146d0 for path 3303/0/5700
[00:03:09.094,909] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x20014700 for path 5/0/7
[00:03:09.095,123] <dbg> net_lwm2m_observation: remove_observer_path_from_list: Removing observer 0x20014730 for path 3303/0/5701
[00:03:09.095,336] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/1, buf:0x2002f577, buflen:1
[00:03:09.095,428] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:0/0/10, buf:0x2002f5a2, buflen:2
[00:03:09.095,489] <dbg> net_lwm2m_registry: lwm2m_engine_get: path:1/0/1, buf:0x2002f580, buflen:4
[00:03:09.095,550] <inf> net_lwm2m_rd_client: RD Client started with endpoint '<REDACTED>' with client lifetime 129600
[00:03:09.095,611] <err> net_lwm2m_message_handling: lwm2m_parse_peerinfo() started
[00:03:09.095,672] <dbg> net_lwm2m_message_handling: lwm2m_parse_peerinfo: Parse url: coap://<REDACTED>
[00:03:09.095,764] <err> net_lwm2m_message_handling: zsock_getaddrinfo() started
```
From here the lwm2m thread is blocked. Next code executed from the system work queue also wants to get hold of the lock:
```
[00:03:09.645,172] <err> at_monitor: start at_monitor_task
[00:03:09.645,233] <dbg> at_monitor: at_monitor_task: AT notif: +CSCON: 1
[00:03:09.645,233] <dbg> at_monitor: at_monitor_task: Dispatching to 0x3cdd1
[00:03:09.645,324] <dbg> at_monitor: at_monitor_task: Dispatching to 0x46295
[00:03:09.645,416] <dbg> lte_lc_helpers: event_handler_list_dispatch: Dispatching events:
[00:03:09.645,416] <dbg> lte_lc_helpers: event_handler_list_dispatch:  - handler=0x0003E389
[00:03:09.645,446] <dbg> lte_lc_helpers: event_handler_list_dispatch:  - handler=0x00049675
[00:03:09.645,446] <dbg> lte_lc_helpers: event_handler_list_dispatch:  - handler=0x0003CAF1
[00:03:09.645,477] <err> net_lwm2m_rd_client: lock rd_client mutex (sm_is_registered)
```
Now also the system work queue is blocked. After some time (usally about 35 seconds) the threads get unblocked (Ignore the following timestamps, I had attached a debugger at this point, and continued the execution later):
```
[00:08:29.091,583] <err> net_lwm2m_message_handling: zsock_getaddrinfo() finished
[00:08:29.091,583] <err> net_lwm2m_message_handling: Unable to resolve address
[00:08:29.091,613] <err> net_lwm2m_rd_client: Cannot init LWM2M engine (-2)
[00:08:29.091,613] <err> net_lwm2m_rd_client: lock rd_client mutex (set_sm_state)
[00:08:29.097,564] <err> net_lwm2m_rd_client: unlock rd_client mutex (set_sm_state)
[00:08:29.097,564] <err> net_lwm2m_rd_client: unlock rd_client mutex (lwm2m_rd_client_service)
[00:08:29.097,595] <err> net_lwm2m_rd_client: unlock rd_client mutex (sm_is_registered)
```

SeppoTakalo, 2023-05-17:
@m-wenzkowski Why are you calling `lwm2m_rd_client_is_registred()` from application? This is not a public API. This is introduced on an internal header.

Application is supposed to follow state changes coming from the callback.
Can you describe what is the purpose of having a separate API for checking the registration status, instead of using the data from callback?

m-wenzkowski, 2023-05-17:
During my debugging I didn't notice that `lwm2m_rd_client_is_registred()` is not part of the public API.

I am using an internal library that bundles lwm2m and modem events. When this library creates a new event it includes the current lwm2m state by calling `lwm2m_rd_client_is_registred()`.But as you already explained this usage is incorrect.

Thank your for your help and sorry for the noise.

closed 2023-05-17 by m-wenzkowski