zephyrproject-rtos/zephyr #57947: Backport Bluetooth fixes to 2.7 branch

sheindel-llt, 2023-05-16:
**Is your feature request related to a problem? Please describe.**
I am working on a project that uses the Zephyr 2.7 release, but have run into several known issues that have been fixed in later releases but are showstoppers for me on 2.7. I would like fixes for these issues to be applied to the 2.7 branch.

**Describe the solution you'd like**
Backport the following PRs to 2.7:
#52536
#48395
#46165

I see there has been at least one other request for some of these PRs to be backported so I think there are others in the community that would benefit:
https://github.com/zephyrproject-rtos/zephyr/issues/43718#issuecomment-1470331996

**Describe alternatives you've considered**
I have attempted backporting these changes myself, but I don't have expertise in the Bluetooth stack and am not confident that my changes are completely effective nor that I have included everything necessary to fix the issues (since there were other changes between the 2.7 release and these PRs being merged that cause conflicts). I could potentially submit my attempt at backporting these changes so they could be reviewed and used by the community, but would need advice on the best way to do that.

Using a more recent release (or main) is not an option for me at this time.

**Additional context**
None

hermabe, 2023-05-16:
https://github.com/zephyrproject-rtos/zephyr/pull/52536 might be relevant, but the other two I think fix an issue that was introduced after 2.7.

sheindel-llt, 2023-05-16:
That's possible, however I was seeing symptoms similar to what those other two described when running 2.7 but it's possible they're separate. I would need to investigate more deeply. #52536 deadlock is definitely affecting me though.

closed 2023-06-07 by theob-pro