zephyrproject-rtos/zephyr #57960: Regression in PWM driver after nrfx 3.0.0 integration

fabiobaltieri, 2023-05-16:
**Describe the bug**
Hi, just bisected a regression related to the Nordic PWM driver to https://github.com/zephyrproject-rtos/zephyr/pull/57259. The `samples/basic/fade_led/` crashes after a couple seconds since that PR, at least on my nrf52dk, could you look into it?

**To Reproduce**
$ git checkout efa3ddd516
$ west update; west build -t flash -p -b nrf52dk_nrf52832 samples/basic/fade_led
<fades a couple times and then stops>
$ git checkout 92a42d8f90
$ west update; west build -t flash -p -b nrf52dk_nrf52832 samples/basic/fade_led
<works>

**Expected behavior**
Fade in... Fade out... Repeat.

**Impact**
Pretty bad, found on an custom board, was a bit of a pain to bisect (most comment in that PR don't build and there was another regression in that range, just my luck).

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK

fabiobaltieri, 2023-05-16:
cc @awojasinski-nordicsemi

fabiobaltieri, 2023-05-16:
Seems like main gets stuck here:

```
nrfx_pwm_stopped_check (p_instance=p_instance@entry=0x77ac <pwm_nrfx_0_config>) at /home/fabiobaltieri/zephyrproject/modules/hal/nordic/nrfx/drivers/src/nrfx_pwm.c:453
453         return ret_val;
(gdb) bt
#0  nrfx_pwm_stopped_check (p_instance=p_instance@entry=0x77ac <pwm_nrfx_0_config>) at /home/fabiobaltieri/zephyrproject/modules/hal/nordic/nrfx/drivers/src/nrfx_pwm.c:453
#1  0x00003a1e in pwm_nrfx_set_cycles (dev=<optimized out>, channel=<optimized out>, period_cycles=<optimized out>, pulse_cycles=<optimized out>, flags=<optimized out>) at /home/fabiobaltieri/zephyrproject/zephyr/drivers/pwm/pwm_nrfx.c:201
#2  0x0000046c in z_impl_pwm_set_cycles (flags=1, pulse=<optimized out>, period=320000, channel=0, dev=<optimized out>) at /home/fabiobaltieri/zephyrproject/zephyr/include/zephyr/drivers/pwm.h:492
#3  pwm_set_cycles (flags=1, pulse=<optimized out>, period=320000, channel=0, dev=<optimized out>) at /home/fabiobaltieri/zephyrproject/zephyr/build/zephyr/include/generated/syscalls/pwm.h:37
#4  pwm_set (flags=1, pulse=400000, period=20000000, channel=0, dev=<optimized out>) at /home/fabiobaltieri/zephyrproject/zephyr/include/zephyr/drivers/pwm.h:558
#5  pwm_set_pulse_dt (spec=<optimized out>, pulse=400000) at /home/fabiobaltieri/zephyrproject/zephyr/include/zephyr/drivers/pwm.h:605
#6  main () at /home/fabiobaltieri/zephyrproject/zephyr/samples/basic/fade_led/src/main.c:38
```

fabiobaltieri, 2023-05-16:
Played with this a bit more, seems like this particular check that was removed as part of the 3.0 upgrade is causing the issue. Or at least, adding it back seems to fix it.

```diff
diff --git a/nrfx/drivers/src/nrfx_pwm.c b/nrfx/drivers/src/nrfx_pwm.c
index 399a9fc..04beeab 100644
--- a/nrfx/drivers/src/nrfx_pwm.c
+++ b/nrfx/drivers/src/nrfx_pwm.c
@@ -448,6 +448,13 @@ bool nrfx_pwm_stopped_check(nrfx_pwm_t const * p_instance)
     {
         ret_val = true;
     }
+    if (nrf_pwm_event_check(p_instance->p_reg, NRF_PWM_EVENT_STOPPED))
+    {
+        p_cb->state = NRFX_DRV_STATE_INITIALIZED;
+        NRFX_LOG_INFO("Disabled.");
+        ret_val = true;
+    }
+
 
     NRFX_LOG_INFO("%s returned %d.", __func__, ret_val);
     return ret_val;
```

awojasinski-nordicsemi, 2023-05-17:
Hi @fabiobaltieri first of all I would like to thank you for spotting the bug. You have a point but this looks a little bit different.
We changed implementation on `nrfx_pwm_stopped_check`. The result from that function is now based on the internal state of the driver that is changed in interrupt from `NRF_PWM_EVENT_STOPPED` event.

The issue here is that MCU is not entering IRQ. Not handling IRQ is caused by two following factors:
- PWM zephyr shim IRQ is not initialized by `IRQ_CONNECT()` macro
- interrupts in nrfx_pwm driver are set only when user event handler is passed to `nrfx_pwm_init()` function

There are at least two solutions I have in my mind. I'll consult them with others and then open a PR.

closed 2023-05-18 by carlescufi