zephyrproject-rtos/zephyr #57965: lib: smf: Exit action gets executed before run action is fully finished

tmilkovic51, 2023-05-16:
**Describe the bug**
State machine framework (`lib/smf`):
When running the current state's `run` function and transitioning to next state by calling `smf_set_state`, the current state's `exit` function immediately gets executed as part of `smf_set_state` call. After `smf_set_state` call finishes, then the current state's `run` function continues with rest of its execution.
This behaviour of `exit` action interrupting the execution of `run` action is unintuitive and not documented anywhere. I am not sure if this behaviour is expected, but it could lead to serious bugs in application logic if developers are not aware of it,
For example, imagine turning the LED off in state's `exit` function, but turning it on at the end of state's `run` function -> the expected behaviour would be that the LED stays off after exiting the current state. Instead, since `exit` gets executed before `run` finishes, the LED would be left on.

Please also mention any information which could help others to understand the problem you're facing:
 - What target platform are you using? 
   - A custom board with NRF51822, but this issue is HW-agnostic and applies to all targets.
 - What have you tried to diagnose or workaround this issue? 
   - One workaround is to write every state's `run` function so that when a state transition is required, `smf_set_state` is called last - at the end of the `run` function (there should be no code after `smf_set_state` call, otherwise that code would get executed after the curent state's `exit` function finishes its execution).
   - A proper workaround is to apply [this custom fix](https://github.com/zephyrproject-rtos/zephyr/compare/main...tmilkovic51:zephyr:fix-smf-exit), which moves all state exit related code to be called after `run` action has finished in `smf_run_state` function. `smf_set_state` function does not execute any action, it only marks that state transition is in progress and saves information about which state to transition into.

**To Reproduce**

Steps to reproduce the behavior:

Execute this modified state machine taken from flat state machine example from [SMF docs](https://docs.zephyrproject.org/latest/services/smf/index.html#flat-state-machine-example):
```
#include <zephyr/kernel.h>
#include <zephyr/smf.h>

/* Forward declaration of state table */
static const struct smf_state demo_states[];

/* List of demo states */
enum demo_state { S0, S1 };

/* User defined object */
struct s_object {
        /* This must be first */
        struct smf_ctx ctx;
} s_obj;

/* State S0 */
static void s0_run(void *o)
{
        smf_set_state(SMF_CTX(&s_obj), &demo_states[S1]);
        printk("Running S0\n");
}
static void s0_exit(void *o)
{
        printk("Exiting S0\n");
}

/* State S1 */
static void s1_run(void *o)
{
       // Stay in this state indefinitely
}

/* Populate state table */
static const struct smf_state demo_states[] = {
        [S0] = SMF_CREATE_STATE(NULL, s0_run, s0_exit),
        [S1] = SMF_CREATE_STATE(NULL, s1_run, NULL),
};

int main(void)
{
        int32_t ret;

        /* Set initial state */
        smf_set_initial(SMF_CTX(&s_obj), &demo_states[S0]);

        /* Run the state machine */
        while(1) {
                /* State machine terminates if a non-zero value is returned */
                ret = smf_run_state(SMF_CTX(&s_obj));
                if (ret) {
                        /* handle return code and terminate state machine */
                        break;
                }
                k_msleep(1000);
        }
}
```

**Expected behavior**
An intuitive behaviour would be to execute the state's `exit` action only after `run` action fully finishes.

**Impact**
This weird behaviour can be avoided if designing all state `run` functions to call `smf_set_state` last, so there is a known, but undocumented workaround.

**Logs and console output**
Should be:
```
Running S0
Exiting S0
```
But in reality, it is:
```
Exiting S0
Running S0
```

**Environment (please complete the following information):**
 - OS: I'm using Debian 12, but host OS should be irrelevant
 - Toolchain: Zephyr SDK v0.16.0
 - Commit SHA or Version used: Relese v3.3.0, but the same behavior is present on current `main` (7039f8ce9c2787f37f5a97c3aca1e1541a0d7d07)

**Additional context**
If this behaviour should be fixed (i.e. it is not intentional), I can open a PR with [the custom fix mentioned above](https://github.com/zephyrproject-rtos/zephyr/compare/main...tmilkovic51:zephyr:fix-smf-exit), if that fix is acceptable.

sambhurst, 2023-05-17:
@tmilkovic51 Thanks for pointing this out. The behavior you describe is not a bug but a limitation of the framework. To mitigate this, you should do an immediate return after all ```smf_set_state``` calls. 

I like your fix, except that it performs the ```if (internal->exit) {``` on every cycle of the state machine.

Or we could update the documentation, pointing out the limitation and possibly add the following macro to the framework:
```
#define SMF_SET_STATE(ctx, state) \
{ \
smf_set_state(ctx, state); \
return; \
}

tmilkovic51, 2023-06-07:
Hi @sambhurst, sorry for taking so long to reply. 

The macro-based approach seems like a quick fix if `smf_set_state` function could be somehow made `static`, so it is only possible to call it by using the provided macro. Otherwise there may be confusion if there were a global function and a similarly named macro which, on a first look, do the same thing, although only one should be used. 

> I like your fix, except that it performs the if (internal->exit) { on every cycle of the state machine.

I don't completely understand why this could be an issue, except for performance reasons. Even then, `smf_run_state()` is usually called from a thread context, and when there is no state transition, the condition in `if (internal->exit)` would evaluate to `false`, meaning the only performance hit when no transition happens would be one extra branch evaluation.

Regardless of the fix to be implemented, I plan to also update the documentation. We have used the SMF on multiple projects already and it has proved to be a nice little tool when you need to quickly implement a state machine.