zephyrproject-rtos/zephyr #57970: LOG macros with CONFIG_NO_OPTIMIZATIONS & CONFIG_LOG disabled causes a blowout of the stack use

dleach02, 2023-05-17:
**Describe the bug**

Functions with LOG macros will use abnormal amounts of stack space if CONFIG_LOG is disabled AND CONFIG_NO_OPTIMIZATIONS=y is set.

Create a simple function (I expanded on hello_world): 

[image]

Then compile the project with:

CONFIG_NO_OPTIMIZATIONS=y
#CONFIG_LOG=y
CONFIG_MYAPP_LOG_LEVEL_ERR=y

Because CONFIG_LOG is disabled, the function myfoo2() only increments the cnt2. But the compiler generates a function preamble that decides to adjust the stack pointer by 0x0154 bytes or 340 bytes.

[image]

If you turn on CONFIG_LOG the SP adjustment is 32 bytes:

[image]

And if you use any other optimization level, the code doesn't have these pre/postambles.

**Expected behavior**
Shouldn't do this. We should be able to reliably use any debug level (within limits).

**Impact**
CONFIG_NO_OPTIMIZATIONS is practically useless without making huge adjustments to the stack sizes.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain (e.g Zephyr SDK, ...) 
 - Commit SHA: 7039f8ce9c2787f37f5a97c3aca1e1541a0d7d07

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

dleach02, 2023-05-17:
It feels like the compiler is getting very confused about the state of things when using the CONFIG_NO_OPTIMIZATIONS (which is -O0). The logging subsystem should null out all the LOG macros if CONFIG_LOG is not set:

```
#ifdef CONFIG_LOG
...
#else

#define LOG_ERR(...)
#define LOG_WRN(...)
#define LOG_INF(...)
#define LOG_DBG(...)
#define LOG_PRINTK(...)

#endif /* CONFIG_LOG */
```
This addresses the stack problem but opens up other warnings of unused variables that are setup for consumption by the logging system.

jgl-meta, 2023-05-23:
This is also observed on LTS.

marcowidmer, 2023-10-30:
The issue is CONFIG_LOG_ALWAYS_RUNTIME. If enabled, it significantly reduces the required amount of stack space.

If logging is enabled, CONFIG_LOG_ALWAYS_RUNTIME defaults to y if CONFIG_NO_OPTIMIZATIONS is set. However, if logging is disabled (CONFIG_LOG=n), CONFIG_LOG_ALWAYS_RUNTIME is not configured by Kconfig, which leads to log_msg.c allocating a lot of stack space.

I was able to fix the issue by adding the following to Kconfig:
```
if !LOG
config LOG_ALWAYS_RUNTIME
	bool
	default y if NO_OPTIMIZATIONS
endif
```

Proposed fix: Kconfig should always set CONFIG_LOG_ALWAYS_RUNTIME=y if CONFIG_NO_OPTIMIZATIONS is set, even if CONFIG_LOG=n.

PR #67164 (2024-06-17): logging: use runtime message creation when logging is disabled
Fixes #57970
closed 2024-06-17 by aescolar