zephyrproject-rtos/zephyr #57971: Error building LittleFs on STM32 nucleo H723zg

Askzephyr, 2023-05-17:
Hi All,
I am getting below error while building LittleFs on STM32 nucleo H723zg.
Error:
undefined reference to z_fsmp_DT_N_S_fstab_S_lfs1

Overlay files used :
&flash0 {
    partitions {
        compatible = "fixed-partitions";
        #address-cells = <1>;
        #size-cells = <1>;

        lfs1_partition: partition@C0000 {
            label = "lfs1_partition";
            reg = <0x0 0x000C0000 DT_SIZE_K(256)>;
        };
    };
};

/ {
    fstab {
        compatible = "zephyr,fstab";
        lfs1: lfs1 {
            compatible = "zephyr,fstab,littlefs";
            mount-point = "/lfs1";
            partition = <&lfs1_partition>;
            automount;
            read-size = <1>;
            prog-size = <1>;
            cache-size = <256>;
            lookahead-size = <32>;
            block-cycles = <512>;
        };
    };
};

DTS Configuration for memory layout:
 1 MB flash memory , Boot partition  with 128 Kb , Image 1 and Image 2 : 256 Kb each
Scratch partition : 128 Kb , Storage partition : 256 Kb

Laczen, 2023-05-17:
Hi @Askzephyr, the reg entry is not corresponding to the #address-cells and #size-cells. Regardless of this error lfs on the internal flash of the stm32h723zg will not work, as configured lfs has only two sectors to work with and this is not sufficient.

Laczen, 2023-05-17:
@tej113, don't put any time into this, littlefs needs sufficient sectors to work correctly. On the internal flash of the stm32h723zg it will just not work due to the large sector size.

You might get it to work for the sample (as this only uses 1 or 2 files), but in a real world scenario it will not be usable.

Laczen, 2023-05-17:
> Thanks @Laczen for this clarification, Yes, As i see H723zg having sector size of 128kb. Just want to understand more on the above point as you mentioned ..can you please let me know how much size it may require to make it work for atleast for 1 or 2sample files .??

IIRC littlefs needs at least 2 sectors for the "superblock" and then one per file/folder and at least one extra sector. To get more details on this you should ask on the littlefs github repo.

Laczen, 2023-05-18:
@tej113, can this issue be closed as "not supported" ?

tej113, 2023-05-22:
@Laczen  Then, What may be the solution to use filesystem on nucleo_h723zg in zephyr ..??

closed 2023-05-22 by Laczen

tej113, 2023-05-25:
Hi @Laczen ,
Can you tell me the solution to get rid of this error atleast for just for 1 or two files.