zephyrproject-rtos/zephyr #57987: hal: renesas: Add precompiled BLE controller image

andrzej-kaczmarek, 2023-05-17:
## Origin

This is a precompiled binary of Apache NimBLE controler that runs on CMAC (BLE controller integrated in Renesas SmartBond(tm) SoCs)

## Type

- [x] Precompiled library
- [ ] Firmware image

## Module

hal_renesas

[zephyrproject-rtos/hal_renesas@`7c1dd85` (#2)](https://github.com/zephyrproject-rtos/hal_renesas/pull/2/commits/7c1dd8530d6db6e3be7aeae640ab72a927f1843f#diff-fc9d30ebe05422c7aae285e96028216a82d788539a9a72a12f2e5c9ff49555a6R17)

## Purpose

The precompiled binary allows to enable BLE support on DA1469x SoCs. The same binary can be built at any time by anyone since the code for Apache NimBLE controller (and Apache Mynewt, which the controller is part of) is publicly available and licensed under Apache-2.0 license (https://github.com/apache/mynewt-core & https://github.com/apache/mynewt-nimble). This however requires to setup complete environment to build Apache Mynewt. The purpose of including this precompiled binary in hal_renesas is to provide BLE on DA1469x out-of-the-box.

## Pull Request

https://github.com/zephyrproject-rtos/hal_renesas/pull/2

## Dependencies

none

## License

The Apache NimBLE controller is licensed under Apache-2.0

carlescufi, 2023-05-18:
@andrzej-kaczmarek this was discussed in yesterday's TSC, here's an excerpt from the minutes:

[image]

https://docs.google.com/document/d/1cMkXkFN-Yje-7zIGZmWMfkrVCju85UNnKDfjng9vEv0/edit#

carlescufi, 2023-05-18:
@andrzej-kaczmarek the firts thing to be fixed is that we do not host `.a` files of any kind in repos in this organization. Please read through the binary blobs page and move the blob to an external hosting provider (can be for example nimble's GitHub):

https://docs.zephyrproject.org/latest/contribute/bin_blobs.html
https://docs.zephyrproject.org/latest/develop/modules.html#binary-blobs

You can find multiple examples in hal_infineon and hal_espressif

carlescufi, 2023-09-14:
This passed the TSC vote on 31st May 2023.

closed 2023-09-14 by carlescufi