zephyrproject-rtos/zephyr #57988: LE Audio: MCS targetted control point notifications

MariuszSkamra, 2023-05-17:
**Is your enhancement proposal related to a problem? Please describe.**
Media Control Point as well as the Search Control Point characteristics shall send notifications to the device that initiated control point operation only. The notification carries information about the operation result. Currently, the connection context is not passed to the `media_proxy_sctrl_cbs.command` nor `media_proxy_sctrl_cbs.search` callbacks and the notification is sent to all devices connected.

**Describe the solution you'd like**
The mentioned callbacks shall take as parameter `struct bt_conn *` reference to use it to send the notification to.  

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Thalley, 2023-06-05:
This does not make sense to spend effort on until fixing https://github.com/zephyrproject-rtos/zephyr/issues/57441 as that will likely fix this issue anyways.

asbjornsabo, 2023-11-28:
I need a refresher.
1) Where in the spec does it say that the server shall send notifications to the device that initiated the connection _only_?  What I find is says it says to notify the client, but it does not forbid notifying others.
2) If I remember correctly, the server uses a quite generic GATT notification call.  But that call (as it used to be) only notifies those clients that have subscribed, I believe.  (And if there is no-one subscribed, there will be an error returned, which can be ignored.)

Thalley, 2023-11-28:
> I need a refresher.
>
>     1. Where in the spec does it say that the server shall send notifications to the device that initiated the connection _only_?  What I find is says it says to notify the [...]

In MCS 3.18.2 (The Media Control Point) it states:

> Notification is sent from the server **to the client** after each Media Control Point opcode write.

I read that as only the client that did the write, should get the notifications. 

However all (or almost all?) characteristics in MCS has the text
> A server device shall notify this characteristic to **the client device**

Which uses similar, but slightly different, words, but I think those notifications (e.g. track changed) should go to all clients, and only a single one.

closed 2024-03-06 by Thalley