zephyrproject-rtos/zephyr #58007: Drop z_ prefix as identifier for internal/private functions and revert back to _

nashif, 2023-05-17:
## Introduction

Revert back to old naming conventions and use _ for private functions and drop the usage of z_ as a prefix in zephyr completely.

### Problem description

The z_ as prefix for internal/private functions is misleading and giving the impression functions with z_ are public APIs. 

### Proposed change

Go back to _ (underscore) as an identifier for private functions and remove all usage of z_ in zephyr.

## Detailed RFC

This is going to be a treewide change that will clarify zephyr APIs and avoid any confusion about what functions should be used out of tree. This is not going to be an API change per se, it will however cause some issues with users depending on internal functions out of tree.

The scope of private functions should also be clarified as part of this RFC and if we should allow usage of internal/private calls across the tree, i.e. subsystem A calls private functions for subsystem B.

### Concerns and Unresolved Questions

Major change that needs lots of preparating and discussions.

## Alternatives

document clearly the z_ prefix and do not allow exporting functions with z_ in public APIs.

carlescufi, 2023-05-17:
CC @stephanosio @aescolar @gmarull @cfriedt @andyross @galak

andyross, 2023-05-17:
I never liked z_ and think this would be nicer.  Double or single underscore doesn't matter much to me.  But I do point out that I really (really) hate the churn from treewide syntax changes like this and think we should implement some kind of blood oath to be taken by all involved to ensure this is the last time we're going to do this.

henrikbrixandersen, 2023-05-23:
Would this be a new policy going forward or do you intend to change existing code?

gmarull, 2023-05-26:
An approach I used in other projects in the past is the one from `libgit2`: https://github.com/libgit2/libgit2/blob/main/docs/conventions.md

ie, public functions look like `mylib_public()` and internals/semi-private use `mylib__internal()` (double underscore). Problem is that Zephyr has never used a standard prefix like `z_`, so we can't have `z_gpio_get()` and `z__gpio_something_internal()`...

aescolar, 2023-05-31:
My 2 cents:
I think this boils down to the same issue we discussed a few times:
* Any symbol/macro we define globally has a chance of colliding with another library/app symbol/macro. The only nice way to minimize this, is to namespace everything zephyr defines out of the way, as we do in many/most areas.
* namespacing symbols with just `_` or `__`, apart from guidelines like MISRA, is also dangerous, as we are likely to hit C libraries and others who feel they own the "reserved" areas. 
* `z_` is indeed not evidently indicating "internal". Something like `zint_ ` or `kint_` would have been, or even `_k_` or `_z_`
* There is the reasoning of "Zephyr is the OS and therefore it is important enough to define any reserved names it wants. And, in case of conflicts it is others who should fix their naming". but, IMHO, this is just asking for trouble.

> But I do point out that I really (really) hate the churn from treewide syntax changes like this and think we should implement some kind of blood oath to be taken by all involved to ensure this is the last time we're going to do this.

I fully agree.

An olden link:
https://github.com/zephyrproject-rtos/zephyr/wiki/Naming-Conventions

nashif, 2023-06-02:
great input, I am well aware of all those constraints and issue with standards, but the current situation is bad enough IMO and this z_ prefix is very confusing and I have seen private functions being used as if they were public APIs. Most notable example is https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys/fdtable.h, where some has decided to use z_ as the preferred prefix for zephyr public APIs...

ejohnso49, 2023-08-28:
As a user, I'd say I haven't hit a ton of confusion understand that `z_` prefixed functions are private and the complications of their use. As others have mentioned, I would be worried about the churn. Regarding the prevalence of the misuse of `z_` for public functions, is it feasible to change the misuse instead?

cfriedt, 2023-08-29:
I've been doing this in a couple of places already, but for reserved, internal Zephyr API, I believe the `__z_` prefix should be OK in that it does not violate any ISO C rules.

ceolin, 2024-10-03:
Have we decided which direction we are going ?

@aescolar made good points. We have being fixing some symbols collision in an ad-hoc way, here is one more https://github.com/zephyrproject-rtos/zephyr/issues/78906 needed.

I honestly don't think we can take the approach that we are the OS and others should adapt to that. It seems to me that we need a proper namespace for, pretty much everything.

cfriedt, 2025-03-18:
> I've been doing this in a couple of places already, but for reserved, internal Zephyr API, I believe the `__z_` prefix should be OK in that it does not violate any ISO C rules.

Just wanted to follow-up - my `__z_` statement above seems to be wrong:

> in C++, identifiers with a double underscore anywhere are reserved everywhere; in C, only the ones that begin with a double underscore are reserved.

> The use of two underscores (`__') in identifiers is reserved for the compiler's internal use according to the ANSI-C standard

carlescufi, 2025-03-18:
Architecture WG:

Options:

- `_z_` and `__z_` are problematic because they are reserved, but we could consider an exception for those
- `zpriv_` and `zint_` are alternatives that can be considered
- We need to consider whether using an exception is the right path forward
- Other RTOS (e.g. ThreadX) use `_`

aescolar, 2025-03-18:
I'd just forget about the `_*` ones to close the discussion about reserved namespace, as that discussion would keep happening every X months.
I'd recommend to just list a few options for the private one and do a vote.

carlescufi, 2025-04-22:
Architecture WG:

- The following links are analyzed to understand whether `_` as a prefix (with lower case after) could be compatible with the standards:
  - https://www.dii.uchile.cl/~daespino/files/Iso_C_1999_definition.pdf (7.1.3)
  - https://www.gnu.org/software/libc/manual/html_node/Reserved-Names.html
  - https://devblogs.microsoft.com/oldnewthing/20230109-00/?p=107685

- We delay the decision until next week to leave some time for additional investigation

carlescufi, 2025-04-29:
Architecture WG:

- When we transitioned from underscore (`_`) to `z_`, many functions got renamed that were global
-

jhedberg, 2025-04-29:
> next step is what we replace it with

I think I mentioned this in some past meeting, and someone had a counterargument, which I forgot, so I'll repeat it: what we essentially have is the desire to represent two independent attributes of an API through some naming convention:

 - public/private
 - which subsystem the API belongs to

There seems to be a consensus that the the naming convention should be based on prefixes, however I don't see it as self-evident that the *order* of these prefixes should be "public/private first, subsystem second". E.g. I could perfectly well imagine using `bt_priv_*` for private Bluetooth APIs. Anyway, if someone wants to repeat the counterargument to this here, feel free, so we have it documented and can move on, or do people think that this alternative is worth considering as well?

carlescufi, 2025-05-01:
> There seems to be a consensus that the the naming convention should be based on prefixes, however I don't see it as self-evident that the _order_ of these prefixes should be "public/private first, [...]

My opinion here is clear: the most important attribute for developers to see first is the public or private nature of the API. So, in my opinion, it should always be `zpriv_bt_*`.

carlescufi, 2025-05-06:
Architecture WG meeting:

- @nashif mentions that today we have multiple public methods prefixed by `z_`, so we should tackle the public prefixes first before we go and fix the private ones
- @carlescufi mentions that there's a difference between private APIs that your application should never call and hooks or other functions that can be overridden, but should not be called directly.

carlescufi, 2025-05-13:
Architecture WG meeting:

Vote:

- Options: 
  - `zpriv_{area prefix}_`
  - `priv_{area prefix}_`
  - `privz_{area prefix}_`
  - `private_{area prefix}_`
  - `{area prefix}_priv_`

cfriedt, 2025-05-14:
A checklist would be really handy to ping maintainers for each area.

carlescufi, 2025-05-20:
Architecture WG meeting:

- @nashif has run some tests on the new naming scheme. In general it works well, it also shows the use of private APIs in some components and tests, which will need to be changed
- @henrikbrixandersen mentions the difference in CAN between APIs that can be used by applications and those used only by CAN driver implementations (in or out of tree)
- @aescolar suggests calling an API "private" because of the fact that they are only used within a single subsystem is wrong. Out-of-tree implementations of that driver that rely on these APIs should be able to count on them.