zephyrproject-rtos/zephyr #58009: how to use mdns_responder sample with run-time configurable hostname

chmorgan, 2023-05-17:
I'm looking to use the mdns approach in samples/net/mdns_responder/ but with a run-time configurable hostname.

However this fails because DNS_SD_REGISTER_SERVICE() is a macro defining a structure in a particular segment, so its defined at compile time and not run-time.

I'd like to generate the hostname at runtime using the hwinfo MAC address, something like:

```
snprintf(runtime_hostname, sizeof(runtime_hostname), "module%X", macValue);
```

so each module has a hostname like 'moduleABCD.local'.

But passing a variable in to that macro results in:

```
mdns_resolver.c:50:41: error: initializer element is not constant
   50 |         DNS_SD_REGISTER_SERVICE(zephyr, net_hostname_get(),
      |                                         ^~~~~~~~~~~~~~~~
/app/zephyr/include/zephyr/net/dns_sd.h:112:17: note: in definition of macro 'DNS_SD_REGISTER_SERVICE'
  112 |                 instance,                                             \
      |                 ^~~~~~~~
/app/adctest/src/mdns_resolver.c:50:41: note: (near initialization for 'zephyr.instance')
   50 |         DNS_SD_REGISTER_SERVICE(zephyr, net_hostname_get(),
      |                                         ^~~~~~~~~~~~~~~~
/app/zephyr/include/zephyr/net/dns_sd.h:112:17: note: in definition of macro 'DNS_SD_REGISTER_SERVICE'
  112 |                 instance,                                             \
      |                 ^~~~~~~~
```

Is there another approach to use? Am I doing this wrong?? I looked through the DNS docs and didn't see another way to register a service.

jukkar, 2023-05-17:
Have you checked `net_hostname_set_postfix()`, it can be used to set the postfix of the hostname. Then the `net_hostname_get()` will return the hostname with some postfix (like a MAC address). Do you need to use the SD macros here, the mDNS responder should work just fine without those macros.

chmorgan, 2023-05-17:
@jukkar you know it isn't obvious that mdns will just work, the mdns_responder example makes it seem like the service() code is a necessary function.

Tested here and with no mdns configuration or calls it appears to be resolving as expected. I've been doing embedded dev for many years, its a surprise that this works without any kind of software config on my part (other than the hostname update which worked well btw), AND that I missed the fact in the docs that no software config is required...

Thoughts on how to clarify/resolve this? Should that code be removed/moved to another example and main() replaced with a while(true) { kmsleep(xxx); }. ??

jukkar, 2023-05-17:
Not sure why we would need to change the sample code, the current one works just fine. But if you think more clarifications are needed, please propose a PR. The mdns_responder sample readme file could be enhanced to clarify what is needed from the user.

chmorgan, 2023-05-17:
I’m not necessarily suggesting we change the function of the app except
that 99% of its code is unrelated to getting mdns working. So as a reader I
copied a good part of the service code before this conversation.

So it’s confusing at the least.

I can propose some language changes to clarify in the main.c file that I
think would help the next person. Bigger questions are whether we should
simplify the example to be mdns only, which would be a zephyr config option
and a sleep in the main()…

chmorgan, 2023-05-18:
@jukkar opened a PR, #58017

henrikbrixandersen, 2023-05-23:
Closing as fixed since https://github.com/zephyrproject-rtos/zephyr/pull/58017 was merged.

closed 2023-05-23 by henrikbrixandersen