zephyrproject-rtos/zephyr #58031: Zephyr Implementation of POSIX select/poll (zsock_select) timeout mechanism does not awaken thread in time

bob2oneil, 2023-05-18:
I have a emulator application running using the Native-POSIX board target that bridges Ethernet frames between to virtual Ethernet interfaces.  I am using a single threaded approach with multiplexed i/o using the standard Linux _select_ statement (_zsock_select_).   I am using a single threaded approach initially to avoid having any form of locking between threads to avoid any locking issues in multi-threaded applications.

I am using Zephyr OS build zephyr-v3.2.0 LTS on desktop Linux as the development and execution platform.

All is well and good w.r.t to triggering on the data flows.  However, the thread timeout values (expressed as Linux standard _timeval_), do not work similar to all my other experiences with both desktop and embedded Linux using the POSIX _select_ API.  I my previous experiences, the thread would awaken on a timeout expiration within a few 100 microseconds of the specified input timeout value.

However, the Zephyr implementation has 2 undesirable characteristics on the awakening of the thread pending in the _select_ statement.

1.  The thread awakens far earlier than the timeout value specified even without the triggering of any file descriptor.  While this causes the main thread to perhaps burn some CPU time needlessly, it is not an error so much as a undesired behavior if it means some other thread has work to do.

2.  More importantly, I have measured that in about 5 to 10 percent of the time, the thread does not awaken close to the timeout value, and can sometimes awaken more than 25 milliseconds late.  This is an error condition for me, as I miss real time deadlines.

To illustrate these results, consider the following log of activities into and at the exit from the Zephyr implementation of the select API.

The **select() IN** statement are the arguments into the select API, **select() OUT** is the output where the actual elapsed time in the thread pend state is labelled **elapsed**, and the **count** value is the number of triggered file descriptors, where a value of 0 represents the timeout mechanism caused the thread to be awakened rather than i/o.

For issue #1, we can see that the thread does not pend to the specified timeout value, but returns with less of a timeout as specified by the measured elapsed time.  Not a big deal, just not what was expected for my embedded Linux experiences.

For issue #2, which is the error condition, at the bottom of the logging below, there is an entry with a timeout value as around 6 ms (in bold) for the _select_ entry, but the thread does not reawaken from the pend until 24 ms (in bold), where I miss a 10 ms epoch deadline.

So is there any explanation for why this might be happening in around 5 to 10 percent of the time (meaning some other Zephyr RTOS thread needs to run for this duration under Native-POSIX blocking the main thread from running)?

I currently have the main thread set to a priority of -2 matching the highest priority threads that are running.  Tweaking this thread's priorities does not seem to change the results. A list of the application threads and priorities are listed below for reference.

Assuming that this is just the way it works and a quick fix is not available, what other recommendations would the Zephyr team have for providing a reliable timing mechanism for 10 ms epochs, and would it require a different thread (e.g. the main thread would be the interface packet read and write to tx queue storage, and a timing thread based on the reliable timer would be the read from queue and interface writer?).

select() IN [epoch 91]: now (8571/918701), timeout (0/**1300**), m_tvNextEpoch (8571/920000)
select() OUT[epoch 91]: now (8571/918832), timeout (0/1300), tvNextEpoch (8571/920000), elapsed (0/**131**), count 0
select() IN [epoch 91]: now (8571/918844), timeout (0/1156), m_tvNextEpoch (8571/920000)
select() OUT[epoch 91]: now (8571/919079), timeout (0/1156), tvNextEpoch (8571/920000), elapsed (0/235), count 0
select() IN [epoch 91]: now (8571/919096), timeout (0/905), m_tvNextEpoch (8571/920000)
select() OUT[epoch 91]: now (8571/920531), timeout (0/905), tvNextEpoch (8571/920000), elapsed (0/1435), count 0
Post select: past next epoch: TDMA epoch 91 at now (8571/920531), timeout (0/905), next epoch (8571/920000)
select() IN [epoch 92]: now (8571/920566), timeout (0/9435), m_tvNextEpoch (8571/930000)
select() OUT[epoch 92]: now (8571/920575), timeout (0/9435), tvNextEpoch (8571/930000), elapsed (0/9), count 1
select() IN [epoch 92]: now (8571/920594), timeout (0/9406), m_tvNextEpoch (8571/930000)
select() OUT[epoch 92]: now (8571/921062), timeout (0/9406), tvNextEpoch (8571/930000), elapsed (0/468), count 0
select() IN [epoch 92]: now (8571/921090), timeout (0/8910), m_tvNextEpoch (8571/930000)
select() OUT[epoch 92]: now (8571/923875), timeout (0/8910), tvNextEpoch (8571/930000), elapsed (0/2785), count 0
select() IN [epoch 92]: now (8571/923898), timeout (0/**6102**), m_tvNextEpoch (8571/930000)
select() OUT[epoch 92]: now (8571/948738), timeout (0/6102), tvNextEpoch (8571/930000), elapsed (0/**24840**), count 0
Post select: past next epoch: TDMA epoch 92 at now (8571/948738), timeout (0/6102), next epoch (8571/930000)
EPOCH timeout: 1 missed epochs detected: epoch timeout called at (8571/948738), current/prev/epoch counter 94/92/105

```
Threads:
 0x15e8d20 ddlt_udp_server
	options: 0x0, priority: 20 timeout: 0
	state: pending, entry: 0x441d08
	stack size 2048, unused 2008, usage 40 / 2048 (1 %)

*0x167c520 shell_telnet
	options: 0x0, priority: 14 timeout: 0
	state: queued, entry: 0x459b0e
	stack size 4096, unused 4056, usage 40 / 4096 (0 %)

 0x174c158 rx_q[0]   
	options: 0x0, priority: -1 timeout: 0
	state: pending, entry: 0x46fc1a
	stack size 1504, unused 1464, usage 40 / 1504 (2 %)

 0x174d180 eth_native_posix_rx-zethblk1.1
	options: 0x0, priority: -2 timeout: 5
	state: suspended, entry: 0x47bf08
	stack size 40, unused 0, usage 40 / 40 (100 %)

 0x174d280 eth_native_posix_rx-zethblk1.0
	options: 0x0, priority: -2 timeout: 0
	state: suspended, entry: 0x47bf08
	stack size 40, unused 0, usage 40 / 40 (100 %)

 0x174c298 tx_q[0]   
	options: 0x0, priority: -1 timeout: 0
	state: pending, entry: 0x46fb42
	stack size 1200, unused 1160, usage 40 / 1200 (3 %)

 0x1680340 net_mgmt  
	options: 0x0, priority: -1 timeout: 0
	state: pending, entry: 0x46c049
	stack size 768, unused 728, usage 40 / 768 (5 %)

 0x174c6c0 tcp_work  
	options: 0x0, priority: -16 timeout: 0
	state: pending, entry: 0x485efa
	stack size 1024, unused 984, usage 40 / 1024 (3 %)

 0x174d5e0 sysworkq  
	options: 0x0, priority: -1 timeout: 0
	state: pending, entry: 0x485efa
	stack size 1024, unused 984, usage 40 / 1024 (3 %)

 0x167bf20 shell_uart
	options: 0x0, priority: 14 timeout: 0
	state: pending, entry: 0x459b0e
	stack size 4096, unused 4056, usage 40 / 4096 (0 %)

 0x174d380 idle      
	options: 0x1, priority: 15 timeout: 0
	state: , entry: 0x48540a
	stack size 256, unused 216, usage 40 / 256 (15 %)

 0x174d480 main      
	options: 0x1, priority: -2 timeout: 477
	state: pending, entry: 0x4849f1
	stack size 2048, unused 2008, usage 40 / 2048 (1 %)
```

aescolar, 2023-05-22:
@bob2oneil From a relatively quick read of this issue description I'm not certain I understand where you think the problem is. I'm guessing you do not think there is an inherent problem with zsock_select()(?), but that instead your expectations wrt to the relationship between Zephyr and the host (Linux) are not met, or more specifically how the Zephyr calls relate to how you would expect the native equivalent Linux call would have performed.
If the later I think the issue may be a misunderstanding: Please note that native_posix is not a translation layer (say, akin to wine) in which Zephyr APIs are converted into Linux APIs. It is a Zephyr test platform. The fact that you can call Linux APIs from it is a practical benefit which allows you to more easily build tests, but should not be understood as your Zephyr application being suddenly "compatible" with, or "running" on Linux. Your application is still running in Zephyr, and the only "time" your application should know about is the time as Zephyr knows it, which is decoupled from the host time.
In this context, building an application for Zephyr and running it in Linux with the intention of using it as an actual application would be misguided.
In relation to the issue description this means:
**You should not call the host clock_gettime() and expect it to be coherent with Zephyr's time**. They are only vaguely related when you run native_posix in real time; And the relationship is that of the Zephyr execution being stalled every system tick timer to allow "the host time to catch up". If you want less divergence/jitter/jerkiness between both you can increase the system tick resolution. But note that there is a limit here given by your host OS load and its own scheduling resolution.
If you run native_posix in no-real-time mode, Zephyr time and the host time are not related in any way. Any timeout in a Zephyr API will have nothing to do with the host time.
I hope this answer helps you. If not you would need to narrow down your question and be a bit more precise about what you use and what you expect.

aescolar, 2023-05-24:
@henrikbrixandersen how come you labelled this as a bug? From the description I gather this is a misunderstanding from the submitter, and not a bug. But without the submitter not having came back yet I preferred to not close the issue.

aescolar, 2023-05-24:
@bob2oneil Does my response https://github.com/zephyrproject-rtos/zephyr/issues/58031#issuecomment-1556832161 clarify things enough? Can we close this issue?

bob2oneil, 2023-05-24:
Note that this could be related in part to the setting associated with Native-POSIX slowdown of the platform assuming the select() timeout is using some of the underlying Zephyr timing APIs.  The results above reflect timing with the platform NOT slowed down, and are relative to the "wall clock" time on Linux, which seems to match the host platform time (meaning 1 second change is what is observed in human time).

As a example, when using the Zephyr timer, I had to use the following to get the Zephyr clock expiration at 10 msecs match somewhat the actual wall clock time, meaning to get a 10 msec period, I had to multiply the 10 msec argument by 500.  This was observed by calculating the elapsed time between successive calls to the expiration function, which is measured relative to the wall clock time.

```
void App::initializeZephyrEpochTimer()
{
    printk("Initializing Zephyr Epoch Timer\n");
    k_timer_init(&m_timer, App::timer_expiry_fn, App::timer_stop_fn);
    // timer.user_data = NON_NULL_PTR;
}

    if (IS_ENABLED(CONFIG_NATIVE_POSIX_SLOWDOWN_TO_REAL_TIME))
    {
        startZephyrEpochTimer(App::getEpochIntervalMs()/2);
    }
    else
    {
        startZephyrEpochTimer(App::getEpochIntervalMs() * 500);
    }

void App::startZephyrEpochTimer(unsigned msec)
{
    // The timer’s duration and period parameters specify minimum delays
    // that will elapse. Because of internal system timer precision (and
    // potentially runtime interactions like interrupt delay) it is possible
    // that more time may have passed as measured by reads from the relevant
    // system time APIs. But at least this much time is guaranteed to have elapsed.

    // Start periodic timer that expires per msecond specified
    printk("Starting up Epoch Timer for period %d ms\n", msec);
    k_timer_start(&m_timer, K_MSEC(msec), K_MSEC(msec));
}

void App::timer_expiry_fn(struct k_timer *timer_id)
{
    static timeval tvLast = {0, 0};

    timeval tvNow;
    TimeUtils::getTime(&tvNow);

    unsigned nowEpoch = timevalToEpoch(tvNow.tv_usec, App::getEpochIntervalUsec());
    timeval tvElapsed = tvNow - tvLast;

    printk("timer_expiry_fn[Epoch %d] invoked at (%ld/%ld), elapsed (%ld/%ld)\n",
           nowEpoch, tvNow.tv_sec, tvNow.tv_usec, tvElapsed.tv_sec, tvElapsed.tv_usec);

    tvLast = tvNow;
}
```

bob2oneil, 2023-05-31:
I have modified my application under Native-POSIX to utilize the real-time option (e.g. **CONFIG_NATIVE_POSIX_SLOWDOWN_TO_REAL_TIME** kernel configuration is set to a value of **y**).  I set the following other 2 kernel options such that Zephyr emulation matches the Linux host platform (**CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC** and **CONFIG_SYS_CLOCK_TICKS_PER_SEC** both set to 1000000).  

This is verified in part by having a Zephyr periodic timer set to a duration of 10 ms, and using the Linux host platform _clock_gettime()_, match the elapsed time for the periodic timer. It is fairly close to 10 ms.  While I am calling the Zephyr  POSIX API  clock_gettime(), it appears that this is linking via the c runtime to host platform's **clock_gettime()** rather than the Zephyr implementation based on k_uptime_ticks().  This does not work badly for me as I need to handle data at 10 ms real-time epochs as well as triggering activities at 1 PPS wall clock time.

In general, these changes resulted in significant improvement to the behavior of the Linux POSIX standard select() API (**zephyr_select** in Zephyr).  This is shown in the log at the bottom of this message where the thread remains pending consistent with the input timeout value for the cases where the exit is not triggered by a file input/output need.

One problem remains.  

I about 10 percent of the times, when my platform is under load with data flowing between 2 virtual Ethernet interfaces, the Zephyr RTOS does not return control to my main thread within the maximum timeout value of 10 ms epochs. Instead, there are times where the main thread (with real-time priority of -2) does not awaken by Zephyr until after 25 or more milliseconds.  This result in my application _not meeting its real-time deadlines_.

Is there any explanation for this or anything that I might explore more closely to understand this behavior, that is Zephyr consistently meeting its real time deadlines under Native-POSIX emulation, or is that just the way it is for this environment?

The following log information displays the test results, where what is being logged is the exit from the **zephyr_select** statement, showing the input timeout value, and the elapsed time between zephyr_select entry and exit as the elapsed time.  The last few lines of this log at the bottom are the error statements where the elapsed time of 24799 usec (about 25 ms) exceeds the input timeout values specified as around 10 ms, and a missed epoch is detected.

```
select() OUT[epoch 89]: now (1685475174/899929), timeout (0/148), tvNextEpoch (1685475174/900000), elapsed (0/77), count 0
select() OUT[epoch 89]: now (1685475174/900013), timeout (0/58), tvNextEpoch (1685475174/900000), elapsed (0/71), count 0
Post select: past next epoch: TDMA epoch 89 at now (1685475174/900013), timeout (0/58), next epoch (1685475174/900000)
select() OUT[epoch 90]: now (1685475174/900049), timeout (0/9953), tvNextEpoch (1685475174/910000), elapsed (0/2), count 1
select() OUT[epoch 90]: now (1685475174/909966), timeout (0/9931), tvNextEpoch (1685475174/910000), elapsed (0/9897), count 0
select() OUT[epoch 90]: now (1685475174/910115), timeout (0/13), tvNextEpoch (1685475174/910000), elapsed (0/128), count 0
Post select: past next epoch: TDMA epoch 90 at now (1685475174/910115), timeout (0/13), next epoch (1685475174/910000)
select() OUT[epoch 91]: now (1685475174/910148), timeout (0/9853), tvNextEpoch (1685475174/920000), elapsed (0/1), count 1
select() OUT[epoch 91]: now (1685475174/919788), timeout (0/9832), tvNextEpoch (1685475174/920000), elapsed (0/9620), count 0
select() OUT[epoch 91]: now (1685475174/919986), timeout (0/189), tvNextEpoch (1685475174/920000), elapsed (0/175), count 0
select() OUT[epoch 92]: now (1685475174/929986), timeout (0/9996), tvNextEpoch (1685475174/930000), elapsed (0/9982), count 0
select() OUT[epoch 93]: now (1685475174/930012), timeout (0/9989), tvNextEpoch (1685475174/940000), elapsed (0/1), count 1
select() OUT[epoch 93]: now (1685475174/930034), timeout (0/9967), tvNextEpoch (1685475174/940000), elapsed (0/1), count 1
select() OUT[epoch 93]: now (1685475174/939985), timeout (0/9952), tvNextEpoch (1685475174/940000), elapsed (0/9937), count 0
select() OUT[epoch 94]: now (1685475174/949924), timeout (0/9988), tvNextEpoch (1685475174/950000), elapsed (0/9912), count 0
select() OUT[epoch 94]: now (1685475174/949949), timeout (0/53), tvNextEpoch (1685475174/950000), elapsed (0/2), count 1
select() OUT[epoch 94]: now (1685475174/950149), timeout (0/29), tvNextEpoch (1685475174/950000), elapsed (0/178), count 0
Post select: past next epoch: TDMA epoch 94 at now (1685475174/950149), timeout (0/29), next epoch (1685475174/950000)
select() OUT[epoch 95]: now (1685475174/950185), timeout (0/9817), tvNextEpoch (1685475174/960000), elapsed (0/2), count 1
select() OUT[epoch 95]: now (1685475174/959777), timeout (0/9798), tvNextEpoch (1685475174/960000), elapsed (0/9575), count 0
select() OUT[epoch 95]: now (1685475174/959925), timeout (0/203), tvNextEpoch (1685475174/960000), elapsed (0/128), count 0
select() OUT[epoch 95]: now (1685475174/959976), timeout (0/57), tvNextEpoch (1685475174/960000), elapsed (0/33), count 0
select() OUT[epoch 95]: now (1685475174/960018), timeout (0/11), tvNextEpoch (1685475174/960000), elapsed (0/29), count 0
Post select: past next epoch: TDMA epoch 95 at now (1685475174/960018), timeout (0/11), next epoch (1685475174/960000)
[... cut ...] 1
select() OUT[epoch 96]: now (1685475174/969985), timeout (0/9931), tvNextEpoch (1685475174/970000), elapsed (0/9916), count 0
select() OUT[epoch 97]: now (1685475174/994811), timeout (0/9988), tvNextEpoch (1685475174/980000), elapsed (0/24799), count 0
Post select: past next epoch: TDMA epoch 97 at now (1685475174/994811), timeout (0/9988), next epoch (1685475174/980000)
EPOCH timeout: 1 missed epoch(s) detected: epoch timeout called at (1685475174/994811), current/prev/epoch counter 99/97/404

aescolar, 2023-05-31:
Bug and priority labels removed, as this is not a bug but a question.

aescolar, 2023-05-31:
@bob2oneil 
I'm sorry to tell you that you are, unfortunately, miss-understanding what native_posix is meant to do.
* native_posix is a test platform. It is not meant to give you that kind of real-timeness.
* native_posix is primarily designed to test Zephyr and its applications fast. The "real-time" mode just provides some alignment with the host clock, so as to allow you to interact with host peripherals at a high level, or interact manually with the Zephyr test target. But that is it.
* native_posix is not a Zephyr-Linux translation layer, and is **not** meant in any way to allow you to build and run hard realtime Linux applications, with a Zephyr API.
* native_posix only ensures a correct <u>simulated time</u>, and correct Zephyr scheduling inside that simulated time.
* Why you see what you see is because:
    * The Linux kernel is the one in charge of the host CPU scheduling. When the zephyr simulation (native_posix) is scheduled is fully up to it. All native_posix can do is slow itself down to real time.
    * If you set CONFIG_SYS_CLOCK_TICKS_PER_SEC so high, you are mostly just creating overhead. Any number that leads to a zephyr system tick timer resolution below 10-50µs is probably just making things worse.

That being said, if you are dead set in continuing this way, you may  want to look into increasing the priority of that executable ([nice](https://man7.org/linux/man-pages/man1/nice.1.html)). But you are shooting yourself in your feet.

bob2oneil, 2023-06-04:
Thanks for the explanation on this.  I will try making the created binary run at a higher priority to eliminate the thread yield issue and will decrease the tick resolution as well per your suggestions.

closed 2023-06-04 by bob2oneil