zephyrproject-rtos/zephyr #58040: SMP kernel (dead?)lock on k_thread_abort

evgeniy-paltsev, 2023-05-18:
**Describe the bug**
I've bumped into issue on SMP kernel and I was able to implement simple test case to reproduce it.

The test is pretty simple:

- The ztest thread has coop priority.
- From ztest thread we spawn N number of coop threads, where N = number of CPUs.
  - The spawned coop are executing infinite loop (so they occupy CPU core until they are aborted).
  - We have (number of CPUs - 1) spawned threads run and executing infinite loop, as current CPU is occupied by ztest coop thread. Due to that the last of spawned threads is ready but not executing.
- We abort spawned threads one-by-one from the ztest thread. 

We expect that all spawned threads will be aborted, however we abort only one spawned thread.
The system will infinitely execute (number of CPUs - 1) spawned threads on (number of CPUs - 1) CPU cores and `idle` thread on the rest CPU core.

I didn't find any hint in documentation which can justify such `k_thread_abort` behavior, hence I treat this as a bug.

The full test case can be found in https://github.com/zephyrproject-rtos/zephyr/pull/58037
The simplified test case code is: 

```c
static void thread_ab_entry(void *p1, void *p2, void *p3)
{
	printk("coop thread on cpu %d\n", arch_curr_cpu()->id);

	/* infinite loop here */
	while (true) { }
}

#define SPAWN_AB_PRIO K_PRIO_COOP(10)

ZTEST(smp, test_coop_abort)
{
	k_tid_t tid[MAX_NUM_THREADS];
	unsigned int i, num_threads = arch_num_cpus();

	/* ZTEST thread (this thread) has coop priority */
	printk("started test case on cpu %d, main prio %d, spawn prio %d, idle prio %d\n",
	       arch_curr_cpu()->id, _current->base.prio, SPAWN_AB_PRIO, K_IDLE_PRIO);

	/* Spawn N number of coop threads, where N = number of CPUs */
	for (i = 0; i < num_threads; i++) {
		tid[i] = k_thread_create(&tthread[i], tstack[i], STACK_SIZE, thread_ab_entry,
					 NULL, NULL, NULL, SPAWN_AB_PRIO, 0, K_NO_WAIT);
	}

	/* Wait for some time to let spawned threads on other cores run, do print and start
	 * executing infinite loop */
	k_busy_wait(DELAY_US * 4);

	/* At this time we have (number of CPUs - 1) spawned threads run and executing infinite loop
	 * on other CPU cores, as current CPU is occupied by this coop thread.
	 * Due to that the last of spawned threads is ready but not executing */

	/* Abort all spawned threads one-by-one */
	for (i = 0; i < num_threads; i++) {
		printk("try to abort thread %d from cpu %d\n", i, arch_curr_cpu()->id);
		k_thread_abort(tid[i]);
	}

	/* If issue reproduces we never get here */
}
```

**Platforms**
I was able to reproduce it on ARC SMP platforms (`nsim_hs_smp`, `nsim_hs5x_smp`), ARM SMP QEMU (`qemu_cortex_a53_smp`), x86 SMP QEMU (`qemu_x86_64`).

Test log on `nsim_hs_smp`
```
*** Booting Zephyr OS build zephyr-v3.2.0-8593-g602178d28db9 ***
Running TESTSUITE smp
===================================================================
START - test_coop_abort
started test case on cpu 0, main prio -1, spawn prio -6, idle prio 15
coop thread on cpu 1
try to abort thread 0 from cpu 0
coop thread on cpu 0
```

Test log on `qemu_cortex_a53_smp`
```
*** Booting Zephyr OS build zephyr-v3.2.0-8593-g602178d28db9 ***
Secondary CPU core 1 (MPID:0x1) is up
Secondary CPU core 2 (MPID:0x2) is up
Secondary CPU core 3 (MPID:0x3) is up
Running TESTSUITE smp
===================================================================
START - test_coop_abort
started test case on cpu 0, main prio -1, spawn prio -6, idle prio 15
coop thread on cpu 1
coop thread on cpu 2
coop thread on cpu 3
try to abort thread 0 from cpu 0
coop thread on cpu 0
```

**To Reproduce**

To simplify test case reproducion it's integrated to `tests/kernel/smp` test. As it's reproducable on the SMP  platforms available in upstream CI we can find failures here: https://github.com/zephyrproject-rtos/zephyr/pull/58037

To reproduce manually you can just run

```
west build -b qemu_cortex_a53_smp tests/kernel/smp -T kernel.multiprocessing.smp
ninja -C build -v run
```

using sources from the branch provided in https://github.com/zephyrproject-rtos/zephyr/pull/58037

**Expected behavior**
The test case should finish and all spawned threads should be aborted.

**Impact**
The issue causes system infinite lock.

**Environment (please complete the following information):**
 - Toolchain: Zephyr SDK, any latest version. Locally tested with 0.15.2.
 - Version used: test in https://github.com/zephyrproject-rtos/zephyr/pull/58037 based on 2bcfd4f7e373ea6ec35b1fed3121e5a3282345e0 commit. The changes are only done to test for issue reproducion.

andyross, 2023-05-19:
Confirmed.  I have this in an isolated rig now, and it's real.  Seems to be something related to thread state, the thread that calls k_thread_abort() is being woken up correctly once the target thread on the other CPU aborts, but the scheduler doesn't choose it to run from within the resulting IPI.

evgeniy-paltsev, 2023-05-19:
@andyross 

I was able to reproduce this with preempt threads.

As I observe the key point to reproduce the issue is to have reschedule happen during the `k_thread_abort` call from the main / ztest thread.

To reproduce it with preempt threads we have to select both main and spawn thread have same preempt priority.

Example to reproduce:

`main.c`:

```c
#include <zephyr/kernel.h>
#include <zephyr/kernel_structs.h>

#ifndef CONFIG_TEST_EXTRA_STACK_SIZE
#define CONFIG_TEST_EXTRA_STACK_SIZE 1024
#endif

#define MAX_NUM_THREADS CONFIG_MP_MAX_NUM_CPUS
#define DELAY_US 50000
#define STACK_SIZE (384 + CONFIG_TEST_EXTRA_STACK_SIZE)

static struct k_thread tthread[MAX_NUM_THREADS];
static K_THREAD_STACK_ARRAY_DEFINE(tstack, MAX_NUM_THREADS, STACK_SIZE);

#define SPAWN_AB_PRIO CONFIG_MAIN_THREAD_PRIORITY

static void thread_ab_entry(void *p1, void *p2, void *p3)
{
	ARG_UNUSED(p1);
	ARG_UNUSED(p2);
	ARG_UNUSED(p3);

	printk("%s thread on cpu %d\n", SPAWN_AB_PRIO < 0 ? "coop" : "preempt", arch_curr_cpu()->id);

	while (true) {
	}
}

int main(void)
{
	k_tid_t tid[MAX_NUM_THREADS];
	unsigned int num_threads = arch_num_cpus();
	unsigned int i;

	printk("started test case on cpu %d, main prio %d, spawn prio %d, idle prio %d\n",
	       arch_curr_cpu()->id, _current->base.prio, SPAWN_AB_PRIO, K_IDLE_PRIO);

	/* Spawn N number of coop threads, where N = number of CPUs */
	for (i = 0; i < num_threads; i++) {
		tid[i] = k_thread_create(&tthread[i], tstack[i],
					 STACK_SIZE, thread_ab_entry,
					 NULL, NULL, NULL,
					 SPAWN_AB_PRIO, 0, K_NO_WAIT);
	}

	printk("waiting on %d\n", arch_curr_cpu()->id);

	/* Wait for some time to let spawned threads on other cores run, do print and start
	 * executing infinite loop.
	 */
	k_busy_wait(DELAY_US * 4);

	/* At this time we have (number of CPUs - 1) spawned threads run and executing infinite loop
	 * on other CPU cores, as current CPU is occupied by this coop thread.
	 * Due to that the last of spawned threads is ready but not executing.
	 */

	/* Abort all spawned threads one-by-one */
	for (i = 0; i < num_threads; i++) {
		printk("try to abort thread %d from cpu %d\n", i, arch_curr_cpu()->id);
		k_thread_abort(tid[i]);
	}

	for (i = 0; i < num_threads; i++) {
		printk("try to join %d\n", i);
		k_thread_join(tid[i], K_FOREVER);
	}

	return 0;
}
```

`prj.conf`:
```
CONFIG_MAIN_THREAD_PRIORITY=5
```

andyross, 2023-05-19:
Turns out to be fairly simple really.  Just a logic bug, not a race or other weirdness.  The selection of a candidate thread to run after an interrupt was done BEFORE the "current thread was aborted asynchronously" check, so it wouldn't select a joined thread that would have been woken up by the abort.

PR #58089 (2023-05-22): kernel/sched: Fix thread selection misordering with aborted threads
When a running thread gets aborted asynchronously (this only happens in SMP contexts, obviously) it gets flagged "aborting", but the actual abort needs to happen in the thread's own context.  For convenience, this was done in the next_up() routine that selects the next thread to run at interrupt exit time.

But this check was being done AFTER the next candidate thread was selected from the run queue.  Thread abort can wake up threads blocked in k_thread_join(), and therefore these weren't seen as runable threads, even if they  [... cut]
commit d537267f (2023-05-22): kernel/sched: Fix thread selection misordering with aborted threads

When a running thread gets aborted asynchronously (this only happens
in SMP contexts, obviously) it gets flagged "aborting", but the actual
abort needs to happen in the thread's own context.  For convenience,
this was done in the next_up() routine that selects the next thread to
run at interrupt exit time.

But this check was being done AFTER the next candidate thread was
selected from the run queue.  Thread abort can wake up threads blocked
in k_thread_join(), and therefore these weren't seen as runable
threads, even if they [... cut]
closed 2023-05-22 by fabiobaltieri