zephyrproject-rtos/zephyr #58057: up_squared: sample: gpio_counter - is failing

KamilxPaszkiet, 2023-05-19:
**Describe the bug**
```
DEBUG   - DEVICE: Succeed to download NBP file.
DEBUG   - DEVICE: *** Zephyr EFI Loader ***
DEBUG   - DEVICE: RSDP found at 0x79c34000
DEBUG   - DEVICE: Zeroing 4816352 bytes of memory at 0x106000
DEBUG   - DEVICE: Copying 32768 data bytes to 0x1000 from image offset
DEBUG   - DEVICE: Copying 24576 data bytes to 0x100000 from image offset 32768
DEBUG   - DEVICE: Copying 2109984 data bytes to 0x59dde0 from image offset 57344
DEBUG   - DEVICE: Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.3.0-2743-g3f06e5ac6069 ***
DEBUG   - DEVICE: ERROR: cannot set HAT pin 35 to OUT (-22)
```
The README contains BIOS settings for this test. I looked in the bios and checked the settings, everything looks fine...

**To Reproduce**
Steps to reproduce the behavior:
```
twister --device-testing --hardware-map $HOME/up2_double.yaml -T samples/boards/up_squared/gpio_counter/ --no-skipped-report -vv
```
or
```
west build -p -b up_squared -T samples/boards/up_squared/gpio_counter/
```
Expected behavior
The test should pass:
```
*** Booting Zephyr OS build zephyr-v2.6.0-2360-gc94e53c9f5b6 ***
counter: 0x0
counter: 0x1
counter: 0x2
counter: 0x3
counter: 0x4
counter: 0x5
counter: 0x6
counter: 0x7
counter: 0x8
counter: 0x9
counter: 0xa
counter: 0xc
counter: 0xd
counter: 0xe
counter: 0xf
```

- OS: Ubuntu for the controller
- Toolchain: Zephyr SDK 15.0
- Latest mainline [2be2fa8](https://github.com/zephyrproject-rtos/zephyr/commit/2be2fa8bd618f616f0b6aea8ecb22249766e52ea)

nashif, 2023-10-25:
@KamilxPaszkiet is the bios set for this test per the requirements?

```

Requirements
************

The application requires an UP Squared board connected to the PC through USB
for serial console. The BIOS settings must be updated as specified in the
source code comments for HAT Configurations (see table above).

* Note:
 * Need to change the BIOS settings:
 * () Advanced -> HAT Configurations:
 *    - HD-Audio / I2S6 Selec -> Disabled
 *    - GPIO / PWM3 Selection -> GPIO
 *    - GPIO / I2S2 Selection -> GPIO
 *
 *    - GPIO 19 (Pin16) Confi -> Input
 *
 *    - GPIO 14 (Pin35) Confi -> Output
 *    - GPIO 15 (Pin37) Confi -> Output
 *    - GPIO 27 (Pin38) Confi -> Output
 *    - GPIO 28 (Pin40) Confi -> Output

```

KamilxPaszkiet, 2023-10-25:
> @KamilxPaszkiet is the bios set for this test per the requirements?
>
> ```
>
> Requirements
> ************
>
> The application requires an UP Squared board connected to the PC through USB
> for serial console. [...]

Yes, ofc.
[image]
[image]
[image]

But output is slightly different than in May:
```
DEBUG   - DEVICE: >>Checking Media Presence......
DEBUG   - DEVICE: >>Media Present......
DEBUG   - DEVICE: Downloading NBP file...
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Succeed to download NBP file.
DEBUG   - DEVICE: *** Zephyr EFI Loader ***
DEBUG   - DEVICE: RSDP found at 0x79c34000
DEBUG   - DEVICE: Zeroing 4870224 bytes of memory at 0x108000
DEBUG   - DEVICE: Copying 32768 data bytes to 0x1000 from image offset
DEBUG   - DEVICE: Copying 32768 data bytes to 0x100000 from image offset 32768
DEBUG   - DEVICE: Copying 2264992 data bytes to 0x5ad060 from image offset 65536
DEBUG   - DEVICE: Jumping to Entry Point: 0x1137 (48 31 c0 48 31 d2 48)
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.5.0-467-gdc431a7397a6 ***
DEBUG   - DEVICE: ERROR: cannot set HAT pin 40 to OUT (-22)
```

jhedberg, 2023-11-22:
I am able to reproduce this with the same `cannot set HAT pin 40 to OUT (-22)` error. The "check_perms()" step in the `gpio_intel.c` driver is failing, i.e. the HW is claiming that the host OS doesn't have permissions to change the specific pin. I think it's likely that this is simply some firmware/BIOS issue, i.e. something has changed in more recent versions, which probably requires different settings, or perhaps updating the sample to use different pins. I don't have any bandwidth to work on this any further however, so the options are to either remove this sample (or possibly mark GPIO as unsupported on APL) or to just close this as not planned.

KamilxPaszkiet, 2023-11-23:
> I am able to reproduce this with the same `cannot set HAT pin 40 to OUT (-22)` error. The "check_perms()" step in the `gpio_intel.c` driver is failing, i.e. the HW is claiming that the host OS [...]

@nashif What do you think?

jhedberg, 2024-04-07:
@KamilxPaszkiet I don't have access to this hardware anymore, so someone else would need to pick this up.

jhedberg, 2024-04-07:
> @KamilxPaszkiet I don't have access to this hardware anymore, so someone else would need to pick this up.

Also see https://github.com/zephyrproject-rtos/zephyr/issues/58057#issuecomment-1822492625 - it doesn't look like this is a bug in a driver or board definition, rather it's BIOS or test case related.

PR #71326 (2024-04-19): samples: gpio_counter: remove obsolete sample
Board specific sample that only works with certain revisions of the
hardware requiring special BIOS settings.

Fixes #58057
commit 132cbdf2 (2024-04-19): samples: gpio_counter: remove obsolete sample

Board specific sample that only works with certain revisions of the
hardware requiring special BIOS settings.

Fixes #58057
closed 2024-04-19 by fabiobaltieri