zephyrproject-rtos/zephyr #58064: latency_measure: Error occurs when setting CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=10000000 on Cortex-R82 class.

zhang10609, 2023-05-19:
I'm trying to do the latency measurement for my FPGA which is based on Cortex-R82 class and the clock frequency is indeed is 10MHz, so I set the parameter as below.
`CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=10000000`	
And the full log can be found as below.
START - Time Measurement
Timing results: Clock frequency: 10 MHz
Average thread context switch using yield                   :    1142 cycles ,   114225 ns
Error: tick occurred
Switch from ISR back to interrupted thread                  :     484 cycles ,    48400 ns
Time from ISR to executing a different thread               :     869 cycles ,    86900 ns
Time to create a thread (without start)                     :     349 cycles ,    34900 ns
Time to start a thread                                      :    1398 cycles ,   139800 ns
Time to suspend a thread                                    :    1303 cycles ,   130300 ns
Time to resume a thread                                     :    1210 cycles ,   121000 ns
Time to abort a thread (not running)                        :     253 cycles ,    25300 ns
Average semaphore signal time                               :     408 cycles ,    40806 ns
Average semaphore test time                                 :     156 cycles ,    15633 ns
Semaphore take time (context switch)                        :    1411 cycles ,   141100 ns
Semaphore give time (context switch)                        :    1839 cycles ,   183900 ns
Average time to lock a mutex                                :     276 cycles ,    27657 ns
Average time to unlock a mutex                              :     105 cycles ,    10539 ns
Average time for heap malloc                                :    1118 cycles ,   111839 ns
Average time for heap free                                  :     725 cycles ,    72505 ns
===================================================================
PROJECT EXECUTION FAILED

I had debugged with this log and found that the error occurs at the line as below in bench_test_end() function. 

`if (timestamp_check >= MSEC_PER_SEC) {
return -1;
`}

I had printed the frequency from the register CNTFRQ_EL0 and found it is indeed 10MHz.
I had found my case hit the error because timestamp_check == 1000 for my platform. I don't know why this value(1000ms) should be considered as an error. 
Is this expected that it will trigger an error if the value of timestamp_check is 1000ms?

peter-mitsis, 2023-06-07:
This test, when it was written, was not expecting the coop_ctx_switch() test to take longer than one second to execute. Why one second? That is the lowest system tick frequency that can be set (CONFIG_SYS_CLOCK_TICKS_PER_SEC=1). The aim is to eliminate skew caused by processing ticks while determining benchmark results.

Additional info.
As it is currently written, the test expects to perform about 20000 yields / context switches.  Or to phrase it differently, this test is expecting that a context switch should not take longer than 50 usec--however, based on the numbers you showed, this is looking to be more around 115 usec.

Given all of this, I think that this test can be enhanced. Rather than *just* reporting the tick error, I think we ought to add a "Notes" column to the output. That is, we always report the numbers as best we can, but add some notes to provide context why it failed.

zhang10609, 2023-07-16:
@peter-mitsis Thanks peter for the fix, sorry for the delay.

PR #59070 (2023-07-19): Update benchmark latency_measure test reporting
Updates the benchmark latency_measure test reporting so that ... 

1. The test description is always shown--even when it fails to make it even easier to identify which test failed.
2. Tests that report average results that take longer than one second (a scenario for a guaranteed system tick processing) will still register as a failure, but they will now display both the calculated results and a note indicating what went "wrong". There is no need to hide the results in this case.

Fixes #58064
closed 2023-07-16 by zhang10609