zephyrproject-rtos/zephyr #58074: samples: board: mec172xevb_assy6906 failed sample

golowanow, 2023-05-19:
**Describe the bug**
Incorrect sample definition at `samples/boards/mec172xevb_assy6906/rom_api/sample.yaml` causes twister to fail on missing regexp

https://github.com/zephyrproject-rtos/zephyr/blob/ce2a7d9a1ad5a0d355e878c294ed3c189f5e7a52/samples/boards/mec172xevb_assy6906/rom_api/sample.yaml#L7-L9

Introduced with [80777594a1cb](https://github.com/zephyrproject-rtos/zephyr/commit/80777594a1cb), #57511 

**To Reproduce**
`twister -p mec172xevb_assy6906 --device-testing --device-serial /dev/ttyUSB0 -T samples/boards/mec172xevb_assy6906/rom_api  -x=CONFIG_BOOT_DELAY=500 -vv --no-skipped-report`

**Expected behavior**
sample passed 

**Impact**
causes twister execution to fail

**Logs and console output**
```
INFO    - Zephyr version: zephyr-v3.3.0-4092-g6697c5aa0bb2
........... 
DEBUG   - build test: mec172xevb_assy6906/samples/boards/mec172xevb_assy6906/rom_api/sample.boards.mec172xevb_assy6906.rom_api
DEBUG   - Building /home/ztest/zephyrproject/zephyr/samples/boards/mec172xevb_assy6906/rom_api for mec172xevb_assy6906
DEBUG   - Determine test cases for test instance: mec172xevb_assy6906/samples/boards/mec172xevb_assy6906/rom_api/sample.boards.mec172xevb_assy6906.rom_api
DEBUG   - Determine test cases for test suite: sample.boards.mec172xevb_assy6906.rom_api
DEBUG   - Test instance mec172xevb_assy6906/samples/boards/mec172xevb_assy6906/rom_api/sample.boards.mec172xevb_assy6906.rom_api already has 1 cases.
DEBUG   - run test: mec172xevb_assy6906/samples/boards/mec172xevb_assy6906/rom_api/sample.boards.mec172xevb_assy6906.rom_api
DEBUG   - Using serial device /dev/ttyUSB0 @ 115200 baud
Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ztest/zephyrproject/zephyr/scripts/pylib/twister/twisterlib/runner.py", line 1203, in pipeline_mgr
    pb.process(pipeline, done_queue, task, lock, results)
  File "/home/ztest/zephyrproject/zephyr/scripts/pylib/twister/twisterlib/runner.py", line 632, in process
    self.run()
  File "/home/ztest/zephyrproject/zephyr/scripts/pylib/twister/twisterlib/runner.py", line 1021, in run
    instance.handler.handle()
  File "/home/ztest/zephyrproject/zephyr/scripts/pylib/twister/twisterlib/handlers.py", line 581, in handle
    harness.configure(self.instance)
  File "/home/ztest/zephyrproject/zephyr/scripts/pylib/twister/twisterlib/harness.py", line 105, in configure
    self.pattern = re.compile(self.regex[0])
IndexError: list index out of range
```

golowanow, 2023-05-19:
@Manimaran-A

Manimaran-A, 2023-05-19:
I check and update the sample.yaml file

Manimaran-A, 2023-05-23:
Created PR #58178, in that have updated sample.yaml regex pattern to fix twister error.

PR #58178 (2023-05-26): sample: crypto: mchp: sample.yaml Regex update
Updated the sample.yaml file with Console message Regex pattern to fix Twister error.

Fix: #58074
commit b6555ee3 (2023-05-26): sample: crypto: mchp: Regex pattern updated in yaml file

Updated the sample.yaml file with Console message Regex
pattern to fix Twister error.

Fix: #58074
closed 2023-05-26 by nashif