zephyrproject-rtos/zephyr #58098: Audio framework that can be used by MCU

hongshui3000, 2023-05-20:
Zephyr has been continuously developing ble audio, but lacks a complete audio framework (SOF is good, but it only supports the xtensa architecture, and it is also very complex and huge). I think it's time to consider developing or porting an audio framework for zephyr.

I think nxp maestro seems like a good choice(https://github.com/nxp-mcuxpresso/maestro). I don't know if the community has plans or ideas in this regard?

nashif, 2023-05-20:
SOF by moving to Zephyr should be able to support other architectures, not only xtensa. And there are already some efforts to do this. @lgirdwood

dbaluta, 2023-05-22:
@hongshui3000 SOF supports arm64 with imx93. Also, because now SOF can run as an app inside Zephyr it should work on all architectures Zephyr works. 

Can you sketch some requirements you have for an audio framework?

hongshui3000, 2023-05-23:
@dbaluta Some single-chip microcomputers have relatively simple needs, that is, to play some music (such as BLE AUDIO), it has some simple dsp instructions, and its memory is not very large. Such as Cortex M4 and Cortex M33 some single-core MCU. At present, SOF is running on some relatively large processors (relatively large memory, relatively strong dsp instructions, and relatively fast processor speed), and I am also using SOF. But when I try to migrate to a weaker mcu, I find it not easy.

It would be great if SOF can support more architectures and do some optimizations for these, such as memory and other resources

hongshui3000, 2024-01-26:
> Hello,
> 
> @hongshui3000 We are glad you mentioned Audio maestro framework. https://github.com/nxp-mcuxpresso/maestro
> 
> We are porting it on Zephyr and you can expect it on Zephyr this year.
> 
> Is there any specific use case you would like to use it with or any specific platform you would like to use?

Great to hear this. What needs attention is the software license issue. If it can be successfully used on zephyr, it should be possible to extend the scope of audio use to ARM's cortex-M mcu (SOF+zephyr currently only supports the xtensa architecture)

Radimli, 2024-01-26:
> > Hello,
> > @hongshui3000 We are glad you mentioned Audio maestro framework. https://github.com/nxp-mcuxpresso/maestro
> > We are porting it on Zephyr and you can expect it on Zephyr this year.
> > Is there any specific use case you would like to use it with or any specific platform you would like to use?
> 
> Great to hear this. What needs attention is the software license issue（Some projects may not fully use NXP's mcu）. If it can be successfully used on zephyr, it should be possible to extend the scope of audio use to ARM's cortex-M mcu (SOF+zephyr currently only supports the xtensa architecture)

Thank you for notice, we are taking it into consideration. The core of Maestro is and will be under BSD 3. That means - it will be open to any Cortex-M MCU. We are still discussing which elements will be supported, in what form and with what license though.

Are you interested in any specific elements? List of elements:
https://github.com/nxp-mcuxpresso/maestro/raw/main/doxygen/common/maestroApp.svg

dbaluta, 2024-01-26:
> Great to hear this. What needs attention is the software license issue（Some projects may not fully use NXP's mcu）. If it can be successfully used on zephyr, it should be possible to extend the scope [...]

Please mind that SOF + Zephyr is now ported to imx93 and SOF runs on a55 core.

hongshui3000, 2024-01-29:
> > > Hello,
> > > @hongshui3000 We are glad you mentioned Audio maestro framework. https://github.com/nxp-mcuxpresso/maestro
> > > We are porting it on Zephyr and you can expect it on Zephyr this year.
> > > Is there any specific use case you would like to use it with or any specific platform you would like to use?
> > 
> > 
> > Great to hear this. What needs attention is the software license issue（Some projects may not fully use NXP's mcu）. If it can be successfully used on zephyr, it should be possible to extend the scope of audio use to ARM's cortex-M mcu (SOF+zephyr currently only supports the xtensa architecture)
> 
> Thank you for notice, we are taking it into consideration. The core of Maestro is and will be under BSD 3. That means - it will be open to any Cortex-M MCU. We are still discussing which elements will be supported, in what form and with what license though.
> 
> Are you interested in any specific elements? Are there any elements you would like maestro to support? List of elements: https://github.com/nxp-mcuxpresso/maestro/raw/main/doxygen/common/maestroApp.svg

I am more interested in the following part
[image]

As for usb audio and file systems, I usually use zephyr, as well as codec and other audio algorithms. I usually use privately customized algorithms.

hongshui3000, 2024-01-29:
> Please mind that SOF + Zephyr is now ported to imx93 and SOF runs on a55 core.

Good news, I will try it on cortex-m. My application has strict requirements on the amount of ram and rom.

ajkontra, 2024-07-13:
> Hello,
> 
> @hongshui3000 We are glad you mentioned Audio maestro framework. https://github.com/nxp-mcuxpresso/maestro
> 
> We are porting it on Zephyr and you can expect it on Zephyr this year.
> 
> Is there any specific use case you would like to use it with or any specific platform you would like to use?

Hello @Radimli. Is there an update available on the progress here? I am specifically looking to use the decoding libraries in the Maestro framework.

Radimli, 2024-07-16:
Hello @ajkontra,
it is in progress. Please, stay tuned.
Is there any specific pipeline you would like to build?

ajkontra, 2024-07-22:
> Hello @ajkontra, it is in progress. Please, stay tuned. Is there any specific pipeline you would like to build?

The pipeline would be STREAM_PIPELINE_MEM to line out.

gWacey, 2024-09-04:
Hello,

Excuse me from jumping in on this thread, but I have a couple of questions on Maestro in regard to porting to Zephyr as an MCU audio system:

1. Do you have a schedule for when this will be ported to Zephyr?
2. Can two streams be merged into a single stream (e.g. two different decoders output mixed into one stream for output)?
3. Is there a resource figure for the minimum stream (e.g. I2S input-to-I2S output or USB in-to-I2S out)?
4. Will any voice and/or post-processing modules be included in the port?
5. Is it possible to dynamically construct a stream(s)?

Thanks in advance for any comments you can give.

Radimli, 2024-09-04:
> Hello,
>
> Excuse me from jumping in on this thread, but I have a couple of questions on Maestro in regard to porting to Zephyr as an MCU audio system:
>
> 1. Do you have a schedule for when this will be [...]

Hello Graham,

thank you for your interest. Here are the answers:

1. Expect end of this year or start of the next one. 
2. Currently no
3. Currently no
4. Yes, module for voice command recognition should be supported in the first release. It will be not part of Maestro itself though
5. Yes

I'd like to ask you, do you have any use-cases you would like to use it with?
Do you have any specific requirements?

Thank you

gWacey, 2024-09-04:
Thanks @Radimli,
I am interested in general audio, but primarily how Zephyr will handle Bluetooth LE audio.
As discussed above a compact configurable audio framework would be very useful in Zephyr.

rhvarrier, 2024-10-17:
Is there any plan to add port Espressif's [esp-adf](https://github.com/espressif/esp-adf) as this seems to provide a well integrated solution for Esp SoCs for audio processing

iuliana-prodan, 2026-05-05:
> I’d really like to see a more complete audio framework in Zephyr for local music playback, including audio decoding, play/pause/resume, and next/previous track controls. I think this would be very useful for many embedded audio projects.

What’s your assessment of SOF? Have you evaluated it yet?
Please refer to this RFC, which proposes using SOF exclusively with Zephyr: https://github.com/zephyrproject-rtos/zephyr/issues/93876.

LibMP is also an option - https://github.com/zephyrproject-rtos/zephyr/pull/98514.

Based on your analysis, would these two provide a solid solution?