zephyrproject-rtos/zephyr #58099: ADXL372 Reset at initialization too long

vChavezB, 2023-05-20:
**Describe the bug**
I am using the adlx372 driver located here

https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/sensor/adxl372

I noticed that the driver has a delay of 1 second in the reset state here

https://github.com/zephyrproject-rtos/zephyr/blob/d01780fc9403166d250ded3d02372a7e1a9db4fd/drivers/sensor/adxl372/adxl372.c#L401

IS the 1 second necessary? I did not see this mentioned in the datasheet. As a comparison, Analog devices the company that makes this sensor uses a reset but is only 1 millisecond long

https://github.com/analogdevicesinc/no-OS/blob/f402d21b039036af5b7b77360d60f5cee96898ff/drivers/accel/adxl372/adxl372.c#L471

So my question here is the delay time correct or should it be 1 ms (similar to the manufacturer´s driver code)

nordicjm, 2023-05-22:
Not sure if the original author of that driver is still around @mhennerich

nordicjm, 2023-05-22:
@amiclaus can you check this?

amiclaus, 2023-05-22:
@nordicjm Indeed there is no delay specified in the datasheet as required after a software reset. Moreover, in the linux driver there is no delay added after this operation: https://github.com/torvalds/linux/blob/44c026a73be8038f03dbdeef028b642880cf1511/drivers/iio/accel/adxl372.c#L697

I will add a patch addressing this issue.

amiclaus, 2023-05-22:
Added a patch addressing this issue with: #58123

closed 2023-05-26 by nashif