zephyrproject-rtos/zephyr #58103: mcuboot/mcumgr: board not upgradable after "Not enough free space to run swap upgrade"

fabiobaltieri, 2023-05-20:
**Describe the bug**
Hi, bumped into this issue with an application setup with two partitions setup for swap upgrade, reproduced it on an nrf52dk but really it should apply to any board with this setup.

It appears that if an image large enough is loaded in slot2 (in my case it's 217820 bytes out of 220 KB, 96.69% used), mcuboot may fail to swap with something like:

```
W: Not enough free space to run swap upgrade
W: required 225280 bytes but only 221184 are available
```

and runs the previous image. This is expected and documented in https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/mcuboot/design.html#swap-without-using-scratch, the problem is that once the board enters this condition, it appears that there's no way to recover it with mcumgr:

```
$ mcumgr -c acm0 image list
Images:
 image=0 slot=0
    version: 0.0.0
    bootable: true
    flags: active confirmed
    hash: 13a9251314c31970101ec3ff4ca441f5c90d36f247644cc0cdc8e30b4c8bf6f1
 image=0 slot=1
    version: 0.0.0
    bootable: true
    flags: pending
    hash: ef1bcc89678e7c523bb25b78d10c946742adafa60694df40b26c35fc16e7d0de
Split status: N/A (0)
$ /home/fabiobaltieri/go/bin/mcumgr -c acm0 image erase
Error: 6
$ /home/fabiobaltieri/go/bin/mcumgr -c acm0 image upload build/zephyr/zephyr.signed.bin
 0 B / 213.04 KiB [-------------------------------------------------------]   0.00% <- stuck, no progress
$ mcumgr -c acm0 image test 13a9251314c31970101ec3ff4ca441f5c90d36f247644cc0cdc8e30b4c8bf6f1
Error: 6
```

So the board is effectively stuck in upgrade limbo as it looks like those calls checks the "upgrade pending" bit and errors out. The only way I've found to recover is to issue a "mcuboot erase 2" on the shell, if that's not available it's debugger time I guess.

This has apparently previously been reported in https://github.com/apache/mynewt-mcumgr/issues/157.

I think that one should be able to cancel the upgrade by erasing the second slot image. Is that something we can fix in the Zephyr fork or should be fixed upstream?

**To Reproduce**
Just try to upgrade a board configured for swap with a big enough image. In my tests I noticed images even bigger to trigger a "E: Image in the secondary slot is not valid!" and cause the bootloader to erase the second slot automatically, so this may only trigger on a range of sizes.

**Expected behavior**
Either mcuboot erases the second image when failing (seems to do it for other cases) or mcumgr allows erasing a pending image.

**Impact**
Seems pretty serious to me, may lead to devices lost and non upgradable in the field.

**Environment (please complete the following information):**
Linux, Zephyr SDK, d01780fc94

fabiobaltieri, 2023-05-20:
cc @d3zd3z @nordicjm @de-nordic

nordicjm, 2023-05-22:
From memory, I believe @de-nordic has been working on an improvement so that images that are too big to be built should error out during the build instead of generating a file that cannot actually be loaded.

PerMac, 2023-05-31:
@gchwier This looks like a good scenario to add to the mcuboot+mcumgr tests you're working on.

ShBiezeman, 2023-08-03:
Should this issue also cover the case that if a big image (remaining free partitions < 1)is loaded into slot number 1 it blocks firmware updates due to no swap partition avaible? Ran into this issue a couple days back and should be solvable by reserving a single partition chunk after the app region during compilation. MCUboot shouldn't reserve this chunk tho. 

Im willing to look more into this issue but have very limited knowledge within the partition management code.

tejlmand, 2023-11-08:
isn't this issue a duplicate of this https://github.com/zephyrproject-rtos/zephyr/issues/27471 ?

fabiobaltieri, 2023-11-08:
@tejlmand from what I see it isn't, this is about a specific range of sizes that is within the partition but beyond what mcumgr can use and mcumgr failing in a way that results in the device not upgradable anymore, which I think it's pretty bad.

PR #64586 (2023-12-12): sysbuild: Reduce maximum application size when MCUboot is enabled
This adds a new optional Kconfig which allows for reducing the available size of an image without needing to adjust the image slot size in dts. The use of this is for sysbuild whereby bootloaders can adjust the maximum size of an image if they e.g. store metadata at the end of an image which is added externally by a tool or outside source.

The initial implementation allows for a sysbuild-configured application which has MCUboot enabled to reduce the maximum image size to account for the TLV and swap data fields which are presen [... cut]
closed 2023-12-12 by carlescufi