zephyrproject-rtos/zephyr #58128: MCUmgr for zephyr 3.4 meta PR list

nordicjm, 2023-05-22:
MCUmgr PRs needed for zephyr 3.4:

- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/57933
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/57974
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/57976
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/57989
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/57991
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/57992
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/58030
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/58065
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/58126
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/58127
- [ ] https://github.com/zephyrproject-rtos/zephyr/pull/57921

nashif, 2023-06-01:
looks like all those PRs are merged, should this be closed?

nordicjm, 2023-06-05:
> looks like all those PRs are merged, should this be closed?

Was away, will close.

closed 2023-06-05 by nordicjm