zephyrproject-rtos/zephyr #58134: Distinguish Multiple boards attached with one CI machine using stm32cubeprogrammer

mithunpandey, 2023-05-22:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/58050

<div type='discussions-op-text'>

<sup>Originally posted by **mithunpandey** May 19, 2023</sup>
We have a CI setup which has multiple boards attached with the CI Agent Machine.
we are building for the following boards using west build
F429
F767
H743
and using west flash using openocd as runner , which is running fine.
**west flash --build-dir build --hex-file C:/app-to-build/build/zephyr/zephyr.hex -r openocd --skip-rebuild --cmd-pre-init "hla_serial 002F00154D4B500820373831"**

we have introduced **nucleo_u575zi_q** board, and flashing using openocd is failing, so we used "stm32cubeprogrammer" as a runner 

**We need to distinguish multiple boards while using west flash using the serial number of the board as we are doing with openocd, so the question is**

1. **is there a option available to do so with  stm32cubeprogrammer as a runner when using west flash command,**

</div>

erwango, 2023-05-22:
Please use recent openocd version (Open On-Chip Debugger 0.11.0+dev-00725-gc5c47943d (2023-03-14-22:08) for instance.

See following log:
```
Open On-Chip Debugger 0.11.0+dev-00725-gc5c47943d (2023-03-14-22:08)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
Info : STLINK V3J7M3 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.266932
Info : Unable to match requested speed 500 kHz, using 200 kHz
Info : Unable to match requested speed 500 kHz, using 200 kHz
Info : clock speed 200 kHz
Info : stlink_dap_op_connect(connect)
Info : SWD DPIDR 0x0be12477
Info : [STM32U575ZITxQ.cpu] Cortex-M33 r0p4 processor detected
Info : [STM32U575ZITxQ.cpu] target has 8 breakpoints, 4 watchpoints
Info : [STM32U575ZITxQ.cpu] external reset detected
Info : starting gdb server for STM32U575ZITxQ.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* STM32U575ZITxQ.cpu cortex_m   little STM32U575ZITxQ.cpu reset

Info : Unable to match requested speed 480 kHz, using 200 kHz
Info : Unable to match requested speed 480 kHz, using 200 kHz
CPU in Non-Secure state
target halted due to debug-request, current mode: Thread 
xPSR: 0xf9000000 pc: 0x080012c0 msp: 0x20001030
Info : Unable to match requested speed 4000 kHz, using 3300 kHz
Info : Unable to match requested speed 4000 kHz, using 3300 kHz
Info : device idcode = 0x20006482 (STM32U57/U58xx - Rev B : 0x2000)
Info : TZEN = 0 : TrustZone disabled by option bytes
Info : RDP level 0 (0xAA)
Info : flash size = 2048kbytes
Info : flash mode : dual-bank
Info : Padding image section 0 at 0x08003bc8 with 8 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x08003bd0 .. 0x08003fff
auto erase enabled
wrote 15312 bytes from file build/b_u585i_iot02a/zephyr/zephyr.hex in 0.207587s (72.033 KiB/s)
```

Note: It's not clear if this is the exact issue you're trying to solve as I don't see the relationship with your issue description "Distinguish Multiple boards attached with one CI machine using stm32cubeprogrammer"

erwango, 2023-05-22:
Note, to use stm32cubeprogrammer, see following configuration on lastest main:
https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/nucleo_u575zi_q/board.cmake

mithunpandey, 2023-05-23:
Will explain the use case:

we have multiple devices attached to the single machine (github agent) which are used by our CI pipelines to run tests.
now when using west flash with open ocd runner, we explicitly pass the device id using the **cmd-pre-init "hla_serial 002F00154D4B500820373831"** option , we need the same option to work with the **nucleo_u575zi_q** board.

erwango, 2023-05-23:
> Will explain the use case:
>
> we have multiple devices attached to the single machine (github agent) which are used by our CI pipelines to run tests. now when using west flash with open ocd runner, we [...]

Please have a check to my previous answers, this should be helping on both directions

erwango, 2023-10-12:
@mithunpandey Did you had a try with recent openocd, as proposed ?