zephyrproject-rtos/zephyr #58150: modules: SEGGER_RTT: Invalid re-use of RAM memory allocated for SEGGER RTT

MarekPieta, 2023-05-22:
**Describe the bug**
By default the RTT data structures are placed in bss memory region. After an application-triggered reboot, the RAM region used for the RTT data structures may be instantly re-used by the bootloader. The bootloader may use the RAM for completely different purpose, so that reading this region may eventually put the connected RTT host computer into strange state and lead to an unexpected behavior. 

On my desk, the host eventually wrote an invalid value to the `RdOff` (value of 215615629) of terminal’s up buffer after RTT initialization was completed on device’s end. The invalid value caused crash later on when RTT was used by the application (application accessed wrong memory address which triggered fault; the application uses SEGGER RTT for logging).

**Expected behavior**
There needs to be a way to inform the RTT connected host computer that the RTT data is no longer valid and should not be read (to prevent the RTT host from entering invalid state). Currently there seem to be no SEGGER API that would allow to uninit the RTT terminal so we may need to expand the API.
Alternatively we may provide a separate RAM region which could be used exclusively for the RTT Terminal. The region needs to be known for both application and bootloader (we may place the region at the end of device's RAM to mitigate chance that an incompatible bootloader would accidentally use it for a other purpose).

**Impact**
Unexpected crash of device working in debug mode. The crash may appear randomly and is hard to investigate.

**Additional context**
I checked the following with gdb to make sure about the root cause:

* If I set breakpoint on `main` function call, the garbage `RdOff` is already there (rest of the RTT structure is properly zero-initialized). Setting the `RdOff` manually to 0 using gdb's set successfully prevents the application's crash.
* If I set breakpoint on `SEGGER_RTT_Init` the spurious set of `RdOff` seems to happen before the RTT structure initialization. In that case, the `SEGGER_RTT_Init` properly initializes the structure after the spurious write which prevents the issue from happening.

nordicjm, 2023-05-22:
There is already a PR that fixes this: https://github.com/zephyrproject-rtos/zephyr/pull/53569

nordicjm, 2023-05-23:
Though @MarekPieta if you want this fix in for 3.4, might be worth picking it up and fixing the outstanding issues as the original author has gone silent and there's only a few days to get something merged.

brycewilkins, 2024-11-15:
@MarekPieta @nordicjm This issue has been solved by #53569 and can be closed.

It's unfortunate the west.yml was not updated in #53569 to bring in the corresponding Segger repo changes (https://github.com/zephyrproject-rtos/segger/pull/17); it was just missed I believe. I will try to bring in the change in #81425

closed 2024-11-15 by nordicjm