zephyrproject-rtos/zephyr #58159: flash_area has no API to get flash block size

vmsh0, 2023-05-22:
**Is your enhancement proposal related to a problem? Please describe.**
I'm using the flash_area API to write on a device where the driver has write block size < flash block size. But it seems like nonetheless the driver doesn't perform the necessary read-update-erase-write by itself, instead corrupting the flash pages.

Thus, I'm am using the flash_area_read, flash_area_erase, and flash_area_write APIs to read the current page contents to a buffer, change it, erase the page, and rewrite it. However, the flash_area module doesn't provide an API to get block size (but only write block size), so the flash library must be used directly to get this information or it must be hard-coded, e.g.:

```
  /* assume we want to write at offset 0 of block FLASH_OFFSET */
  const struct flash_area *f = open_flash();
  struct flash_pages_info info;
  flash_get_page_info_by_offs(f->fa_dev, FLASH_OFFSET, &info);
  size_t bs = info.size;  /* 4096 in my case */
  assert(bs <= sizeof(flash_page_buffer));
  flash_area_read(f, FLASH_OFFSET, &flash_page_buffer, bs);
  flash_area_erase(f, FLASH_OFFSET, bs);
  memcpy(&flash_page_buffer, &write_buffer, sizeof(write_buffer));
  flash_area_write(f, FLASH_OFFSET, &flash_page_buffer, bs);
```

**Describe the solution you'd like**
Since the flash_area module supports an erase operation, it should provide an API to get the flash block size, which is information necessary to perform the erase operation.

The flash_area module also already supports returning similar information (write block size), so it wouldn't be unreasonable to add a similar operation to get the block size.

**Describe alternatives you've considered**
Bypassing the flash_area module and using the flash module to get the information, or hardcoding the information.

**Additional context**
I would be happy to submit a PR to support this new operation.

Do you think the driver behaviour I described above is also a bug that should be reported? I.e. write block size < block size, but the driver doesn't really perform write block sized write operations.

nordicjm, 2023-05-23:
> However, the flash_area module doesn't provide an API to get block size (but only write block size)

Are you sure it's not returning the page size i.e. erase size? Because this is working fine for others e.g. 4KB sector sizes with 1-4 byte writes

Laczen, 2023-05-23:
@vmsh0, working with flash has some limitations as flash can only write to erased pages. The use of a read, modify, write procedure should be discouraged because it can leave whatever that is using this method (e.g. a filesystem, ...) in a unrecoverable state when the procedure is interrupted. That is why this behavior is not included in the flash area write API.

Also the erase-block-size can vary over the flash region (the write-block-size on the other hand is constant), that is why the flash page api needs to be used to find out the erase-block-size. Adding this to the flash area api would be more complex than it seems.

vmsh0, 2023-05-23:
> Are you sure it's not returning the page size i.e. erase size? Because this is working fine for others e.g. 4KB sector sizes with 1-4 byte writes

Sorry for the terminology confusion. It's returning 4, what I referred to as "write page size", and an erase of size 4 results in:
`E: unaligned address: 0x0007a000:4`

I haven't tested other erase sizes other than 4 and 4K (which I referred to as "page size" and I think you are referring to as "sector size" - anyhow, the size in `flash_get_page_info_by_offs(..., 0x7a000, ...).size`).

> Also the erase-block-size can vary over the flash region (the write-block-size on the other hand is constant), that is why the flash page api needs to be used to find out the erase-block-size. Adding this to the flash area api would be more complex than it seems.

The new flash_area API I was imagining would be offset-based, like its flash counterpart, so one can individually deal with single pages. This would make sense to me as the entire flash_area API is offset-based, including the `flash_area_erase` function which would be the primary user of the information returned by the new API.

de-nordic, 2023-05-23:
> The new flash_area API I was imagining would be offset-based, like its flash counterpart, so one can individually deal with single pages. This would make sense to me as the entire flash_area API is [...]

I do not get the idea. Everything here is offset based, you just are bound to some alignment, which differs between write, erase and read. Nothing stops you from selecting the highest alignment and pretending that you work with "blocks" of desired size.
That does not even require API change as it would be just utility functions using API calls to do this.

Laczen, 2023-05-23:
@vmsh0 it is not clear to me how this new flash_area API would be different from the present one.

A long time ago I presented an alternative to flash_area (https://github.com/zephyrproject-rtos/zephyr/pull/36979) that would allow overriding the flash erase-block-size and (for the first proposal) would even support subpartitions and read-only partitions. Overriding the erase-block-size allows support for areas with non-constant erase-block-size and allows packing multiple flash erase-blocks into one erase-block (as is used e.g. by nvs). This was not accepted and since then I have lost interest in trying to improve this part of zephyr. I am never using the flash_area API and always make a shim around the flash interface that erases a block when it is first written to. I am only using the dts definitions to extract the flash partition info.

Laczen, 2023-05-23:
OK, it is becoming clear. I see the following disadvantages:

1. It would make the flash_area API dependent on `CONFIG_FLASH_PAGE_LAYOUT`,
2. It would require the routine to use the same methodology as in `FLASH_PAGE_LAYOUT`, because just a size does not give sufficient information in the case of non-constant block sizes (also the offset is needed).

Laczen, 2023-05-23:
> > OK, it is becoming clear. I see the following disadvantages:
> > ```
> > 1. It would make the flash_area API dependent on `CONFIG_FLASH_PAGE_LAYOUT`,
> > ```
> 
> That is already the case. `flash_map_layout.c` is an existing part of the API that's conditionally compiled in when `FLASH_PAGE_LAYOUT` is enabled
> 
But this is not required for the flash_area API (it does not use these routines).
> > ```
> > 2. It would require the routine to use the same methodology as in `FLASH_PAGE_LAYOUT`, because just a size does not give sufficient information in the case of non-constant block sizes (also the offset is needed).
> > ```
> 
> I believe my previous post addresses that. The information returned by the proposed API is offset-dependant. Am I misunderstaing something?

I think you might be misunderstanding the problem with non-constant block flashes: to perform a correct erase you need both the offset and the size (the "page"-offset is not guaranteed to be aligned with the "page"-size), this is why the page layout is returning the flash_pages_info (that contains both the offset and the size of a page).

vmsh0, 2023-05-23:
> > That is already the case. `flash_map_layout.c` is an existing part of the API that's conditionally compiled in when `FLASH_PAGE_LAYOUT` is enabled
> 
> But this is not required for the flash_area API (it does not use these routines).

The `flash_map_layout.c` file _is_ part of the flash_area API. It implements the `flash_area_get_sectors` API function, which is a public function of the API declared [here](https://github.com/zephyrproject-rtos/zephyr/blob/b9ea2c27053f41c9b2d51aa461b51d77d1a8d27f/include/zephyr/storage/flash_map.h#L212) (albeit its interface is inconsistent with the rest of the API as described previously).

> I think you might be misunderstanding the problem with non-constant block flashes: to perform a correct erase you need both the offset and the size (the "page"-offset is not guaranteed to be aligned [...]

If you have the offset, and you call a function that, given an offset, gets you the size, then you have both the offset and the size. This is the intended usage of the proposed API function. A concrete example:

```
#define SOME_PAGE 0x0
const struct flash_area *f = open_flash();
size_t page_sz = flash_area_page_size(f, SOME_PAGE);
flash_area_erase(f, SOME_PAGE, page_sz);
```

Of course, that assumes that the user has already obtained the page offset of the page they want to manipulate. This is an assumption that is already made throughout the entire `flash_area` module, as the only way to escape the assumption is calling an API function which performs an unknown-sized memory allocation.

About that, the new proposed API function `flash_area_page_size` offers a concrete alternative, as it would allow a user to iterate over the available flash pages without memory allocations:

```
size_t off = 0;
size_t page_sz;
size_t total_sz = fa->fa_size;
while (total_sz < off) {
  page_sz = flash_area_page_size(fa, off);
  /* do something with page_sz */
}
```

(Or an additional non-allocating API function for iteration could be defined similarly to `flash_area_foreach`, e.g. `flash_area_page_foreach`, to implement this same functionality in a consistent way.)

Laczen, 2023-05-24:
After going through the code again: flash_area API is equivalent to flash API on a flash area:
* flash API: the erase will work when provided with the correct parameters without needing the page_layout information. When it is needed to retrieve page information the user should enable  `CONFIG_FLASH_PAGE_LAYOUT` and use this information before executing an erase.
* flash_area API: the same should be true (in this case the `flash_area_get_sectors` should be used if needed to find out sector information before executing an erase).

This is how it is at the moment.

vmsh0, 2023-05-24:
@Laczen Indeed, that what I was trying to say! As exemplified, the proposed API function would allow to find out sector information without calling `flash_area_get_sectors`, which performs as unknown-sized memory allocation - something that is often preferable to avoid on devices where operating systems like Zephyr run (hence the design of e.g. `flash_area_foreach` in the very same module)

de-nordic, 2023-05-25:
@vmsh0 There was attempt to avoid "foreaching" flash layout to get page info that went through TSC twice but failed to gather reviews from platform maintainers:
https://github.com/zephyrproject-rtos/zephyr/pull/36484
It was supposed to use direct knowledge on flash layout by driver to speed up getting info on page size by given offset.
As you can imagine Flash Map API wrapper for such call would be quite simple.

Laczen, 2023-05-25:
> @Laczen Indeed, that what I was trying to say! As exemplified, the proposed API function would allow to find out sector information without calling `flash_area_get_sectors`, which performs as [...]

You could add a `flash_area_get_page_info_by_offs` that adds the area offset and then calls `flash_get_page_info_by_offs` and then substract the area offset again from the `page_info`. As it is not guaranteed that a `flash_partition` is aligned with the erase blocks a check should be added that the page_info still belongs to the flash_area  and error out if it doesn't. This routine should also only be available when `CONFIG_FLASH_PAGE_LAYOUT` is selected.

Regarding the name of the routine, it would be preferable to use `flash_area_get_eraseblock_info_by_offs` instead of `flash_area_get_page_info_by_offs`.

Laczen, 2023-05-25:
> Particularly, you raise the interesting issue of an erase block (page/sector) not being aligned with a partition. Given that, indeed, nowhere it is claimed that this alignment would exist, are we aware of any use cases (or in general any sane situations) where the opposite could be true? A shared erase block kind of clashes with the traditional semantics of a storage partition.
>

This could intentionally be used to create a region that cannot be erased, it does not have to be a shared block, if it is just defined one byte less than the erase-block to avoid the erase by flash_area_erase. As far as I know it is not used by anyone in this way, and I would not encourage it.
 
> > Regarding the name of the routine, it would be preferable to use `flash_area_get_eraseblock_info_by_offs` instead of `flash_area_get_page_info_by_offs`.
> 
> Any particular reason for this? I see that the terms page and sector are used somehow interchangeably to indicate the concept of an "erase block". But in the API of either the flash and flash_area modules I have never come across the term, so I'm not sure that introducing it now would make things better.

The name `page` should really be reserved for the largest supported write block size, the use of `page` in `flash_page` has been confusing a lot of times and zephyr uses `erase-block` (except in the `flash_page` support). It is something difficult to remove from zephyr and nobody has volunteered to change the confusing name.

Laczen, 2023-05-25:
@de-nordic, the `flash_get_page_info_by_offs` routine is not really using the `foreach` approach in cases where the layout is constant. Then they are using the same approach as proposed in #36484. The speed increase would be very little. A possible improvement would be to suppose that any size is a 2's complement, but maybe the compiler is smart enough to figure this out.

vmsh0, 2023-05-26:
I think these last messages clarify everything for me. Thank you.

Would you be interested in a PR? I'm not trying to oversell the idea. If you are not convinced this would be useful we can close this.

Laczen, 2023-05-26:
> I think these last messages clarify everything for me. Thank you.
> 
> Would you be interested in a PR? I'm not trying to oversell the idea. If you are not convinced this would be useful we can close this.

I am not against a PR for this, as long as it is enclosed in the correct `CONFIG_FLASH_PAGE_LAYOUT` dependency.

Maybe you could also make a PR for the function `flash_area_get_sectors()` that should only be available when `CONFIG_FLASH_PAGE_LAYOUT` is enabled.