zephyrproject-rtos/zephyr #58170: device crash while including CONFIG_FILE_SYSTEM

nintyninebottlesx, 2023-05-23:
am using NCS 2.2.0 Zephyr  - nrf2840 Xiao BLE board with a a xiao_ble.overlay file to specify SPI

I am in the process of getting a zephyr,sdmmc-disk SPI device working.

The device appears to crash when I specifiy CONFIG_FILE_SYSTEM=y

the serial out shows
<err> fs: fs_stat: mount point not found!!
<err> fs: fs_mkdir: mount point not found!!

other related CONFIG Settings I am including without issue are
CONFIG_FS_FATFS_MOUNT_MKFS=n

CONFIG_FAT_FILESYSTEM_ELM=y

CONFIG_DISK_ACCESS=y
CONFIG_SPI_NRFX_RAM_BUFFER_SIZE=8
CONFIG_DISK_DRIVER_SDMMC=y
CONFIG_SPI=y
CONFIG_SDMMC_SUBSYS=y

The device itself seems to work after adding the code from zephyr\samples\subsys\fs\fat_fs\src
I get serial printout of
Sector Size 512
Memory Size(MB) 476
Disk Mounted

Listing dir /SD: ...

If I strip out any code associated with the Card Reader and include files.
Just purely adding the CONFIG_FILE_SYSTEM=y  

will give on serial 
<err> fs: fs_stat: mount point not found!!
<err> fs: fs_mkdir: mount point not found!!

and it will seemingly crash never getting to run Bluetooth BLE or printing out other printk lines.

**Expected behavior**
I would expect the included libraries and the code in Main to specify the mount point and allow the the card reader to either work or fail.

**Impact**
Stops the rest of the code from running in Main function

**Logs and console output**

*** Booting Zephyr OS build v3.2.99-ncs1 ***
[00:00:02.014,801] [0m<inf> sd: sd_check_card_type: Detected SD card[0m
[00:00:02.301,116] [0m<inf> sd: sdmmc_card_init: Maximum SD clock is under 25MHz, using clock of 24000000Hz[0m
Sector size 512
Memory Size(MB) 476
Disk mounted.

Listing dir /SD: ...
Counter alarm

Starting Bluetooth Central and Peripheral 
Button Debouncing
[00:00:05.613,250] [1;31m<err> fs: fs_stat: mount point not found!![0m
[00:00:05.613,281] [1;31m<err> fs: fs_mkdir: mount point not found!![0m
[00:00:05.613,281] [1;31m<err> bt_settings: bt_settings_init: settings_subsys_init failed (err -2)[0m

**Environment (please complete the following information):**
 NCS 2.2.0 Zephyr

**Additional context**
My device is using Bluetooth
SSD1306 Display using I2C
MMC using SPI
USB_ACM for serial output.

nordicjm, 2023-05-23:
NCS differs vastly from zephyr, you will need to ask this question on https://devzone.nordicsemi.com/

closed 2023-05-23 by nordicjm

nintyninebottlesx, 2023-05-23:
OK I was hoping to glean insight into what exactly is going on just simply with the including of the CONFIG flag itself.
If its going against something like how my USB_ACM logging has deferred.

CONFIG_LOG_MODE_MINIMAL=y
# Since the default LOG_MODE_MINIMAL is synchronous, it will mess with the USB
# driver/subsys. Use the deferred mode instead.
CONFIG_LOG_MODE_DEFERRED=y

or if its related to something else hidden like buffering such as CONFIG_SPI_NRFX_RAM_BUFFER_SIZE=8.

It just seems hard knowing the interactions of various CONFIG flag combinations sometimes.