zephyrproject-rtos/zephyr #58175: Coverage broken in main for ztests

aescolar, 2023-05-23:
**Describe the bug**
One cannot build ztests with coverage enabled by default after https://github.com/fabiobaltieri/zephyr/commit/f5830f3c3fe278bdecea12d124575760664dfae2 

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p native_posix -T tests/kernel/ --coverage
2. See all cmake failures

**Expected behavior**
Coverage working for ztests as before (with no optimizations wherever possible so the coverage report is accurate :) )

**Logs and console output**
```
...
ERROR   - native_posix              tests/kernel/queue/kernel.queue                     ERROR : Cmake build failure
ERROR   - Cmake build failure: zephyr/tests/kernel/mem_slab/mslab_api for native_posix
ERROR   - native_posix              tests/kernel/mem_slab/mslab_api/kernel.memory_slabs.api  ERROR : Cmake build failure
ERROR   - Cmake build failure: zephyr/tests/kernel/mem_heap/mheap_api_concept for native_posix
ERROR   - native_posix              tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap  ERROR : Cmake build failure
ERROR   - Cmake build failure: zephyr/tests/kernel/events/event_api for native_posix
ERROR   - native_posix              tests/kernel/events/event_api/kernel.events         ERROR : Cmake build failure
...
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Host gcc 12
 - Commit SHA or Version used: f5830f3c3fe278bdecea12d124575760664dfae2 

**Additional context:**
Please don't hurry a fix/merge just disabling no optimizations when coverage is enabled.

aescolar, 2023-05-23:
CC @fabiobaltieri @nashif @henrikbrixandersen @carlescufi

henrikbrixandersen, 2023-05-23:
Could we turn on `CONFIG_ZTEST_ALLOW_NO_OPTIMIZATIONS` when running the coverage tests?

henrikbrixandersen, 2023-05-23:
> @henrikbrixandersen We could, but it would seem like that cmake error is not really what we want if we just bypass it when we enable coverage.

The CMake error was introduced as a means to avoid all the "false" bug reports coming in from people running tests with no optimizations and seeing stack overflows. The idea was to only allow no optimizations for tests known to be designed to run without optimizations. I don't believe anybody had considered coverage, when we reviewed this.

aescolar, 2023-05-23:
@henrikbrixandersen Thanks, I see. I wonder if there would be some other better option with less collateral effects..
What about (just brainstorming):
1. Adding a clear note in the bug report template telling people about this.
2. Not have cmake error but just warn (an actual warning would still break CI), with a warning that is more descriptive ("With this option enabled you may see you run out of stack, code may be too slow to pass, etc..") and which can be turned off for some targets.
3. Have the kconfig description for the option warn people of the same.
4. The opposite of 2 (have some targets or tests declare themselves as constrained) and cmake warn about the combination of no_opts and that flag. (this could be tedious though)
5. A combination of 1,2,3

henrikbrixandersen, 2023-05-23:
> 1. Adding a clear note in the bug report template telling people about this.

Good idea.

> 2. Not have cmake error but just warn (an actual warning would still break CI), with a warning that is more descriptive ("With this option enabled you may see you run out of stack, code may be too slow to pass, etc..") and which can be turned off for some targets.

That might catch most issues.

> 3. Have the kconfig description for the option warn people of the same.

I already added that in 26ebceed55b94b6e9bc98e2bf4c892a05404bf16 - or at least I tried to make people aware of it.

fabiobaltieri, 2023-05-23:
Hi Alberto, sorry for the noise, to answer https://github.com/zephyrproject-rtos/zephyr/pull/58163#issuecomment-1558673033 (maybe let's try to keep the conversation on this issue here):

> what was the motivation for adding this https://github.com/zephyrproject-rtos/zephyr/blob/main/CMakeLists.txt#L709-L710 ?

The bit about "don't file a bug if a random test fails if you build with no optimizations", it's really not not like we get a huge amount of these but we do triage bugs manually, so it's a small bit to try and avoid a death by thousand cuts situation as the project grows.

>     1. Adding a clear note in the bug report template telling people about this.

Sure it's an option but that template is starting to get a bit bloated, and something one read the first time (maybe) and then just ignore and delete every time. Since there's a clear trigger for this situation I think it's worth trying to use it to warn the user on the specific event if possible.

>     2. Not have cmake error but just warn (an actual warning would still break CI), with a warning that is more descriptive ("With this option enabled you may see you run out of stack, code may be too slow to pass, etc..") and which can be turned off for some targets.

That's an option, my concern in this case is that we'd start adding more of these (again thinking of project growth) and then having warning firing all the time and lost in the noise.

@aescolar I'd like to do another try and just disable the error on coverage runs (https://github.com/zephyrproject-rtos/zephyr/pull/58176) before we downgrade it to a warning entirely. Let me know what you think.

On a separate note, and still on the long term operational front, what bugs me even more than this is the fact that the breakage was not caught by the PR CI run, and as far as I'm aware it's not first time that coverage breaks and it gets noticed downstream. Can we just do a simple reference test that build with coverage enable and check that it actually worked and runs in all PRs?

aescolar, 2023-05-23:
>>  1. Adding a clear note in the bug report template telling people about this.

> Sure it's an option but that template is starting to get a bit bloated, and something one reads the first time (maybe) and then just ignore and delete every time. 

I agree many would miss it. Still it could help a bit(?).

>>   2. Not have cmake error but just warn ..

> That's an option, my concern in this case is that we'd start adding more of these (again thinking of project growth) and then having warning firing all the time and lost in the noise.

Two thoughts:
a) The issue with erroring out is that it is a too dramatic response. Building ztests with no optimizations, is not a failure at all, it is actually working well in many cases, and helpful.
b) I would still have a waiver for that warning (just like you added for the error), and have that waiver option be set up by default for all sensible combinations in which no problems are expected.

> Can we just do a simple reference test that build with coverage enable and check that it actually worked and runs in all PRs?

That is a good idea. Though note that all babblesim tests are built with coverage by default in CI, so in this case it would have been the combination of ztests and coverage that was not covered in CI.

closed 2023-05-23 by fabiobaltieri