zephyrproject-rtos/zephyr #58202: rimage not fetched with west update

teburd, 2023-05-23:
**Describe the bug**
west update should pull in rimage and its configuration files, either using submodules or the sof manifest or in the zephyr manifest

sof specified rimage both as a relative submodule and in the sof west.yml manifest. In either case rimage isn't cloned and it should be.

Using submodules: true in zephyr's manifest for sof doesn't work however as its a relative path submodule.
Using import: true in zephyr's manifest for sof doesn't work as sof depends on zephyr.

The result is building any tests or samples for intel adsp platforms fails with the default signing parameters because rimage's config directory isn't where its expected to be.

**To Reproduce**
Steps to reproduce the behavior:

Setup a clean zephyr work space and checkout zephyr main

west build -p always -b intel_adsp_cavs25 samples/hello_world

Results in...

```
rimage -o /home/tburdick/z/zephyr/build/zephyr/zephyr.ri -e -k /home/tburdick/z/modules/audio/sof/keys/otc_private_key_3k.pem -c /home/tburdick/z/modules/audio/sof/rimage/config/tgl.toml /home/tburdick/z/zephyr/build/zephyr/boot.mod /home/tburdick/z/zephyr/build/zephyr/main.mod
/home/tburdick/z/modules/audio/sof/rimage/config/tgl.toml:	error: unable to open file for reading. No such file or directory (errno = 2)
```

**Expected behavior**
Working builds with signing

marc-hb, 2023-05-23:
Note the current instructions (thanks @laurenmurphyx64 !) provide a good workaround:
https://docs.zephyrproject.org/latest/boards/xtensa/intel_adsp_cavs25/doc/intel_adsp_generic.html#signing-tool
```
cd zephyrproject/modules/audio/sof/
git clone https://github.com/thesofproject/rimage --recurse-submodules
cd rimage
```

marc-hb, 2023-05-24:
If you want to clone `rimage` in a different directory for some (not recommended) reason, then these extra steps are enough:

```shell
rm -rf build
west config rimage.path /home/me/build-rimage/rimage 
west config build.cmake-args -- -DRIMAGE_CONFIG_PATH=/path/to/rimage/config
west config -l
west build ... 
```

I'm working on a documentation update (before #58212)

marc-hb, 2023-05-24:
Relevant documentation update submitted in:
- https://github.com/zephyrproject-rtos/zephyr/pull/58215

nashif, 2023-11-05:
rimage is now part of SOF.

closed 2023-11-05 by nashif