zephyrproject-rtos/zephyr #58205: eswifi: easy to deadlock eswifi driver with socket offload disabled and concurrent send/recv

bdunlay, 2023-05-23:
**Describe the bug**

With socket offload disabled, concurrent threads for socket send/recv eswifi can easily deadlock.

My understanding of the bug is based on these observations:

1. The `send()` thread takes a **socket mutex** and likely gets interrupted by a higher priority background eswifi task. 
1. The eswifi background task takes the **eswifi mutex**, and then may try to take the **socket mutex** in the socket receive callback, which is presently held by the `send()` thread.
1. Blocked, the `send()` thread resumes and tries to take the **eswifi mutex**, which is held by the background task.

Additional information:

 - What target platform are you using? STM32 B_L5S4I_IOT01A
 - What have you tried to diagnose or workaround this issue? I tried diagnosing with a test program and gdb stack analysis.
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? No.

**To Reproduce**
See attached program. Use `netcat -lu 5678` to receive and send data back to the device after each message. Repeat until deadlock, which doesn't take long.

**Expected behavior**
Harmonious concurrency in socket communications.

**Impact**
Concurrent threads for send/receive is consistently unreliable when disabling socket offload. Socket offload is disabled because offload sockets don't fully support zsock_poll features.

**Logs and console output**

Send thread
```
#0  arch_swap (key=key@entry=0) at /Users/<user>/zephyrproject/zephyr/arch/arm/core/aarch32/swap.c:53
53              return _current->arch.swap_return_value;
(gdb) bt
#0  arch_swap (key=key@entry=0) at /Users/<user>/zephyrproject/zephyr/arch/arm/core/aarch32/swap.c:53
#1  0x080610ce in z_swap_irqlock (key=0) at /Users/<user>/zephyrproject/zephyr/kernel/include/kswap.h:185
#2  z_swap (key=..., lock=0x2003d504 <announce_remaining>) at /Users/<user>/zephyrproject/zephyr/kernel/include/kswap.h:196
#3  z_pend_curr (lock=lock@entry=0x2003d4ec <pending_cancels>, key=key@entry=..., wait_q=wait_q@entry=0x20004754 <eswifi0+2028>, timeout=...) at /Users/<user>/zephyrproject/zephyr/kernel/sched.c:866
#4  0x08060822 in z_impl_k_mutex_lock (mutex=mutex@entry=0x20004754 <eswifi0+2028>, timeout=<error reading variable: DWARF-2 expression error: DW_OP_GNU_uninit must always be the very last op.>) at /Users/<user>/zephyrproject/zephyr/kernel/mutex.c:146
#5  0x0803c1fc in k_mutex_lock (timeout=..., mutex=0x20004754 <eswifi0+2028>) at /Users/<user>/zephyrproject/applications/my-app/build/zephyr/include/generated/syscalls/kernel.h:955
#6  eswifi_lock (eswifi=eswifi@entry=0x20003f68 <eswifi0>) at /Users/<user>/zephyrproject/zephyr/drivers/wifi/eswifi/eswifi.h:98
#7  0x08079570 in eswifi_off_sendto (pkt=0x2004fef8 <_k_mem_slab_buf_tx_pkts+192>, dst_addr=0x20002bec <tls_contexts+132>, addrlen=<optimized out>, cb=0x0, timeout=0, user_data=0x0) at /Users/<user>/zephyrproject/zephyr/drivers/wifi/eswifi/eswifi_offload.c:309
#8  0x080327ea in net_offload_sendto (iface=<optimized out>, pkt=pkt@entry=0x2004fef8 <_k_mem_slab_buf_tx_pkts+192>, dst_addr=dst_addr@entry=0x20002bec <tls_contexts+132>, addrlen=addrlen@entry=8, cb=cb@entry=0x0, timeout=..., timeout@entry=..., user_data=user_data@entry=0x0) at /Users/<user>/zephyrproject/zephyr/include/zephyr/net/net_offload.h:375
#9  0x08071ec4 in context_sendto (context=context@entry=0x2003a5ac <contexts+160>, buf=buf@entry=0x20052060, len=len@entry=188, dst_addr=dst_addr@entry=0x20002bec <tls_contexts+132>, addrlen=addrlen@entry=8, cb=cb@entry=0x0, timeout=timeout@entry=..., user_data=user_data@entry=0x0, sendto=sendto@entry=true) at /Users/<user>/zephyrproject/zephyr/subsys/net/ip/net_context.c:1721
#10 0x08072340 in net_context_sendto (context=context@entry=0x2003a5ac <contexts+160>, buf=buf@entry=0x20052060, len=len@entry=188, dst_addr=dst_addr@entry=0x20002bec <tls_contexts+132>, addrlen=addrlen@entry=8, cb=cb@entry=0x0, timeout=..., user_data=0x0) at /Users/<user>/zephyrproject/zephyr/subsys/net/ip/net_context.c:1887
#11 0x080268ae in zsock_sendto_ctx (ctx=0x2003a5ac <contexts+160>, buf=0x20052060, len=188, flags=<optimized out>, dest_addr=dest_addr@entry=0x20002bec <tls_contexts+132>, addrlen=addrlen@entry=8) at /Users/<user>/zephyrproject/zephyr/subsys/net/lib/sockets/sockets.c:788
#12 0x0806b8ca in sock_sendto_vmeth (obj=<optimized out>, buf=<optimized out>, len=<optimized out>, flags=<optimized out>, dest_addr=0x20002bec <tls_contexts+132>, addrlen=8) at /Users/<user>/zephyrproject/zephyr/subsys/net/lib/sockets/sockets.c:2543
#13 0x0806b928 in z_impl_zsock_sendto (sock=<optimized out>, buf=0x20052060, len=188, flags=64, dest_addr=dest_addr@entry=0x20002bec <tls_contexts+132>, addrlen=addrlen@entry=8) at /Users/<user>/zephyrproject/zephyr/subsys/net/lib/sockets/sockets.c:818
#14 0x08026d80 in zsock_sendto (addrlen=8, dest_addr=0x20002bec <tls_contexts+132>, flags=<optimized out>, len=<optimized out>, buf=<optimized out>, sock=<optimized out>) at /Users/<user>/zephyrproject/applications/my-app/build/zephyr/include/generated/syscalls/socket.h:258
```

Background work thread

```
#0  arch_swap (key=key@entry=0) at /Users/<user>/zephyrproject/zephyr/arch/arm/core/aarch32/swap.c:53
53              return _current->arch.swap_return_value;
(gdb) bt
#0  arch_swap (key=key@entry=0) at /Users/<user>/zephyrproject/zephyr/arch/arm/core/aarch32/swap.c:53
#1  0x080610ce in z_swap_irqlock (key=0) at /Users/<user>/zephyrproject/zephyr/kernel/include/kswap.h:185
#2  z_swap (key=..., lock=0x2003d504 <announce_remaining>) at /Users/<user>/zephyrproject/zephyr/kernel/include/kswap.h:196
#3  z_pend_curr (lock=lock@entry=0x2003d4ec <pending_cancels>, key=key@entry=..., wait_q=wait_q@entry=0x200007b4 <fdtable+140>, timeout=...) at /Users/<user>/zephyrproject/zephyr/kernel/sched.c:866
#4  0x08060822 in z_impl_k_mutex_lock (mutex=0x200007b4 <fdtable+140>, timeout=<error reading variable: DWARF-2 expression error: DW_OP_GNU_uninit must always be the very last op.>) at /Users/<user>/zephyrproject/zephyr/kernel/mutex.c:146
#5  0x0806b36a in k_mutex_lock (timeout=..., mutex=<optimized out>) at /Users/<user>/zephyrproject/applications/my-app/build/zephyr/include/generated/syscalls/kernel.h:955
#6  zsock_received_cb (ctx=0x2003a5ac <contexts+160>, pkt=0x2004ffb8 <_k_mem_slab_buf_rx_pkts+128>, ip_hdr=<optimized out>, proto_hdr=<optimized out>, status=0, user_data=0x0) at /Users/<user>/zephyrproject/zephyr/subsys/net/lib/sockets/sockets.c:377
#7  0x0803c978 in eswifi_off_read_work (work=0x200048b8 <eswifi0+2384>) at /Users/<user>/zephyrproject/zephyr/drivers/wifi/eswifi/eswifi_socket.c:166
#8  0x08060bf2 in work_queue_main (workq_ptr=0x20003f78 <eswifi0+16>, p2=<optimized out>, p3=<optimized out>) at /Users/<user>/zephyrproject/zephyr/kernel/work.c:668
#9  0x080692a8 in z_thread_entry (entry=0x8060b69 <work_queue_main>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /Users/<user>/zephyrproject/zephyr/lib/os/thread_entry.c:36

```

Examining the mutexes held by each thread:

SEND CLIENT THREAD (pending lock held by eswifi thread, already holds socket lock)
```
  9    Thread 536883928 "SEND_CLIENT_THREAD_tid" (Name: SEND_CLIENT_THREAD_tid, prio:-2,useropts:0)      arch_swap (key=key@entry=0) at /Users/<user>/zephyrproject/zephyr/arch/arm/core/aarch32/swap.c:53

(gdb) print eswifi->mutex
$24 = {wait_q = {waitq = {{head = 0x20004b40 <eswifi_spi0+8>, next = 0x20004b40 <eswifi_spi0+8>}, {tail = 0x200032d8 <_k_thread_obj_SEND_CLIENT_THREAD_tid>,
        prev = 0x200032d8 <_k_thread_obj_SEND_CLIENT_THREAD_tid>}}}, owner = 0x20003f78 <eswifi0+16>, lock_count = 1, owner_orig_prio = -2}
```

BACKGROUND WORKQUEUE THREAD (pending lock held by client thread, already holds eswifi lock)
```
  13   Thread 536887160 "thr_8060b69_20003f78" (Name: thr_8060b69_20003f78, prio:-14,useropts:0) arch_swap (key=key@entry=0) at /Users/<user>/zephyrproject/zephyr/arch/arm/core/aarch32/swap.c:53

(gdb) print *ctx->cond.lock
$23 = {wait_q = {waitq = {{head = 0x20003f78 <eswifi0+16>, next = 0x20003f78 <eswifi0+16>}, {tail = 0x20003f78 <eswifi0+16>, prev = 0x20003f78 <eswifi0+16>}}}, owner = 0x200032d8 <_k_thread_obj_SEND_CLIENT_THREAD_tid>, lock_count = 1, owner_orig_prio = 0}

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): MacOS
 - Toolchain (e.g Zephyr SDK, ...): 0.15.2
 - Commit SHA or Version used: v3.3.0

**Additional context**

Socket offload disabled.

bdunlay, 2023-05-23:
Sample program. [udp.zip](https://github.com/zephyrproject-rtos/zephyr/files/11547360/udp.zip) 

Use `netcat -lu 5678` to receive and send data back to the device after each message. Repeat sending data back to the device until deadlock, which doesn't take long.

bdunlay, 2023-05-23:
@loicpoulain 😀

rftafas, 2023-05-31:
@jgl-meta Question: why is this under platform ESP32 if the target is STM32? I think there have been a misunderstanding, as ESP-AT runs ESP-IDF firmware and its bugs or misbehavior should be reported at proper (other) forums.

I believe this is a networking issue.

jgl-meta, 2023-05-31:
> @jgl-meta Question: why is this under platform ESP32 if the target is STM32? I think there have been a misunderstanding, as ESP-AT runs ESP-IDF firmware and its bugs or misbehavior should be reported at proper (other) forums.
> 
> I believe this is a networking issue.

The bug triage group interpreted this as the reporter is going through esp32 for wifi I believe. That was the reasoning for adding the tag.

bdunlay, 2023-06-01:
I can confirm that this bug is not related to ESP32. This is somewhat related to STM32 as it's on an STM32 dev board, but if my analysis is correct, it's just a driver bug for the Inventek es-WiFi module, which makes it platform independent.

closed 2023-08-16 by github-actions[bot]