zephyrproject-rtos/zephyr #58212: RFC: Add rimage and tomlc99 to Zephyr's west manifest

teburd, 2023-05-23:
## Introduction

Include rimage and tomlc99 in the manifest for use with west sign when building for intel adsp targets

See issue #58202 for more details

### Problem description

rimage is an elf metadata and signing tool use by the sof project and needed for intel adsp platforms to build loadable images. Having rimage and tomlc99 in the zephyr west manifest would let zephyr build and have available to it various rimage configurations needed for signing.

### Proposed change

Directly add rimage and tomlc to zephyr's west.yml, host rimage and tomlc under the zephyr project github organization

marc-hb, 2023-05-31:
Approved:
- https://lists.zephyrproject.org/g/tsc/message/1676
- https://docs.google.com/document/d/1cMkXkFN-Yje-7zIGZmWMfkrVCju85UNnKDfjng9vEv0

aescolar, 2023-06-01:
Is the plan for this to be added to the zephyr manifest disabled by default with a group filter, and have the sof manifest enable that group by default?

marc-hb, 2023-09-07:
@nashif 's suggestion to merge rimage and SOF:
- https://github.com/thesofproject/sof/issues/8178

marc-hb, 2023-10-09:
rimage transfer to sof.git submitted in:
- https://github.com/thesofproject/sof/pull/8297