zephyrproject-rtos/zephyr #58219: I2C Polling Mode in STM32H7 for 1000 byte of data

RadhikaMaheshwari-eaton, 2023-05-24:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/58131

<div type='discussions-op-text'>

<sup>Originally posted by **RadhikaMaheshwari-eaton** May 22, 2023</sup>
Hi Community,

I am trying to write and read data of 1 byte after the transaction of 1000 data of 1 byte through i2c_stm32_transfer() in i2c_ll_stm32.c, I am getting incorrect response (tx_data != rx_data) in I2C polling mode (controller). Is there a global buffer which is getting filled and not able to clear the contents, the pointer is stuck somewhere, or the data register is pre-occupied. What could be the possible reason? And what is the maximum no. of data bytes we can transfer in I2C Polling mode?</div>

On further debugging we found out that there is a timeout issue in stm32_i2c_msg_read() function in i2c_ll_stm32_v2.c driver for H743 controller whereas this issue is fixed in v1 driver. Want the same configurations and specification for v2 as well.

erwango, 2023-05-24:
@RadhikaMaheshwari-eaton If you have the code that help to reproduce the issue, can you share it ?
This will be much helpful as it allows to quickly reproduce, analyse and fix the issue.
Also, if you already know the fix to apply, don't hesitate to contribute directly.

RadhikaMaheshwari-eaton, 2023-05-24:
Here is the code which I am using:
[image]

- If we transfer 1000 data of 1 byte through this program, we get correct response in terms of written data is equal to read data.
- But in second iteration if we send 1 byte of data and try to read it then i2c_stm32_msg_read() API requires time to process the query.

Like in i2c_ll_stm32_v1.c driver, we have timeout delay in the same function i2c_stm32_msg_read() for I2C polling mode. We can get reference from there.

RadhikaMaheshwari-eaton, 2023-05-25:
[new 1.txt](https://github.com/zephyrproject-rtos/zephyr/files/11561245/new.1.txt)
[new 2.txt](https://github.com/zephyrproject-rtos/zephyr/files/11561255/new.2.txt)
Here is text version of the code and prj.conf
I am using STM32H743ZI nucleo board.

closed 2023-08-08 by github-actions[bot]